home-assistant/operating-system #3149: Host lost static IP address after update to 11.5

SmartHomeFanatic, 2024-02-05:
### Describe the issue you are experiencing

Updated my instance to 11.5 and the host lost its static IP configuration and became unreachable (running DHCP on HA).

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Updated HAOS

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

## System Information

version | core-2024.1.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.74-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4587
Installed Version | 1.34.0
Stage | running
Available Repositories | 1395
Downloaded Repositories | 74

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 16. Februar 2024 um 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
remote_server | eu-central-1-0.ui.nabu.casa
certificate_status | ready
instance_id | 5a50d9881a094593a412cbfd346b0be1
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.5
-- | --
update_channel | stable
supervisor_version | supervisor-2024.01.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 916.2 GB
disk_used | 37.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.2.0), AdGuard Home (5.0.2), ESPHome (2023.12.9), Mosquitto broker (6.4.0), RPC Shutdown (2.4), Terminal & SSH (9.8.1), Valetudo Mapper (1.12.0), Matter Server (5.0.4), Ring-MQTT with Video Streaming (5.6.3), Home Assistant Google Drive Backup (0.112.1), AirCast (4.2.0), Studio Code Server (5.15.0), DOODS-cuda (2), TasmoAdmin (0.29.1), Frigate (0.13.1), openWakeWord (1.8.2), Docker Container Stats (1.5.0), MQTT Explorer (browser-1.0.3), EMQX (0.4.0), Jellyfin Server (1.1.4), Samba NAS (12.2.0-nas1), go2rtc (1.8.5), Netbox (3.7.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 41
views | 20
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 26. Januar 2024 um 14:16
-- | --
current_recorder_run | 5. Februar 2024 um 20:04
estimated_db_size | 1670.79 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.4 (a4a8c5f)
-- | --
cloud_online | 1 / 1
local_online | 1 / 1

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

gfn256, 2024-02-05:
Except for IP address is your instance of HA running? What does a display connected (if available) show?
Also, I don't know what you mean by "Static IP" in your title description, when later you state  "running DHCP on HA"

SmartHomeFanatic, 2024-02-05:
Had to connect a display to configure the static IP again. Interface showed "connected". Everything work after that.

"Running DHCP on HA" meant that I am running a DHCP server addon on the HA host and therefore was trapped in HA not getting an IP address.

agners, 2024-02-05:
Hm, the most realistic case how this could happen is if the network device got a different name.

Can you use `login` and check the network device name using `ip addr`? 

Then downgrade to 11.4 using `ha os update --version 11.4` and check the device name again?

tommes, 2024-02-06:
I'm experiencing the same issue on RPi 4 after upgrade to 11.5.

closed 2024-05-14 by github-actions[bot]