home-assistant/operating-system #3150: Fresh Odroid N2+ installation fails

BarBaar44, 2024-02-05:
### Describe the issue you are experiencing

### Hardware

New hardware:
- Odroid N2+ 4GB
- eMMC 32GB
- Power adapter that came with the unit

Old Hardware:
- Odroid C2, fully operational

### Issue

Fresh installation of haos 11.5 fails with a blank screen and no further system action

### What operating system image do you use?

odroid-c2 (Hardkernel ODROID-C2)

### What version of Home Assistant Operating System is installed?

n.a.

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Download haos 11.5 image
2. Follow installation instructions on https://www.home-assistant.io/installation/odroid/
3. boot system
4. Blank screen, no blue heartbeat

Second scenario is upgrading to haos 11.5 from existing install. Leads to same situation. See below for details and steps taken
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Blank screen... no logs...
```

### Anything in the Host logs that might be useful for us?

```txt
Reminds me of this one I had a while ago...

https://github.com/home-assistant/operating-system/issues/1142
```

### Additional information

1. First installed the latest image Odroid N2+ image, version 11.4. Copied to the eMMC as described on [this](https://www.home-assistant.io/installation/odroid/) page. Installed the eMMC on the board and powered up. No action. Red light on the board is on, blue stays off (no heartbeat). Blank screen, no messages or anything. The system just doesn't boot.
2. Version 11.5 became available. Tried this version, but the same result as before. No luck
3. Tried installing using the [Flashing Guide](https://www.home-assistant.io/common-tasks/os#flashing-an-odroid-n2). The same result for both the 11.4 and 11.5 versions of the image.
4.  Doubting if there is a hardware issue. Downloaded an Ubuntu image from the Odroid website and installed it. Worked flawlessly.
5. Tried copying the image (version 11.5) to the eMMC using dd:
`dd if=<image> of=<disk> bs=4M`
Same result. So ruled out hardware issues and Etcher issues.

Working with dd from now on..

6. Zeroing the disk:
`dd if=/dev/zero of=/dev/disc bs=4M`
8. Tried image version 10.5. Same result.
9. Tried version 8.0 of the image. Finally! The system booted! But... a failure during setup of the system. But, foremost happy I finally had HA booting on the hardware.
10. Tried version 9.5 of the image. YES! fully working system and finally possible to connect to HA using the website (http://xx.xx.xx.xx:8123)
11. Restoring image from the Old Hardware (see above). Long wait, but did work.
12. Working HA environment that needs to be updated to Home Assistant Operating System 11.5. Applied to update, and.... system no longer boots. Back at the initial situation with a blank screen without any messages. The red light on the board is on but no heartbeat.

Trying again:
1. Installing version 10.0 of the image. Fail
2. Installing version 9.5 of the image. Fail ... What?
3. zeroing the disk
4 Installing version 10.0 of the image.. boot!
5. Appears to be stuck in "waiting for the Home Assistant cli to be ready". Power cycle..
6.  No boot, no blue light. Copying a fresh 10.0  image back to the eMMC
7. Boot! And web interface is ready in very little time.
8. Restoring backup
9. Updating to haos 11.5 using buildin HA-update functionality
10. reboot.. blank screen and no heartbeat...
11. giving up for now...

sairon, 2024-02-07:
Since the behavior you describe is rather erratic, my guess is that the eMMC card/module is somehow damaged. The other, less likely scenario, is that something wrong is with your ODROID itself. ODROID N2+ has an SD card slot as well, could you try copying the system image to that and try again?

BarBaar44, 2024-02-07:
Ok...

Using a 64HB Samsung SD:
1. updated Petitboot (a couple of days ago already); pb_update
2. Copied the 11.5 image to the SSD using Etcher (running on Windows)
3. Switched the Odroid to SPI-boot
4. Inserted the SSD and power-on the Oidroid
5. Booting to Petitboot, but stuck there: "Processing new Disk Device".

Repeated with a different SD-card -> Same result
Repeated with a different image (non HA): works

Used a different converter (USB to SD) to copy the image to the SD-card (64Gb one) --> Stuck at "Processing new Disk Device" (same as (5)).

To summarize:
- Not working with SD or eMMC
- Not working using the OTG-USB method

--> I noticed I am missing a boot.ini in the image. See [here](https://forum.odroid.com/viewtopic.php?p=252597&sid=1a380d4ab7e6d2dbc055fdba4d2ffb61#p252597). Not the expert, but just noticed this.

I have no indications there is a hardware problem:
- Tried different discs all with the same result
- Tried OTG-USB, which rules out any issues during copying the image to the eMMC
- Successful installed the Ubuntu image as provided by Hardkernel

But I do run out of options to try and get it to work :/

sairon, 2024-02-07:
> Switched the Odroid to SPI-boot

That is wrong. HAOS on N2 does not use Petitboot but U-Boot in the first partition of the SD card/eMMC, so the boot switch should be set to MMC no matter whether you're booting from an eMMC or SD card. See the [diagram in ODROID wiki](https://wiki.odroid.com/odroid-n2/software/boot_sequence#boot_from_emmcmicro_sd). That also answers why there's no `boot.ini`.

BarBaar44, 2024-02-09:
Tnx..

The system seems to be working with the SD-Card. Guess the eMMc is bad. Weird as it is a new one.

Closing this one. Thnx for your support @sairon

closed 2024-02-09 by BarBaar44

piercy, 2024-03-25:
@sairon I dont think this is an issue with this guys MMC port, theres 2 more issues raised with a bug to do with N2+ not booting.  

see:
#3227 3227
#3250 3250

agners, 2024-04-02:
@piercy thanks for connecting these issues, I agree this seems all to be related. Let's track it in #3227 as this seems to be the issue with most information at this point.