home-assistant/operating-system #3154: Bluetooth Disappeared for Realtek c821

Veldkornet, 2024-02-07:
### Describe the issue you are experiencing

I'm currently using a [Bkouen Mini PC AK7 Pro](https://www.amazon.nl/Bkouen-Mini-pc-mini-computer-Ethernet-Busniess/dp/B0BHWZSQLJ); after installing OS 11.5, my bluetooth stopped working. I've done reboots etc, but still nothing.
[image]
```
Bluetooth Radio (0bda:c821)
by Realtek
Firmware: hassio
Hardware: usb:v1D6Bp0246d0546
```
That being said, I don't think it's related to OS 11.5 specifically, as I have another one of these devices and the bluetooth disappeared there too in an earlier version...

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update OS from 11.4 to 11.5

### Anything in the Supervisor logs that might be useful for us?

```txt
24-02-07 08:58:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-07 08:58:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
24-02-07 08:58:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
24-02-07 08:58:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret red_username
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret red_password
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret red_username
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret red_password
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret red_secret
24-02-07 09:03:55 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret ssh_username
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret ssh_key1
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret ssh_key2
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret portainer_agent_secret
24-02-07 09:03:55 WARNING (MainThread) [supervisor.addons.options] Option 'ssl' does not exist in the schema for Portainer (a0d7b954_portainer)
24-02-07 09:03:55 WARNING (MainThread) [supervisor.addons.options] Option 'certfile' does not exist in the schema for Portainer (a0d7b954_portainer)
24-02-07 09:03:55 WARNING (MainThread) [supervisor.addons.options] Option 'keyfile' does not exist in the schema for Portainer (a0d7b954_portainer)
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_username
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_username
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
24-02-07 09:03:55 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
24-02-07 09:05:22 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-02-07 09:05:56 INFO (MainThread) [supervisor.backups.manager] Backup 588ceddf starting stage addon_repositories
24-02-07 09:05:56 INFO (MainThread) [supervisor.backups.manager] Backup 588ceddf starting stage docker_config
24-02-07 09:05:56 INFO (MainThread) [supervisor.backups.manager] Creating new partial backup with slug 588ceddf
24-02-07 09:05:56 INFO (MainThread) [supervisor.backups.manager] Backup 588ceddf starting stage addons
24-02-07 09:05:56 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on a0d7b954_zwavejs2mqtt
24-02-07 09:05:58 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon a0d7b954_zwavejs2mqtt
24-02-07 09:05:58 INFO (MainThread) [supervisor.backups.manager] Backup 588ceddf starting stage finishing_file
24-02-07 09:05:58 INFO (MainThread) [supervisor.backups.manager] Creating partial backup with slug 588ceddf completed
24-02-07 09:05:58 INFO (MainThread) [supervisor.docker.addon] Updating image ghcr.io/hassio-addons/zwave-js-ui/amd64:3.3.1 to ghcr.io/hassio-addons/zwave-js-ui/amd64:3.3.2
24-02-07 09:05:58 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/hassio-addons/zwave-js-ui/amd64 with tag 3.3.2.
24-02-07 09:06:05 INFO (SyncWorker_1) [supervisor.docker.manager] Stopping addon_a0d7b954_zwavejs2mqtt application
24-02-07 09:06:35 INFO (SyncWorker_1) [supervisor.docker.manager] Cleaning addon_a0d7b954_zwavejs2mqtt application
24-02-07 09:06:35 INFO (MainThread) [supervisor.addons.addon] Add-on 'a0d7b954_zwavejs2mqtt' successfully updated
24-02-07 09:06:35 INFO (SyncWorker_3) [supervisor.docker.manager] Cleanup images: ['ghcr.io/hassio-addons/zwave-js-ui/amd64:3.3.1']
[... 250 lines cut ...]

<details><summary>NextDNS</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 21 January 2024 at 11:27
-- | --
current_recorder_run | 6 February 2024 at 23:33
estimated_db_size | 1816.28 MiB
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

Veldkornet, 2024-02-07:
dmesg:
```
[    0.034682] TSC deadline timer available
[    0.034683] smpboot: Allowing 4 CPUs, 0 hotplug CPUs
[    0.034701] PM: hibernation: Registered nosave memory: [mem 0x00000000-0x00000fff]
[    0.034703] PM: hibernation: Registered nosave memory: [mem 0x0009f000-0x000fffff]
[    0.034704] PM: hibernation: Registered nosave memory: [mem 0x75346000-0x75346fff]
[    0.034706] PM: hibernation: Registered nosave memory: [mem 0x760c8000-0x785c7fff]
[    0.034706] PM: hibernation: Registered nosave memory: [mem 0x785c8000-0x78847fff]
[    0.034707] PM: hibernation: Registered nosave memory: [mem 0x78848000-0x78947fff]
[    0.034708] PM: hibernation: Registered nosave memory: [mem 0x78948000-0x78efefff]
[    0.034708] PM: hibernation: Registered nosave memory: [mem 0x78eff000-0x78ffefff]
[    0.034709] PM: hibernation: Registered nosave memory: [mem 0x79000000-0x7fbfffff]
[    0.034710] PM: hibernation: Registered nosave memory: [mem 0x7fc00000-0xbfffffff]
[    0.034711] PM: hibernation: Registered nosave memory: [mem 0xc0000000-0xcfffffff]
[    0.034711] PM: hibernation: Registered nosave memory: [mem 0xd0000000-0xfdffffff]
[    0.034712] PM: hibernation: Registered nosave memory: [mem 0xfe000000-0xfe010fff]
[    0.034712] PM: hibernation: Registered nosave memory: [mem 0xfe011000-0xfebfffff]
[    0.034713] PM: hibernation: Registered nosave memory: [mem 0xfec00000-0xfec00fff]
[    0.034713] PM: hibernation: Registered nosave memory: [mem 0xfec01000-0xfecfffff]
[    0.034714] PM: hibernation: Registered nosave memory: [mem 0xfed00000-0xfed00fff]
[    0.034714] PM: hibernation: Registered nosave memory: [mem 0xfed01000-0xfed1ffff]
[    0.034715] PM: hibernation: Registered nosave memory: [mem 0xfed20000-0xfed7ffff]
[    0.034715] PM: hibernation: Registered nosave memory: [mem 0xfed80000-0xfedfffff]
[    0.034716] PM: hibernation: Registered nosave memory: [mem 0xfee00000-0xfee00fff]
[    0.034716] PM: hibernation: Registered nosave memory: [mem 0xfee01000-0xfeffffff]
[    0.034717] PM: hibernation: Registered nosave memory: [mem 0xff000000-0xffffffff]
[    0.034718] [mem 0x7fc00000-0xbfffffff] available for PCI devices
[    0.034720] Booting paravirtualized kernel on bare hardware
[    0.034723] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1910969940391419 ns
[    0.040764] setup_percpu: NR_CPUS:64 nr_cpumask_bits:4 nr_cpu_ids:4 nr_node_ids:1
[    0.041070] percpu: Embedded 54 pages/cpu s181352 r8192 d31640 u524288
[    0.041077] pcpu-alloc: s181352 r8192 d31640 u524288 alloc=1*2097152
[    0.041079] pcpu-alloc: [0] 0 1 2 3 
[    0.041103] Fallback order for Node 0: 0 
[    0.041105] Built 1 zonelists, mobility grouping on.  Total pages: 2025013
[    0.041106] Policy zone: Normal
[    0.041108] Kernel command line: BOOT_IMAGE=(hd0,gpt4)/bzImage root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id=f241819b9704463ea6ab754e75e99a73 fsck.repair=yes console=tty1 rauc.slot=B
[    0.041174] Unknown kernel command line parameters "BOOT_IMAGE=(hd0,gpt4)/bzImage", will be passed to user space.
[    0.041207] random: crng init done
[    0.042134] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.042624] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.042656] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.042666] software IO TLB: area num 4.
[    0.079565] Memory: 7949156K/8229276K available (16394K kernel code, 2646K rwdata, 5008K rodata, 1572K init, 3044K bss, 279860K reserved, 0K cma-reserved)
[    0.079610] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.079673] Dynamic Preempt: full
[    0.079699] rcu: Preemptible hierarchical RCU implementation.
[    0.079700] rcu:     RCU event tracing is enabled.
[    0.079700] rcu:     RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=4.
[    0.079702]  Trampoline variant of Tasks RCU enabled.
[    0.079703]  Tracing variant of Tasks RCU enabled.
[    0.079703] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
[    0.079704] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.080746] NR_IRQS: 4352, nr_irqs: 1024, preallocated irqs: 16
[    0.081004] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.081178] Console: colour dummy device 80x25
[    0.081427] printk: console [tty1] enabled
[    0.081442] ACPI: Core revision 20220331
[    0.081748] hpet: HPET dysfunctional in PC10. Force disabled.
[    0.081751] APIC: Switch to symmetric I/O mode setup
[    0.085781] clocksource: tsc-early: mask: 0xffffffffffffffff max_cycles: 0x3990bec8342, max_idle_ns: 881590769617 ns
[    0.085792] Calibrating delay loop (skipped), value calculated using timer frequency.. 3993.60 BogoMIPS (lpj=1996800)
[    0.085828] CPU0: Thermal monitoring enabled (TM1)
[    0.085831] x86/cpu: User Mode Instruction Prevention (UMIP) activated
[    0.085873] process: using mwait in idle threads
[    0.085876] Last level iTLB entries: 4KB 0, 2MB 0, 4MB 0
[    0.085878] Last level dTLB entries: 4KB 0, 2MB 0, 4MB 0, 1GB 0
[    0.085882] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
[    0.085886] Spectre V2 : Mitigation: Enhanced IBRS
[... 886 lines cut ...]
[  148.557853] eth0: renamed from veth8363c91
[  148.565259] IPv6: ADDRCONF(NETDEV_CHANGE): veth4eb7a5e: link becomes ready
[... cut ...]
[  179.859664] kauditd_printk_skb: 319 callbacks suppressed
[  179.859669] audit: type=1334 audit(1707258918.281:508): prog-id=133 op=LOAD
[  179.859832] audit: type=1334 audit(1707258918.281:509): prog-id=134 op=LOAD
[  179.859863] audit: type=1334 audit(1707258918.281:510): prog-id=135 op=LOAD
[  180.144125] audit: type=1334 audit(1707258918.565:511): prog-id=136 op=LOAD
[  180.144275] audit: type=1334 audit(1707258918.566:512): prog-id=137 op=LOAD
[  180.144307] audit: type=1334 audit(1707258918.566:513): prog-id=138 op=LOAD

```

sairon, 2024-02-07:
According to the USB ID, the card is RTL8821CU. Until c1b936716dd875d5b59162819a032f8abb08118e we used a 3rd party driver, because it was not available in the Linux kernel, yet since the above commit it should be replaced by an upstream Linux driver. Unfortunately I do not have the adapter available to test why it doesn't properly load the 3rd party one, but hopefully the next OS release should fix it for you :crossed_fingers: 

I'm a bit confused though - you said the steps to reproduce were "update OS from 11.4 to 11.5", yet you also say it disappeared in previous version. So does it mean it didn't work in 11.4 either? Or only on some other device?

Veldkornet, 2024-02-09:
Found this in journalctl FYI, if it's onf any use:
```
Feb 04 23:01:26 hassio homeassistant[467]: 2024-02-05 00:01:26.300 WARNING (MainThread) [bluetooth_auto_recovery.recover] There are no BT controllers present in the system!
Feb 04 23:01:26 hassio homeassistant[467]: 2024-02-05 00:01:26.301 ERROR (MainThread) [bluetooth_auto_recovery.recover] hciNone [38:CA:73:0F:05:AF] seems not to exist (anymore), check BT interface mac address in your settings; Available adapters: {} 
Feb 04 23:01:27 hassio homeassistant[467]: 2024-02-05 00:01:27.806 ERROR (MainThread) [habluetooth.scanner] hci0 (38:CA:73:0F:05:AF): Failed to restart Bluetooth scanner: hci0 (38:CA:73:0F:05:AF): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
Feb 04 23:01:27 hassio homeassistant[467]: Traceback (most recent call last):
Feb 04 23:01:27 hassio homeassistant[467]:   File "src/habluetooth/scanner.py", line 237, in habluetooth.scanner.HaScanner._async_start
Feb 04 23:01:27 hassio homeassistant[467]:   File "src/habluetooth/scanner.py", line 238, in habluetooth.scanner.HaScanner._async_start
Feb 04 23:01:27 hassio homeassistant[467]:   File "/usr/local/lib/python3.11/site-packages/bleak/__init__.py", line 198, in start
Feb 04 23:01:27 hassio homeassistant[467]:     await self._backend.start()
Feb 04 23:01:27 hassio homeassistant[467]:   File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/scanner.py", line 185, in start
Feb 04 23:01:27 hassio homeassistant[467]:     self._stop = await manager.active_scan(
Feb 04 23:01:27 hassio homeassistant[467]:                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 23:01:27 hassio homeassistant[467]:   File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/manager.py", line 405, in active_scan
Feb 04 23:01:27 hassio homeassistant[467]:     assert_reply(reply)
Feb 04 23:01:27 hassio homeassistant[467]:   File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
Feb 04 23:01:27 hassio homeassistant[467]:     raise BleakDBusError(reply.error_name, reply.body)
Feb 04 23:01:27 hassio homeassistant[467]: bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
Feb 04 23:01:27 hassio homeassistant[467]: 
Feb 04 23:01:27 hassio homeassistant[467]: 
Feb 04 23:01:27 hassio homeassistant[467]: The above exception was the direct cause of the following exception:
Feb 04 23:01:27 hassio homeassistant[467]: 
Feb 04 23:01:27 hassio homeassistant[467]: Traceback (most recent call last):
Feb 04 23:01:27 hassio homeassistant[467]:   File "src/habluetooth/scanner.py", line 367, in habluetooth.scanner.HaScanner._async_restart_scanner
Feb 04 23:01:27 hassio homeassistant[467]:   File "src/habluetooth/scanner.py", line 320, in _async_start
```

sairon, 2024-02-19:
Now that 12.0.rc1 is out, could you try it there, please? You don't need to switch to beta channel for that, you can only update OS from the CLI using `ha os update --version 12.0.rc1`.

Veldkornet, 2024-02-19:
Yes, it has returned to life! 🎉

sairon, 2024-02-19:
Awesome, thanks for the quick reply! I'm closing this as resolved then.

closed 2024-02-19 by sairon