home-assistant/operating-system #3155: Fresh Installation over Raspberry Pi Imager doesn`t work because of "no supervisor internet connection"

preko69, 2024-02-07:
### Describe the issue you are experiencing

Initial situation:
- Raspberry Pi 4B - 8GB
- Verbatim Vi560 S3 (SATA3) SSD 1 TB connected via USB3
- Network/Internet access via Ethernet cable

Installed directly to SSD via Raspberry Pi Imager
- Home Assistant 11.5 for RPi4 Image
[Downloaded raw logs.txt](https://github.com/home-assistant/operating-system/files/14192409/Downloaded.raw.logs.txt)
[image]
[Messages out of the Error Screen.txt](https://github.com/home-assistant/operating-system/files/14192410/Messages.out.of.the.Error.Screen.txt)
[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Installed the Home Assistant Image via Raspberry Pi Imager directly on the SSD
2. Connected afterwards the Raspberry Pi 4 with Ethernet Cable to the Internet/Router
3. Started RPi4
4. Recieved the Error Screen: "Error installing Home Assistant" with Error-Logs showed

Cannot enter the System setting or anything out of the Error screen.

The Home Assistant Observer reports as follows:
Supervisor: Connected
Supported: Supported
Healthy: Healthy 

[Messages out of the Error Screen.txt](https://github.com/home-assistant/operating-system/files/14192364/Messages.out.of.the.Error.Screen.txt)
[image]
[image]
[Downloaded raw logs.txt](https://github.com/home-assistant/operating-system/files/14192365/Downloaded.raw.logs.txt)

### Anything in the Supervisor logs that might be useful for us?

```txt
Supervisor logs are not reachable. Ended in a white screen with "Not found".
```

### Anything in the Host logs that might be useful for us?

```txt
No access to Settings.
```

### System information

No access to System information. Ended in a white screen with "Not found".

### Additional information

No Firewall blocking activated.

MilanCiprian, 2024-02-14:
Same problem to me

HAOS 12.0.dev20240214
RaspberryPi 5 8Gb

1. fresh install
2. start raspberry without internet connection
3. internet line up
[image]

[logs.txt](https://github.com/home-assistant/operating-system/files/14289363/logs.txt)

preko69, 2024-02-16:
I threw this stuff in the trash and switched to **Homebridge**. Installed - running! How cleanly programmed software should work. I am now very satisfied and no longer frustrated at having lost a whole day for nothing like before with Home Assistant!!!

Can be closed from my side, don't need it anymore!

closed 2024-02-16 by preko69

agners, 2024-02-16:
@preko69 I am sorry that Home Assistant (or rather Supervisor) caused issue in your environment :cry: 

The Supervisor is really an infrastructure project, who's purpose is to mainly enable [Home Assistant Core](https://github.com/home-assistant/core) (and Add-ons). That is also where most contributions go to, and that also means code quality is much higher! But any contribution or help to make Supervisor is *very* welcome!

Now to your issue: The first thing I am noticing is that the time is off. This is typical for Raspberry Pi as there is no RTC. Home Assistant OS tries to synchronize time on startup using NTP. By default DHCP provided NTP servers are used, if that is not available, the system falls back to the Cloudflare NTP servers. But if both don't work, we give up after 90s and continue to boot. So one question here would be why did NTP not work in your environment? This could be a local setup or maybe a bug in Home Assistant OS (we use systemd-timesyncd).

Now this unsynced time causes issues with the connectivity check, similar to what https://github.com/home-assistant/supervisor/issues/4883 reported. In theory a "poor man" NTP via http should trigger in Supervisor, but that seems not the case. It is definitely something I'll investigate into.

In any case, Homebridge is certainly a good project too. As long as you dont' go to the dark side (as in non-open-source) :wink:  Enjoy home automation!