home-assistant/operating-system #3156: Bluetooth driver missing?

MartinKlecka, 2024-02-07:
### Describe the issue you are experiencing

I have installed HAOS on Acer Aspire E15 (E5-575-597P). But bluetooth adapter not working, propably some driver issue. It is eaither Atheros (more likely) or Intel BT adapter.
Is there some way to install driver for this adapter?

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.74-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Screenshot...
```

### Anything in the Host logs that might be useful for us?

```txt
Nope
```

sairon, 2024-02-08:
The `qca` in firmware patch suggests it is indeed an Atheros card, likely QCA9377, but I'd like to confirm first. Can you post the output of `lsusb`?

MartinKlecka, 2024-02-10:
Here is lsusb print:

[core-ssh ~]$ lsusb
Bus 001 Device 004: ID 04f2:b571
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 003: ID 04ca:3015
Bus 002 Device 001: ID 1d6b:0003
[core-ssh ~]$

closed 2024-05-18 by github-actions[bot]