home-assistant/operating-system #3160: Following update HAOS fails to start on [drm] TMDS error

owlen, 2024-02-08:
### Describe the issue you are experiencing

I updated to the latest HOS version two days ago and updated the core yesterday. Today, my HA instance failed without me changing anything in the configuration.
The standard URL I use returns 
```
500 Internal Server Error 
Server got itself in trouble

```
I run HAOS on a small dedicated minisforum PC.
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. got 500 HTTP response
2. restarted the server (unfortunately before connecting a monitor and checking if there is any error)
3. Server no longer responds to HTTP requests
4. connected a monitor and took a snapshot

### Anything in the Supervisor logs that might be useful for us?

```txt
I can't access the UI or machine.
```

### Anything in the Host logs that might be useful for us?

```txt
I can't access the UI or machine.
```

### System information

I can't access the UI or machine.

### Additional information

Since my SSL certificates are not set up properly, I can't access the machine with ssh.
Any idea on how to proceed?

owlen, 2024-02-08:
I restarted it a few times, and now the [drm] error message is no longer shown. I also have a responsive `ha >` prompt.

`ha >core logs` shows errors. I'm afraid there is no way for me to copy, so here is two screen grabs.
[image]
[image]

If any info can help, please guide me on what to run.

sairon, 2024-02-21:
The drm error was likely result of a glitch in the communication with the display, it's unrelated to your problems. The issue itself seems to be caused by FS corruption in the Core container data. In cases like this, `ha supervisor repair` could help, or `ha core rebuild` if the core container is corrupted. But if nothing of this helps, your only option may be to do a fresh install and restore from backup (which can be done using HA CLI if the system is not completely fubared). Anyway, this is not an OS issue.

closed 2024-02-21 by sairon

owlen, 2024-02-21:
Thanks a lot, @sairon! I'll try to report back, if only for future reference.
Out of curiosity, can you say what makes you feel it's an FS issue? I'm trying to learn how the OGs analyze a crash like this.

sairon, 2024-02-21:
The "bad marshal data" error usually indicates the .pyc files are corrupted. This does not (must not!) happen on itself, most common causes are unclean shutdown or similar effects, which scramble the data on the disk. Given there might be more files damaged this way, this is why I recommend rather radical approach with the attempt to get to a clean state.

owlen, 2024-02-21:
As a last resource, I've run `ha core update,` and after physically restarting the PC - The home Assistant is back alive!
THe only thing I notice is the zigbee2mqtt app could not find specified some CSS.  in the Console:
```
Could not find specified theme
(anonymous) @ index-a782d339.js:1058
web-client-content-script.js:2 Uncaught (in promise) Error: Orphaned iframed
    at Hn (web-client-content-script.js:2:283879)
    at zn (web-client-content-script.js:2:283964)
    at web-client-content-script.js:2:377750
    at web-client-content-script.js:2:378513
    at Object.stateSync (web-client-content-script.js:2:378519)
    at web-client-content-script.js:2:377670
    at d (web-client-content-script.js:2:189771)
    at Generator.<anonymous> (web-client-content-script.js:2:191108)
    at Generator.next (web-client-content-script.js:2:190134)
    at b (web-client-content-script.js:2:195761)
```

Since I'm still not sure everything ok, are there any other commands that I might want to run to check the system?

This is the supervisor log:
```
24-02-21 16:10:06 INFO (MainThread) [supervisor.store] Loading add-ons from store: 105 all - 0 new - 0 remove
24-02-21 16:10:06 WARNING (MainThread) [supervisor.addons.options] Option 'allow_anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
24-02-21 16:10:06 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:15:06 WARNING (MainThread) [supervisor.addons.options] Option 'allow_anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
24-02-21 16:15:06 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:20:06 WARNING (MainThread) [supervisor.addons.options] Option 'allow_anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
24-02-21 16:20:06 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:22:05 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:22:11 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:22:19 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:23:03 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-02-21 16:23:20 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:24:32 INFO (SyncWorker_2) [supervisor.docker.manager] Stopping addon_45df7312_zigbee2mqtt application
24-02-21 16:24:32 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_45df7312_zigbee2mqtt application
24-02-21 16:24:32 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:24:32 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:24:33 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on zigbee2mqtt/zigbee2mqtt-amd64 with version 1.35.3-1
24-02-21 16:24:33 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:24:33 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:25:06 WARNING (MainThread) [supervisor.addons.options] Option 'allow_anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
24-02-21 16:25:06 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:26:30 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:26:35 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:27:15 INFO (MainThread) [supervisor.backups.manager] Found 12 backup files
24-02-21 16:27:15 ERROR (MainThread) [supervisor.backups.backup] Can't read backup tarfile /data/backup/39ce82c6.tar: unexpected end of data
24-02-21 16:27:15 INFO (MainThread) [supervisor.backups.manager] Found 12 backup files
24-02-21 16:27:15 ERROR (MainThread) [supervisor.backups.backup] Can't read backup tarfile /data/backup/39ce82c6.tar: unexpected end of data
24-02-21 16:27:25 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
24-02-21 16:27:25 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
24-02-21 16:27:38 INFO (MainThread) [supervisor.backups.manager] Backup 6ba7b25d starting stage addon_repositories
24-02-21 16:27:38 INFO (MainThread) [supervisor.backups.manager] Backup 6ba7b25d starting stage docker_config
24-02-21 16:27:38 INFO (MainThread) [supervisor.backups.manager] Creating new full backup with slug 6ba7b25d
24-02-21 16:27:38 INFO (MainThread) [supervisor.backups.manager] Backup 6ba7b25d starting stage addons
24-02-21 16:27:38 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on a0d7b954_plex
24-02-21 16:27:41 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon a0d7b954_plex
24-02-21 16:27:41 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on a0d7b954_vscode
24-02-21 16:27:42 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon a0d7b954_vscode
24-02-21 16:27:42 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on cebe7a76_hassio_google_drive_backup
24-02-21 16:27:42 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon cebe7a76_hassio_google_drive_backup
[... cut ...]
24-02-21 16:32:02 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
24-02-21 16:35:06 WARNING (MainThread) [supervisor.addons.options] Option 'allow_anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
24-02-21 16:35:06 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mqtt_password
```

I don't see anything special in the host logs, attached:
[ha host logs.txt](https://github.com/home-assistant/operating-system/files/14360971/ha.host.logs.txt)

Thanks in advance!