home-assistant/operating-system #3163: mport OVA in vmware >7 --> Line 30: Unsupported hardware family 'vmx-10'.

andrewt81, 2024-02-09:
### Describe the issue you are experiencing

When trying to import OVA in vmware >7 i receive error

Line 30: Unsupported hardware family 'vmx-10'.

I can bypass the error manually editing with ovftool but should be >vmx-11 by default

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

When trying to import OVA in vmware >7 i receive error

Line 30: Unsupported hardware family 'vmx-10'.

I can bypass the error manually editing with ovftool but should be >vmx-11 by default

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

no

jens-maus, 2024-02-12:
I am sorry to say so, but is vmWare ESX at all a valid target virtualization platform for HomeAssistant anymore since Broadcom crushes vmWare stuff to the ground, thus no free ESXi, etc.? I actually do feel that in future using the ova images with vmWare products is about to end at some point, thus only forum community might help to keep up some compatibility...

jens-maus, 2024-02-13:
> I'm aware of the broadcom plans.. but until free esxi are available and .ova are released regularly i think is reasonable to give to the community an "effort free" tool

First of all, the "plans" have just gone reality, see:

https://kb.vmware.com/s/article/2107518

Thus, no free ESXi anymore. However, I do of course not suggest to drop the OVA platform altogether, since this is still a standard and the ova images can be used by other virtualization solutions like Synology VMM, etc. which are Qemu/KVM based. I am just saying that now after the End-of-Life of free ESXi the focus has clearly shifted now to free virtualization platforms like ProxmoxVE. Thus, IMHO the focus of the OVA support in HomeAssistantOS should also clearly focus on these free and IMHO also more powerful platforms and every still existing user of ESX should consider moving over better now than too late...

closed 2024-05-22 by github-actions[bot]