home-assistant/operating-system #3172: VCGENCMD error in release 2024.0

cjmjmartens, 2024-02-12:
### Describe the issue you are experiencing

I use the following command to display the current version of the Raspberry firmware (RPI4). Since the upgrade to HA 2024.2.0, I receive the following error:

Command failed (with return code 1): sh -c 'ts=$(/usr/bin/vcgencmd bootloader_version | sed -E -n -e "s/timestamp (.*)/\1/p") && date -u -d @$ts -I'

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Add the code below to configuration.yaml to display (see the details below). Home Assistant version should be older than 2024.2.0. The firmware version of the Raspberry Pi will be displayed on the Dashboard.
2. Upgrade to Home Assistant 2024.2.0. 
3. The error message above will be displayed in the system log and the sensor will report "unknown".

`- sensor:
    name: RPi Firmware
    command: |
      sh -c 'ts=$(/usr/bin/vcgencmd bootloader_version | sed -E -n -e "s/timestamp (.*)/\1/p") && date -u -d @$ts -I'
`

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: homeassistant.components.command_line.utils
Source: components/command_line/utils.py:54
Integration: Command Line (documentation, issues)
First occurred: 21:42:19 (22 occurrences)
Last logged: 21:52:49

Command failed (with return code 1): sh -c 'ts=$(/usr/bin/vcgencmd bootloader_version | sed -E -n -e "s/timestamp (.*)/\1/p") && date -u -d @$ts -I'
```

### Anything in the Host logs that might be useful for us?

```txt
Feb 12 20:30:40 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a30eaf19e35ab883373a823298e6b33493b0db682463f2ba87b7119bd50a821a-runc.btXEiA.mount: Deactivated successfully.
Feb 12 20:30:47 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-e4b66d5947997a57970684bd140cea64e9cb66c6dbecbfb9fac11544902e1e61-runc.kdSZob.mount: Deactivated successfully.
Feb 12 20:31:07 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2851917923f8d7fb9cb73cf4ef75e52f237758c154fa7a200f075e69fa3a2088-runc.s1uQil.mount: Deactivated successfully.
Feb 12 20:31:17 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-e4b66d5947997a57970684bd140cea64e9cb66c6dbecbfb9fac11544902e1e61-runc.Yue6D6.mount: Deactivated successfully.
Feb 12 20:31:30 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2c232e963b665e5758521346e7293eb2e9ab3dccb9983bef9a711a34485dafc1-runc.3cz0Xs.mount: Deactivated successfully.
Feb 12 20:31:33 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-27a374791b538840b5f9711ff60892e80f6fdd4ff084403b4894d02c5a76994d-runc.3uIt8O.mount: Deactivated successfully.
Feb 12 20:31:41 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a30eaf19e35ab883373a823298e6b33493b0db682463f2ba87b7119bd50a821a-runc.0on21R.mount: Deactivated successfully.
Feb 12 20:32:03 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-27a374791b538840b5f9711ff60892e80f6fdd4ff084403b4894d02c5a76994d-runc.m0EnAz.mount: Deactivated successfully.
Feb 12 20:32:11 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a30eaf19e35ab883373a823298e6b33493b0db682463f2ba87b7119bd50a821a-runc.lWOB4V.mount: Deactivated successfully.
Feb 12 20:32:15 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-9501801975cab417412b1c433d71f156a958e6de24cd8cd4b7ec691e256397bb-runc.jfpppq.mount: Deactivated successfully.
Feb 12 20:32:38 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2851917923f8d7fb9cb73cf4ef75e52f237758c154fa7a200f075e69fa3a2088-runc.5jkzyK.mount: Deactivated successfully.
Feb 12 20:32:41 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a30eaf19e35ab883373a823298e6b33493b0db682463f2ba87b7119bd50a821a-runc.qb04JU.mount: Deactivated successfully.
Feb 12 20:33:31 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2c232e963b665e5758521346e7293eb2e9ab3dccb9983bef9a711a34485dafc1-runc.n1o0ZM.mount: Deactivated successfully.
Feb 12 20:33:38 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2851917923f8d7fb9cb73cf4ef75e52f237758c154fa7a200f075e69fa3a2088-runc.Q8rYsC.mount: Deactivated successfully.
Feb 12 20:33:42 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a30eaf19e35ab883373a823298e6b33493b0db682463f2ba87b7119bd50a821a-runc.vlah6h.mount: Deactivated successfully.
Feb 12 20:34:31 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-9501801975cab417412b1c433d71f156a958e6de24cd8cd4b7ec691e256397bb-runc.4IH3Ur.mount: Deactivated successfully.
Feb 12 20:34:31 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2c232e963b665e5758521346e7293eb2e9ab3dccb9983bef9a711a34485dafc1-runc.fFrPyo.mount: Deactivated successfully.
Feb 12 20:34:48 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-e4b66d5947997a57970684bd140cea64e9cb66c6dbecbfb9fac11544902e1e61-runc.Hot7fW.mount: Deactivated successfully.
Feb 12 20:35:05 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-9501801975cab417412b1c433d71f156a958e6de24cd8cd4b7ec691e256397bb-runc.BIA8Zs.mount: Deactivated successfully.
Feb 12 20:36:02 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2c232e963b665e5758521346e7293eb2e9ab3dccb9983bef9a711a34485dafc1-runc.m23EsI.mount: Deactivated successfully.
Feb 12 20:36:04 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-27a374791b538840b5f9711ff60892e80f6fdd4ff084403b4894d02c5a76994d-runc.5JPJYf.mount: Deactivated successfully.
Feb 12 20:36:09 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2851917923f8d7fb9cb73cf4ef75e52f237758c154fa7a200f075e69fa3a2088-runc.EEjPch.mount: Deactivated successfully.
Feb 12 20:36:34 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-27a374791b538840b5f9711ff60892e80f6fdd4ff084403b4894d02c5a76994d-runc.B3fOXs.mount: Deactivated successfully.
Feb 12 20:37:02 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-2c232e963b665e5758521346e7293eb2e9ab3dccb9983bef9a711a34485dafc1-runc.xOcJq4.mount: Deactivated successfully.
Feb 12 20:37:04 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-27a374791b538840b5f9711ff60892e80f6fdd4ff084403b4894d02c5a76994d-runc.BpRxGS.mount: Deactivated successfully.
[... 155 lines cut ...]

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 2 februari 2024 om 18:12
-- | --
current_recorder_run | 12 februari 2024 om 21:41
estimated_db_size | 498.83 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

In June '23 I reported the same error. Stefan Agners replied:

"Note that when you use a command sensor, this command is executed in the context of Home Assistant Core, which is not the same Home Assistant OS. So this is not a OS related issue.
That said, it seems that vcgencmd moved from /opt/vc/bin/vcgencmd to /usr/bin/vcgencmd, presumably because Alpine Linux package raspberrypi-userland changed the path of that program between v3.17 and v3.18. So you'll have to adjust the path of your command."

Based on the current issue, I have reported a bug related to command_line. Gjohansson-ST posted the following reaction and closed the ticket. That is the reason that I report it again in the HA OS section. 

"HA OS does not include vcgencmd (or at least not on my own) so it's not an issue with command_line really. You would have to make an issue on HAOS repository as the issue is really not here.
Thanks for understanding 👍"

djjoakim, 2024-02-12:
Same issue here, can’t get the temperature from my CPU as I used to do with vcgencmd

closed 2024-02-13 by github-actions[bot]

agners, 2024-02-13:
I am sorry, this is indeed a Core issue. I've reopened https://github.com/home-assistant/core/issues/109913.