home-assistant/operating-system #3180: Bluetooth stopped working after upgrading HA Yellow from 11.4 to 11.5

brunokc, 2024-02-17:
### Describe the issue you are experiencing

I've been trying to upgraded my HA Yellow from HAOS 11.4 to 11.5 but it used to just come back to 11.4 after reboot. Today I tried again and it did upgrade. However, upon reboot, Bluetooth was non-functional (Bluetooth was working fine on 11.4):

[image]

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade from 11.4 to 11.5
2. Observe Bluetooth is not working

### Anything in the Supervisor logs that might be useful for us?

```txt
24-02-16 17:24:50 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 11.5 / BootSlot A
24-02-16 17:24:50 INFO (SyncWorker_1) [supervisor.docker.manager] Starting homeassistant
24-02-16 17:24:50 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
24-02-16 17:24:51 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
24-02-16 17:24:51 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
24-02-16 17:24:51 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/229fc559 repository
24-02-16 17:24:51 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
24-02-16 17:24:53 INFO (MainThread) [supervisor.store] Loading add-ons from store: 78 all - 78 new - 0 remove
24-02-16 17:24:53 INFO (MainThread) [supervisor.addons.manager] Found 5 installed add-ons
24-02-16 17:24:53 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/zwave-js-ui/aarch64 with version 3.3.2
24-02-16 17:24:53 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-silabs-multiprotocol with version 2.4.4
24-02-16 17:24:53 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/z-wave-me/aarch64-ha-z-wave-me-addon with version 4.1.0
24-02-16 17:24:53 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/aarch64 with version 5.15.0
24-02-16 17:24:53 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.9.0
24-02-16 17:24:53 INFO (MainThread) [supervisor.backups.manager] Found 89 backup files
24-02-16 17:24:53 INFO (MainThread) [supervisor.discovery] Loaded 2 messages
24-02-16 17:24:53 INFO (MainThread) [supervisor.ingress] Loaded 2 ingress sessions
24-02-16 17:24:53 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup
24-02-16 17:24:53 INFO (MainThread) [supervisor.resolution.check] System checks complete
24-02-16 17:24:53 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
24-02-16 17:24:53 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-02-16 17:24:53 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup
24-02-16 17:24:53 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
24-02-16 17:24:53 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-02-16 17:24:53 INFO (MainThread) [__main__] Running Supervisor
24-02-16 17:24:53 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
24-02-16 17:24:53 INFO (MainThread) [supervisor.addons.manager] Phase 'initialize' starting 0 add-ons
24-02-16 17:24:53 INFO (MainThread) [supervisor.addons.manager] Phase 'system' starting 1 add-ons
24-02-16 17:24:53 INFO (SyncWorker_1) [supervisor.docker.manager] Cleaning addon_a0d7b954_zwavejs2mqtt application
24-02-16 17:24:54 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/zwave-js-ui/aarch64 with version 3.3.2
24-02-16 17:25:00 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state startup
24-02-16 17:25:00 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-02-16 17:25:00 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-02-16 17:25:00 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant Core state changed to NOT_RUNNING
24-02-16 17:25:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/z-wave-me/ha-z-wave-me-addon repository
24-02-16 17:25:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
24-02-16 17:25:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
24-02-16 17:25:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
24-02-16 17:25:03 INFO (MainThread) [supervisor.store] Loading add-ons from store: 78 all - 0 new - 0 remove
24-02-16 17:25:03 INFO (MainThread) [supervisor.store] Loading add-ons from store: 78 all - 0 new - 0 remove
24-02-16 17:25:24 INFO (MainThread) [supervisor.addons.manager] Phase 'services' starting 3 add-ons
24-02-16 17:25:24 INFO (SyncWorker_1) [supervisor.docker.manager] Cleaning addon_a0d7b954_vscode application
24-02-16 17:25:25 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/vscode/aarch64 with version 5.15.0
24-02-16 17:25:25 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_core_ssh application
[... 191 lines cut ...]
        CPU: 14ms

Feb 17 01:24:27 homeassistant systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Feb 17 01:24:27 homeassistant systemd[1]: hciuart.service: Skipped due to 'exec-condition'.
Feb 17 01:24:27 homeassistant systemd[1]: Condition check resulted in Configure Bluetooth Modems connected by UART being skipped.
Hint: Some lines were ellipsized, use -l to show in full.
```

Looking at the exec-condition, it check the result of `cat "/proc/device-tree/$(cat /proc/device-tree/aliases/bluetooth)/status"`, which is:

```
# cat "/proc/device-tree/$(cat /proc/device-tree/aliases/bluetooth)/status"
okay#
```

Don't we want `cat "/proc/device-tree/$(cat /proc/device-tree/aliases/bluetooth)/status"` to be **equal** to `okay` instead, though?

Regardless, manually starting btuart seems to get me nowhere. As far as I can tell, it's just exiting.

```
# /usr/bin/btuart
# ps aux | grep btuart
root       29265  0.0  0.0   3344   572 pts/0    S+   03:47   0:00 grep btuart
#
```

brunokc, 2024-02-17:
Just saw that btuart is a script and it seems to exit early by design on HA Yellow devices?

```
[...]
# Bail out if the kernel is managing the Bluetooth modem initialisation
if ( dmesg | grep -q -E "hci[0-9]+: BCM: chip" ); then
  # On-board bluetooth is already enabled
  exit 0
fi
[...]
```

And we have this:

```
# dmesg | grep -E "hci[0-9]+: BCM: chip"
[    6.645148] Bluetooth: hci0: BCM: chip id 107
#
```

Still, no controllers seem to have been detected:

```
# bluetoothctl list
#
```

sairon, 2024-02-21:
> Just saw that btuart is a script and it seems to exit early by design on HA Yellow devices?

It's designed to exit early on the whole RPi platform, because the HCI driver is not initialized by the userspace anymore, but by the kernel driver itself early in the boot. In that case, userspace initialization scripts must not be used anymore. You can revert to using them by adding `dtparam=krnbt=off` to `config.txt` in the boot partition, but it is not the default and should work without that. Also make sure there are no other overlays or modifications in your config, i.e. it's the same as [provided in a fresh install](https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/board/raspberrypi/yellow/config.txt).

However, you say it started to misbehave after upgrade from 11.4 to 11.5. There was no change that could have effect on this, so I think that something went wrong only during that particular initialization, what was that could be seen early in `dmesg`. Have you tried whether just simple restart (ideally a cold one, with unplugging the power for a while) doesn't fix that?

brunokc, 2024-02-22:
@sairon, great analysis! I first checked my config.txt and it was still identical to the fresh install one. Then I went ahead with a cold reboot and sure enough it came back to life. `dmesg` output looks different than when it was erroring out and going to Settings -> Devices shows a working Bluetooth device with live connections.

I'll try to remember this for next time I hit an issue like this during an upgrade.  

Thank you!

closed 2024-02-22 by brunokc

Bertk, 2025-03-08:
I had the identical issue after upgrade to 2025.3.1 and unplugging the power from home assistant yellow fixed it.