home-assistant/operating-system #3182: "A start job is running for HAOS swap" - takes forever

tweety88, 2024-02-17:
### Describe the issue you are experiencing

I am trying to freshly install HAOS on flash drive or SSD and run it on Raspberry Pi 5 8Gb.
I tried fresh install on 2 different storages - USB 3.0 Flash Drive and on SSD connected via cable to USB 3.0.
The image was created through 2 different ways - using the Raspberry Imager and the existing there HA option and via manually downloading the HA image and use it (tried with 11.5 and even 12.0). 
In all those cases, when I start the Pi 5, it stops on the step of "running for HAOS swap" and doesn't continue.

After many restarts and tries, it happens to skip the problematic step and to continue. Then, I managed to run and configure properly the HA. At that moment I tried to restart it several times, and it was properly working during every try.
After having the Raspberry off for few days and I turned it on, the situation was the same - same message and doesn't continue.
I have no idea what I can do and what other information I can extract.

[image]

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Open Raspberry Imager
2. Select Raspberry 5
3. Select HA
4. Select SSD storage
5. Write the image to the SSD
6. Connect the SSD to the Raspberry and turn it on
...

### Anything in the Supervisor logs that might be useful for us?

```txt
I cannot start the OS
```

### Anything in the Host logs that might be useful for us?

```txt
I cannot start the OS
```

123marvin123, 2024-02-19:
Same problem here

tweety88, 2024-02-19:
it seems as normal. I tested with:
- ssd 256 GB - it stopped on swap step for around 10 minutes and then continued
- usb 32 Gb - it stopped on swap step for around 2 minutes and then continued.
I am closing the ticket for now as I do not experience the issue anymore.

closed 2024-02-19 by tweety88

humantex, 2024-03-12:
I'm also running into this same issue after explicitly following the install steps for x86 installations. I started by downloading both the latest '.xy' image file and Etcher last night, then today, I flashed a 64GB Verbatim ToughMax thumb drive (USB 3.0) with the image, and tried the setup on an HP ProDesk 400 G2 mini, with a 256GB SSD & 8GB ram.

21 minutes later and still showing the process

[image]

At 25+ minutes, it finally continued to load -then- "Waiting for the Home Assistant CLI to be ready...", and after several minutes...

"[WARN] Home Assistant CLI not starting! Jump into emergency console...", following with a root prompt. This is a first time install for me, and no idea how to proceed from here.

I do see HA is on the network and assigned an IP, but it presents an error screen when accessed.

[image]

Downloaded logs.txt:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[20:38:16] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[20:38:16] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
24-03-12 20:38:27 INFO (MainThread) [__main__] Initializing Supervisor setup
24-03-12 20:38:27 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
24-03-12 20:38:28 INFO (MainThread) [supervisor.bootstrap] Setting up coresys for machine: generic-x86-64
24-03-12 20:38:28 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2024.02.0
24-03-12 20:38:28 INFO (MainThread) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
24-03-12 20:38:28 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
24-03-12 20:38:28 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-03-12 20:38:28 INFO (MainThread) [__main__] Setting up Supervisor
24-03-12 20:38:29 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
24-03-12 20:38:29 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
24-03-12 20:38:29 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
24-03-12 20:38:29 INFO (MainThread) [supervisor.host.services] Updating service information
24-03-12 20:38:29 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
24-03-12 20:38:29 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
24-03-12 20:38:29 INFO (MainThread) [supervisor.host.network] Updating local network information
24-03-12 20:38:29 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
24-03-12 16:38:30 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
24-03-12 16:38:30 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
24-03-12 16:38:30 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-cli versions: [<AwesomeVersion CalVer '2023.11.0'>]
24-03-12 16:38:30 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2023.11.0
24-03-12 16:38:30 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
24-03-12 16:39:12 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2023.11.0 - 172.30.32.5
24-03-12 16:39:12 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-dns versions: [<AwesomeVersion CalVer '2023.06.2'>]
24-03-12 16:39:12 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2023.06.2
24-03-12 16:39:12 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
24-03-12 16:40:02 INFO (MainThread) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2023.06.2 - 172.30.32.3
[... cut ...]
24-03-12 16:45:28 INFO (MainThread) [supervisor.host.network] Updating local network information
24-03-12 16:45:28 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
24-03-12 16:45:30 INFO (MainThread) [supervisor.host.manager] Host information reload completed
24-03-12 16:49:00 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-03-12 16:49:00 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not healthy - supervisor
```

humantex, 2024-03-12:
An update...

Flashing with Etcher directly to the SSD worked, without any extra steps. Total time to get running was under 10 minutes, including the time to flash and verify the image, physically install the SSD, and reboot/configure/run HA.

ssbarnea, 2024-04-27:
I just flashed 12.5 using official rasberry imager on a mac, to a 64GB sd card, put it in rpi 5 and.... still waiting for swap to initialize.

AlixAbbasi, 2024-04-28:
Same here. I just flashed it to Pi 5 and stuck here.

bjornpijnacker, 2024-06-24:
I am also experiencing this. It continued after ~4 minutes and I'm experiencing exactly what @humantex describes with the "waiting for CLI".

OP @tweety88 would you consider re-opening this issue? Since multiple people have problems and there's clearly not a solution yet.

supershopping, 2024-06-26:
Having same issue. Home assistant CLI not starting. Jump into emergency console.

dantmobi, 2024-07-06:
Also having same issue too with new 12.4 install on RP4 first boot "A start job is running for HAOS"

IMakeTP, 2024-07-13:
Having the same issue installing with generic x86 on an old hp g3 mini

mattxcnm, 2024-07-19:
Having the same issue on an old Dell Inspiron 15 laptop.

Moozzuzz, 2024-07-20:
Similar issue on old x86-64 hardware: 33min & counting. I notice an error message on amdgpu_discovery_init (fatal error on gpu init, not sure if it has any link with the issue at hand).

Could it be linked with the size of the SSD? (1TB in my case)

mike6789, 2024-08-14:
Same problem here... 64gb USB drive as boot media... simply flashed via rPi imager and it will not get past HAOS swap step...

aleabo336, 2024-09-01:
same here, generic x86_64, i used it for casaOS and ubuntu server completely fine but with HAOS after flashing it with balenaetcher to the ssd it stops at "a start job is running for HAOS swap" for more than 20 minutes

lebinho, 2024-10-04:
Same here. I had Home Assistant working fine. Installed mqqt2zigbee and wanted to restart HA and now in console on proxmox having +20 min and still on "A start job is running for HAOS swap"