home-assistant/operating-system #3184: Home Assistant Crashes when ESPHome runs out of memory (used to just crash ESPHome)

roblamoreaux, 2024-02-19:
### Describe the issue you are experiencing

When I compile NS-Panel Blueprint on my Odroid N2+ (4GB) with too many other things running it used to restart the ESPHome addon. Now with ESPHome 2024.2.0beta (and 2023.12.2) and OS version 12.0.rc1, supervisor 2024.02.0, and Corte 2024.2.2, it causes Home Assistant to restart. 
Turning off other add-ons such as Music Assistant reduces the chance of crash since ESPHome can get more resources.

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

12.0.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Compile NS-Panel Blueprint esphome code with more than 80% memory free on Odroid N2+ which used to cause ESPHome to restart.
2. It gets most of the way through and then HA restarts and supervisor log is restarted leaving no evidence of pre-crash info
3. Note I already reduced process limit to 1 on ESPHome.
... I am not use if this is ESPHome, Core or OS, but it used to just kill ESPHome which I could deal with

### Anything in the Supervisor logs that might be useful for us?

```txt
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/ad4c456f/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/d557e82e/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/a6ab9a92/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/4ded7296/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/49cb62d8/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/13d28fa7/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/d7a74702/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/7c6163a7/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/ecafd656/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/833ebb4e/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/a68d9e51/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/19ec4a64/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/dbb5f422/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/8998f60f/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/68fada8f/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/610b6b9e/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/f8b4fa7a/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/c8ccbc97/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/80f54172/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/c5de3154/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/50e40a8b/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/0f7903f9/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/730f3cd2/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/e57555ea/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/c2f90a79/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/3958110a/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/504f88eb/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/6794c7bb/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/8ac1a916/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/c5b54025/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/65c7614d/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/fefd3657/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/087bdfb9/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/4bdfc79d/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/5f86353b/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/4c6fe0d5/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/c068652c/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/4e3eec7f/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/562fd439/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/b58b769b/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/690209a6/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/50441d45/info access from cebe7a76_hassio_google_drive_backup
24-02-19 09:28:08 INFO (MainThread) [supervisor.api.middleware.security] /backups/5620eb4e/info access from cebe7a76_hassio_google_drive_backup
[... 147 lines cut ...]
Feb 19 14:29:20 homeassistant kernel: audit: type=1334 audit(1708352960.887:586): prog-id=157 op=UNLOAD
Feb 19 14:29:25 homeassistant kernel: audit: type=1701 audit(1708352965.835:587): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=docker-default pid=9326 comm="genie-client" exe="/usr/local/bin/genie-client" sig=5 res=1
Feb 19 14:29:25 homeassistant kernel: audit: type=1334 audit(1708352965.851:588): prog-id=160 op=LOAD
Feb 19 14:29:25 homeassistant kernel: audit: type=1334 audit(1708352965.851:589): prog-id=161 op=LOAD
Feb 19 14:29:25 homeassistant kernel: audit: type=1334 audit(1708352965.851:590): prog-id=162 op=LOAD
Feb 19 14:29:25 homeassistant systemd[1]: Started Process Core Dump (PID 9329/UID 0).
Feb 19 14:29:26 homeassistant systemd-coredump[9330]: Process 9326 (genie-client) of user 0 dumped core.
Feb 19 14:29:26 homeassistant systemd[1]: systemd-coredump@11-9329-0.service: Deactivated successfully.
Feb 19 14:29:26 homeassistant kernel: audit: type=1334 audit(1708352966.435:591): prog-id=162 op=UNLOAD
[... cut ...]
Feb 19 14:29:27 homeassistant systemd-journal-gatewayd[9341]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.
```

### System information

None

### Additional information

As noted above I have had an issue where compiling NS-Panel Blueprint ESPHome Yaml code would crash ESPHome due to insufficent resources and stopping other add-ons (especially Music Assistant) would free up enough resources for the compile to work. Now instead of restarting ESPHome the whole system restarts. This behavior is much worse where restarting ESPHome was acceptable if not desirable.

XiaoXianNv-boot, 2024-02-19:
Looks like there's not enough RAM

XiaoXianNv-boot, 2024-02-20:
add swapfile

sairon, 2024-02-20:
@roblamoreaux By the "whole system" you mean the OS as whole or just main system components running in containers? I expect when it runs out of memory, the OOM killer starts killing things. It may eventually kill containerd, which would probably end up with all containers being restarted. Also I'd expect some messages to show in the host logs then, but maybe they're only in the host `dmesg`? :thinking: Can you check with HDMI and keyboard attached?

In any case, if you're running low on resources and the compilation is **very** demanding, there's not much we can do on OS side to make it pass. But I absolutely agree it should try to behave well and it should be obvious what went bad.

roblamoreaux, 2024-02-21:
I'll plug the monitor in tonight and try compiling again. I'm not sure how much restarts, just know that everything stops responding then the browser shows the core start up messages.

I agree well behaved is the addon getting restarted and the core HA tasks keep running. That is what I saw before this.

daviss57, 2024-05-27:
Same issue here.
Installation onto S3 BOX stops at the same point everytime and causes HA to restart.
[image]

[image]

[image]

cogentcoder, 2024-07-17:
My home assistant is also crashing while updating or compiling yaml for ESP32 Devkit. I can update and compile for ESP8266 without issue. I am using Raspberry Pi 3B with 1GB, so it should be RAM issue and I want to update but not sure to go with NUC system or Raspberry Pi. Till then, I have installed python and esphome on windows 11 and can compile esphome yaml under a minute and then update the program using esphome web.

edm00se, 2024-08-29:
I've also started seeing esphome in my HA add-on take down my entire HA instance on my Raspberry Pi 4 (8GB) which had been previously more than capable. It started for me in late July. It appears I it doesn't make it far into compiling before it takes down HA and I lose my connection.

TexTown, 2024-08-29:
Interesting, I thought this was memory related because my Raspberry Pi only has 2gb of memory. But if this happens on 8gb systems as well, it becomes even harder to explain.

I can still work around the issue by shutting down most addons (Grafana, smb, ssh, matter etc) and then performing the esp update.
Does that work for you?

edm00se, 2024-08-30:
@TexTown after stopping a couple of other add ons I was able to successfully compile and install for the update in esphome. I suspect your observation on the memory is accurate, in whatever way that HA does or doesn't manage memory for and between add ons.

dpgh947, 2024-09-04:
Install this to increase your swap space. It's an addon, but it only runs at startup to add the extra swap space, and then ends. https://github.com/TazzerMAN/increase_swap_addon

TexTown, 2024-09-06:
I've installed the addon, increased the swap size to 2GB and can now successfully update ESPHome devices without the need to stop other addons. This is great, thanks @dpgh947 for the tip.

dpgh947, 2024-09-06:
Brilliant, you're welcome. I also have a 2gb pi4 and I also had trouble with esphome compiles.

nielsaust, 2024-10-16:
> Install this to increase your swap space. It's an addon, but it only runs at startup to add the extra swap space, and then ends. https://github.com/TazzerMAN/increase_swap_addon

I want to thank you greatly for this tip. My HA install rebooted not only when compiling for ESPHome but also other instances and this seemed to have solved it. 

So... Thanks!

dpgh947, 2024-10-16:
Cheers, you're welcome

supertr0n, 2025-01-15:
I've just experienced this same issue and it's taken a while to figure out what was going wrong!

It's fine if ESPHome requires a certain amount of memory in order to run compilations but the unintuitive part is there is seemingly no warning that available memory is insufficient and the entire Home Assistant system reboots unexpectedly.

I think this experience could be improved by showing a warning on installation of the ESPHome add-on, if the host system memory appears to be too low eg. in the case of Home Assistant OS where it presumably could detect this.

Or in the absence of detecting if available memory is insufficient, perhaps just display a warning at the install stage of what the likely memory requirements are.

closed 2025-04-23 by github-actions[bot]

edm00se, 2025-04-23:
This issue has now been marked as stale and will be closed if no further activity occurs. Thank you for your contributions.

[image]