home-assistant/operating-system #3186: qemu: no ha-cli in headless mode

532910, 2024-02-20:
### Describe the issue you are experiencing

I run HAOS from haos_ova-11.5.qcow2
On ttyS0 it starts it prints kernel log and starts agetty.
On tty1 it prints boot log and starts ha-cli.

1. I'd like to run ha-cli on ttyS0 or even on both: ttyS0 and tty1
2. agetty login has no reason on ttyS0

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

## System Information

version | core-2024.2.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.1.74-haos
arch | x86_64
timezone | Etc/UTC
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1468
Downloaded Repositories | 11
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.5
-- | --
update_channel | stable
supervisor_version | supervisor-2024.01.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 30.8 GB
disk_used | 8.9 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Piper (1.4.0), Whisper (1.0.2), openWakeWord (1.8.2), ESPHome (2023.12.9), Terminal & SSH (9.9.0), Matter Server (5.1.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 6
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 9, 2024 at 12:31
-- | --
current_recorder_run | February 20, 2024 at 06:26
estimated_db_size | 214.18 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

sairon, 2024-02-22:
Currently getty is started on serial (ttyS0) via the standard systemd flow, while tty1 (graphical console) is handled by a special service `ha-cli@.service` with the default instance set to `tty1`. I don't think it's possible to set multiple default instances, moreover other platforms are implemented different, so it may not be applicable universally. May I ask what is the use case? Even on the serial console without ha-cli as the default shell, you can simply use `ha` commands.

532910, 2024-03-05:
OK, if is't impossible to set multiple default instances the single one will be enough.

1. getty on ttyS0 is completely useless: it's impossible to login there
2. ha-cli@.service on graphical console is completely useless: what is the reason to run qemu with videocard, while serial console is enough

instead of two completely useless things I'd like to have only one, but useful: ha-cli@.service on ttyS0. No ha-cli on tty0, no getty on ttyS0, only one ha-cli on ttyS0

532910, 2024-03-05:
May I ask what is the use case to run ha-cli on graphical console?