home-assistant/operating-system #3188: PCI Errors booting on NVME on Raspberry Pi 5

corvy, 2024-02-21:
### Describe the issue you are experiencing

I got NVME to work with the new HA build. I first installed on SD card, and when I got my HAT I used dd to copy all of my data to the NVME. After booting with Raspberry Pi OS I got to change the boot order and it worked just fine.

But now I had the following errors in dmesg:

```
[20757.013590] nvme 0000:01:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[20757.013594] nvme 0000:01:00.0: device [1d79:2263] error status/mask=00009100/0000e000
[20757.013598] nvme 0000:01:00.0: [ 8] Rollover
[20757.013601] nvme 0000:01:00.0: [12] Timeout
```

I have tried both PCI gen 2 and 3, same issue. Reseating the PCI cable makes no difference so I think the connection is good.

I have read a little and I find this in the Raspberry Pi Forums:
[https://forums.raspberrypi.com/viewtopic.php?t=363682#p2193646 1](https://forums.raspberrypi.com/viewtopic.php?t=363682#p2193646)

One of the symptoms of this problem is that it looses network connectivity. I get dropped ping echo replies from the device, and I loose SSH connections. Both on wired and wireless. The device does not crash, but it lags. 

The “fix” seems to be to recompile the bcm2712-rpi-5-b.dtb file. But I do not find dtc as a command on Hass OS so I am unable to do that. The dtb file is placed here:

```
find /mnt/boot/ | grep bcm2712
/mnt/boot/slot-A/bcm2712-rpi-5-b.dtb
/mnt/boot/slot-B/bcm2712-rpi-5-b.dtb
```

One relevant comment (it seems!): “The default MSIx target has an address above 4GB and that’s confusing the device - switching to the internal MSI target means the controller can put the target address below 4GB. I’m pretty sure the overlay should have a way of doing this automatically.”

It it possible to recompile the dtb file to test if the proposed fix from the Raspberry Pi forums could also work for Hass OS? 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.0.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Happens all the time by itself. 

### Anything in the Supervisor logs that might be useful for us?

```txt
24-02-21 09:46:44 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
24-02-21 09:46:44 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/raspberrypi5-64-homeassistant with version 2024.2.2
24-02-21 09:46:44 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 12.0.rc1 / BootSlot A
24-02-21 09:46:44 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/1f3d020e repository
24-02-21 09:46:44 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
24-02-21 09:46:44 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
24-02-21 09:46:44 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
24-02-21 09:46:44 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/2d5c2698 repository
24-02-21 09:46:44 INFO (MainThread) [supervisor.store] Loading add-ons from store: 85 all - 85 new - 0 remove
24-02-21 09:46:44 INFO (MainThread) [supervisor.addons.manager] Found 6 installed add-ons
24-02-21 09:46:44 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 6.4.0
24-02-21 09:46:44 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/nginxproxymanager/aarch64 with version 1.0.1
24-02-21 09:46:44 INFO (MainThread) [supervisor.docker.interface] Attaching to sbarmen/aarch64-addon-gpsd2mqtt with version 2024.2.1
24-02-21 09:46:44 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/aarch64 with version 5.15.0
24-02-21 09:46:44 INFO (MainThread) [supervisor.docker.interface] Attaching to local/aarch64-addon-gpsd2mqtt with version 2024.2.0.dev
24-02-21 09:46:44 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/aarch64 with version 17.1.0
24-02-21 09:46:44 INFO (MainThread) [supervisor.backups.manager] Found 5 backup files
24-02-21 09:46:44 INFO (MainThread) [supervisor.discovery] Loaded 1 messages
24-02-21 09:46:44 INFO (MainThread) [supervisor.ingress] Loaded 1 ingress sessions
[... 478 lines cut ...]

<details><summary>Recorder</summary>

oldest_recorder_run | February 17, 2024 at 7:38 AM
-- | --
current_recorder_run | February 21, 2024 at 9:47 AM
estimated_db_size | 62.75 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

config.txt (the first part for a GPS HAT using UART. Should not be relevant. 

[all]
enable_uart=1
dtparam=uart0_console

# Enable external PCIe port
dtparam=pciex1

# Optionally, you can control the PCIe lane speed using this parameter
# dtparam=pciex1_gen=3

corvy, 2024-02-21:
Further to this I tried to follow the recompile suggested in the link above (https://forums.raspberrypi.com/viewtopic.php?t=363682#p2193646). I copied out the dtb files from /mnt/boot/slot-A and slot-B, made the changes and recompiled. Moved them back to the folders and rebooted. But no change. 

To recompile I installed dtc in the Home Assistant container (apk add dtc) and copied the files from /mnt/boot via ssh to 22222 and placed them in the /share folder (/mnt/data/supervisor/share/ in host OS). From there I did the uncompress, change and recompress of the dtb files and moved them back. Are these two the correct files?

vin100ross, 2024-02-21:
Not sure if it's related. I get some of these once in a while, also on a Pi 5 but still on 11.5:

> [103280.003182] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00                                                                       .0
[103280.003197] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=                                                                       Data Link Layer, (Receiver ID)
[103280.003202] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00                                                                       000040/00002000
[103280.003207] pcieport 0000:00:00.0:    [ 6] BadTLP
[103883.906442] pcieport 0000:00:00.0: AER: Multiple Corrected error received: 0                                                                       000:00:00.0
[103883.906465] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=                                                                       Data Link Layer, (Transmitter ID)
[103883.906470] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00                                                                       0010c0/00002000
[103883.906474] pcieport 0000:00:00.0:    [ 6] BadTLP
[103883.906478] pcieport 0000:00:00.0:    [ 7] BadDLLP
[103883.906482] pcieport 0000:00:00.0:    [12] Timeout
[103883.906489] nvme 0000:01:00.0: PCIe Bus Error: severity=Corrected, type=Phys                                                                       ical Layer, (Transmitter ID)
[103883.906493] nvme 0000:01:00.0:   device [144d:a809] error status/mask=000011                                                                       01/0000e000
[103883.906497] nvme 0000:01:00.0:    [ 0] RxErr
[103883.906500] nvme 0000:01:00.0:    [ 8] Rollover
[103883.906504] nvme 0000:01:00.0:    [12] Timeout

This is with gen 2 speed. I tried gen 3 but lost the connection with the NVMe entirely after a few days so reverted back. These messages don't exactly flood my logs, but happen every few hours maybe. Not sure if it is software or hardware related. I am using a geekworm X1003 HAT with a Samsung PM991 SSD. I might try to change the FPC cable eventually to see if it has any impact if this becomes a problem.

sairon, 2024-02-22:
@corvy You don't say what NVMe module  you are using, but I think the thread you refer to is unrelated, as it's discussing issues with one particular SATA controller. I can't recommend using the same method of modifying the DTB, as the one used in HAOS may have different numbering, and changing `msi-parent` blindly to `0x66` may lead to unexpected results. Moreover, the AER errors may look similar but if you look closely, the message is different from yours.

The communication with your drive is having timeouts that can be caused by various factors - it can be a hardware issue (bad contacts, damaged cables) or general hardware incompatibility. One of the Yellow users recently even encountered a [similar problem](https://community.home-assistant.io/t/home-assistant-yellow-pcie-bus-errors-reboots/685736), seemingly caused by some RF interference in his setup. From the other possible issues which come up to my mind, it can also be a power issue, so replacing the power supply might be worth a shot, or problem with power management - in that case you can try adding `pcie_aspm=off` to `cmdline.txt`. In any case, it will be worth checking if the same drive works with RPi OS in a longer term too, and if it's so, then open discussion on RPi forums or issue tracker.

vin100ross, 2024-02-22:
Disabling Bluetooth reduced my errors 90%, so interference is certainly possible. Geekworm NVMe HAT owners take note... I only got one error during the entire night saying my nvme controller had become unresponsive. So power management also seems to be an issue. I'll try to turn it off in cmdline.txt as suggested.

In any case, I am now fairly certain that this has nothing to do with Homme Assistant and looks to be hardware-related.

corvy, 2024-02-22:
Problem solved for me with pcie_aspm=off to cmdline.txt. Thank you!

kitopopo, 2024-02-28:
> Problem solved for me with pcie_aspm=off to cmdline.txt. Thank you! 

Dear friend, thanks for solve the problem. I had had this problem for months, i add by mistake the pcie_aspm=off to \boot\firmware\config.txt and also the problem has been solved. Maybe is possible add in this two files or only in cmdline.txt? Thanks in advanced. Best regards

kitopopo, 2024-03-02:
Dear friends, adding pcie_aspm=off in cmdline.txt does not solve the problem but rather reduces the number of pci error corrected errors on my raspberry pi 5. If you perform several reboots and/or total power shutdowns, looking at the raspberry log at each boot You will be able to see the error mentioned. I have a pineberry pi button and lexar nm620 nvme.  Can you reopen the issue so that the problem can be resolved? thanks in advance

kitopopo, 2024-03-04:
Have you tried turning the raspberry off and on repeatedly through the plug and also doing reboots? the pcie error appears again. the pcie_aspm=off don't solve the problem. any ideas? thanks in advanced

agners, 2024-03-04:
@kitopopo if this is still reproducible for your configuration, feel free to open a new issue directly.

Also, it would generally be interesting if you can reproduce this also with Raspberry Pi OS. We use the very same kernel as Raspberry Pi OS, so if it is reproducible there, I'd suggest to report it with the Raspberry Pi folks (as they are maintaining that kernel tree and are the hardware manufacturer).

kitopopo, 2024-03-04:
> @kitopopo if this is still reproducible for your configuration, feel free to open a new issue directly.
>
> Also, it would generally be interesting if you can reproduce this also with Raspberry Pi OS. [...]

Dear friend, 
I am using rasperri pi OS in a raspberry pi 5. It is the same issue.

kitopopo, 2024-03-10:
Dear friends,

Here they explain the procedure to eliminate pci:corrected errors, in my case it works, anyone who has errors can try it.

[https://github.com/geerlingguy/raspberr ... issues/559](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/559)

kind regards