home-assistant/operating-system #3195: Error message on console after upgrade to 12.0

derolli1976, 2024-02-26:
### Describe the issue you are experiencing

[image]

Running on Hyper-V VM.
Looks like everything else is working fine.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update to 12.0 from previous release

### Anything in the Supervisor logs that might be useful for us?

```txt
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
24-02-26 16:26:27 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
24-02-26 16:26:27 INFO (MainThread) [supervisor.host.services] Updating service information
24-02-26 16:26:27 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
24-02-26 16:26:27 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
24-02-26 16:26:27 INFO (MainThread) [supervisor.host.network] Updating local network information
24-02-26 16:26:27 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
24-02-26 16:26:27 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
24-02-26 16:26:27 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/beta.json
24-02-26 16:26:27 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2024.02.0
24-02-26 16:26:27 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
24-02-26 16:26:27 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2024.02.0 - 172.30.32.5
24-02-26 16:26:27 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2023.06.2
24-02-26 16:26:27 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2023.06.2 - 172.30.32.3
24-02-26 16:26:28 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0
24-02-26 16:26:28 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0 - 172.30.32.4
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2023.06.0
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-multicast with version 2023.06.2
24-02-26 16:26:28 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2023.06.2 - Host
24-02-26 16:26:28 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/qemux86-64-homeassistant with version 2024.2.3
24-02-26 16:26:28 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 12.0 / BootSlot A
24-02-26 16:26:28 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
24-02-26 16:26:28 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
24-02-26 16:26:28 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
24-02-26 16:26:28 INFO (MainThread) [supervisor.store] Loading add-ons from store: 77 all - 77 new - 0 remove
24-02-26 16:26:28 INFO (MainThread) [supervisor.addons.manager] Found 4 installed add-ons
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/amd64 with version 17.1.1
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/log-viewer/amd64 with version 0.17.0
24-02-26 16:26:28 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/node-red/amd64 with version 17.0.7
24-02-26 16:26:29 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.15.0
24-02-26 16:26:29 INFO (MainThread) [supervisor.backups.manager] Found 1 backup files
24-02-26 16:26:29 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
[... 231 lines cut ...]
views | 1
mode | storage

</details>

<details><summary>MercedesME 2020</summary>

api_endpoint_reachable | ok
-- | --
websocket_connection_state | connected
cars_connected | 1
version | v0.11.1

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 14, 2024 at 7:10 AM
-- | --
current_recorder_run | February 26, 2024 at 4:27 PM
estimated_db_size | 9.50 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

DanDon, 2024-02-26:
Same issue and my entire HA OS instance crashes out and becomes unresponsive. Full Host reboot only runs stable for a minute or two and then crashes out with:

[image]

maxakash, 2024-02-26:
Same here, my Home Assistant is not coming online after updating

agners, 2024-02-26:
@DanDon @maxakash @ang3lx you all get `panic: "ro" shorthand is more than one ASCIII character` on the console?

Are you using Hyper-V VM as well?

ang3lx, 2024-02-26:
@agners I'm out from home at this moment, so a can't see console
I'm running on odroid c4

DanDon, 2024-02-26:
> @DanDon @maxakash @ang3lx you all get `panic: "ro" shorthand is more than one ASCIII character` on the console?
> 
> Are you using Hyper-V VM as well?

I'm not using Hyper-V ... I'm running the HA OS in a VM on VMware Workstation

agners, 2024-02-26:
Oh I see, this is on the beta channel. This is actually not related to the OS release itself, but to the current CLI instance. The OS lead to a reboot, which started the CLI again, and ultimately printed the error.

Underlying issue is already resolved, see https://github.com/home-assistant/cli/issues/460.

I'll create an updated HA CLI plug-in release, the issue should be resolved soon.

ang3lx, 2024-02-26:
sorry, it's the first time something like this has happened to me... so I'm in trouble.
Is the only way to boot properly is to format or restore from backup? or is there any operation to solve the problem

epoullet, 2024-02-26:
Same for me on Raspberry. What can be done ? Only restore from backup ?

rawzone, 2024-02-26:
Not sure if related but 12.0 also broke my VM (VM running under Proxmox). Just boots to EFI shell.
Had to restore full VM backup to get it going again.

agners, 2024-02-26:
> Not sure if related but 12.0 also broke my VM (VM running under Proxmox). Just boots to EFI shell.
> Had to restore full VM backup to get it going again.

This is not really related.

That said, we have been reports of such issues already before, e.g. #2136. Unfortunately I was never able to reproduce this on my Proxmox setup, and we double checked all possible race conditions and make sure things are written to disk before the reboot. We even mount the boot partition "sync" mode, to make sure things hit the disk. Yet on every release (and mainly on virtual machines) we have such reports. If this is reproducible, or you have any logs/traces (maybe from the Hypervisor) which could be helpful, please open a new issue with the details.

agners, 2024-02-26:
> Same for me on Raspberry. What can be done ? Only restore from backup ?

Your Home Assistant installation should remain working. You'll get the new CLI update 2024.02.1 within the next few hours. From then on the CLI will work normally again as well.

derolli1976, 2024-02-27:
> > Same for me on Raspberry. What can be done ? Only restore from backup ?
> 
> Your Home Assistant installation should remain working. You'll get the new CLI update 2024.02.1 within the next few hours. From then on the CLI will work normally again as well.

New version fixes the issue for me.
CLI is working again. At least for me, HA webclient has been available all the time.

agners, 2024-02-27:
The main issue here, `panic: "ro" shorthand is more than one ASCIII character` on the system console, is resolved by the latest CLI update, as confirmed by @derolli1976 . I am closing the issue.

@nicolazilio @atus your issues are not related to this issue really. Please open a new issue along with all required information.