home-assistant/operating-system #3196: Unable to boot from SD or SSD

cyclobob, 2024-02-26:
### Describe the issue you are experiencing

I am running a 4Gb Pi 5 with a 500Gb SSD connected via nvme. I can run the standard Pi 64bit OS from either the SD or SSD. I have updated the EEPROM and tried several different SD cards and I have run the standard OS on SD and used Pi Imager to build the SSD. The system is set to boot from SD first then SSD so I can boot the standard OS on the SD and create an image on the SSD using pi imager. I have HA already running on a Pi 3 but wanted to increase performance with the Pi5. The PI5 starts to boot HAOS but then stops with the yellow Ethernet light on solid and the green system led on. I have tried leaving it for several hours but nothing works. Have ordered an HDMI cable so I can monitor what's going on. I have used VNC to access the graphical interface using the standard Pi OS.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Build SD card for Pi5 using Pi Imager on Mac (or using Pi 5 to build image on SSD)
2. Boot Pi5 using HAOS image on SD or SSD
3. System hangs

### Anything in the Supervisor logs that might be useful for us?

```txt
The system won't boot so no logs available
```

### Anything in the Host logs that might be useful for us?

```txt
The system won't boot so no logs available
```

agners, 2024-02-27:
> The PI5 starts to boot HAOS but then stops with the yellow Ethernet light on solid and the green system led on.

My Raspberry Pi 5 shows solid green and yellow on the Ethernet port (no activity blinking even when I access my instance) as well as solid green. At first sight, this seems "regular behavior" I'd say.

Do you see an IP address acquired on your router or similar? Is `homeassistant.local:8123` as well as `homeassistant:8123` not reachable?

In any case, the HDMI screen should really show what is going on.

cyclobob, 2024-02-27:
Thanks for the reply Agners. So far no, there hasn't been a DHCP request. Will start from scratch again today....

cyclobob, 2024-02-27:
Fixed it! I found some errors with the SSD and had to remove it from the pi, connect to my Mac and initialise it. Then used Pi imager to flash the SSD and put it back in the Pi. As for the SD cards, I think I must have some bad ones.

sairon, 2024-02-28:
@cyclobob Thanks for reporting back! My two cents is that I would not put my trust into reliability of such SSD anymore, but glad you got it sorted out. If you got your HDMI cable now, any debugging should be a bit easier the next time.

closed 2024-02-28 by sairon