home-assistant/operating-system #3200: WiFi card not detected - 04W3772 Intel Wireless-N 105 WiFi

bjornpri, 2024-02-26:
### Describe the issue you are experiencing

Possibly missing driver.
105bnhmw wifi card for lenovo thinkcenter e73 e93 m62z m72e m73 m82 m83 m92 m93 EDGE-72Z series, from 04w3772

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.16-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

New install (27/2-2024)

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

## System Information

version | core-2024.2.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.6.16-haos
arch | x86_64
timezone | Europe/Oslo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1410
Downloaded Repositories | 28

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 28. mars 2024 kl. 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | false
remote_connected | false
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-17.ui.nabu.casa
certificate_status | ready
instance_id | b474bb9405544d658027fdb182065779
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Easee elbillader</summary>

component_version | 0.9.57
-- | --
reach_easee_cloud | ok
connected2stream | true

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 116.7 GB
disk_used | 32.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Glances (0.21.0), Log Viewer (0.17.0), Terminal & SSH (9.9.0), ESPHome (2024.2.1), OneDrive Backup (2.3), Z-Wave JS UI (3.4.0), Samba share (12.3.0), Node-RED (17.0.7), Mosquitto broker (6.4.0), Frigate (Full Access) (0.13.2), File editor (5.8.0), Advanced SSH & Web Terminal (17.1.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 16
views | 24
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 17. februar 2024 kl. 09:41
-- | --
current_recorder_run | 26. februar 2024 kl. 23:15
estimated_db_size | 3004.29 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

darkxst, 2024-02-27:
Intel 100 series ucode seems to be missing from Buildroot linux-firmware package.

sairon, 2024-02-27:
My personal opinion is that this card is kind of obsolete in these days, with single-band 802.11b/g/n, but we can add it if there's demand for it. I'll prepare a patch and enable it for the x86 platform.

PR #3219 (2024-03-04): Add Intel Wireless-N 105 firmware for generic-x86-64
* buildroot 0ab96d7c0d...15dc3b6e0c (1):
  > linux-firmware: add options for missing Intel b/g/n cards

Fixes #3200
commit 47f86719 (2024-03-04): Add Intel Wireless-N 105 firmware for generic-x86-64 (#3219)

* buildroot 0ab96d7c0d...15dc3b6e0c (1):
  > linux-firmware: add options for missing Intel b/g/n cards

Fixes #3200
closed 2024-03-04 by sairon