home-assistant/operating-system #3203: Wifi and Lan somehow required for core update in 12.0

Z3R0-CDS, 2024-02-27:
### Describe the issue you are experiencing

I updated to 12.0 today. Afterwards I wanted to update core. Wich did not work because it said no internet connection.
I connected WIFI (I dont use it I just use LAN/ETH) and I can update.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.1.73-haos-raspi

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update to OS 12.0
2. Try to update to core 2024.2.4
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nope
```

### Anything in the Host logs that might be useful for us?

```txt
24-02-27 14:07:16 WARNING (MainThread) [supervisor.jobs] 'HomeAssistantCore.update' blocked from execution, no host internet connection
```

### Additional information

Failed to call service update/install. Error updating Home Assistant Core 'HomeAssistantCore.update' blocked from execution, no host internet connection is the original error I got.
I used ping 8.8.8.8 to verify it works even tho it should anyway because I could use it before and refresh my duckdns ip

Z3R0-CDS, 2024-02-28:
Friend of mine tryed it too. Somehow it works without wifi connected on his side. Not sure if my setup just had some random bug

sairon, 2024-02-28:
The connectivity check is performed every 10 minutes, maybe you had some temporary outage on the network or hardware (i.e. cable was disconnected during boot) level. Anyway it seems this have been a one-time thing, so there is not much more to do.  Rather than ping, try doing `curl -v http://checkonline.home-assistant.io/online.txt` the next time. If that fails, then there's something indeed wrong with the connectivity. If you are able to reproduce the issue reliably, please let me know and I'll reopen the issue.

closed 2024-02-28 by sairon