home-assistant/operating-system #3207: Bluetooth Integration Fails to Load After Upgrade to 12.0

dai80027, 2024-02-27:
### Describe the issue you are experiencing

Bluetooth integration stays in "Failed setup - will retry" after upgrade

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to 12.0
2. Restart HA
3. Bluetooth integration not working
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-02-26 21:26:55.782 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not determine the power state of the Bluetooth adapter hci0 [00:00:00:00:00:00] due to timeout after 5 seconds
2024-02-26 21:27:16.644 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not cycle the Bluetooth adapter hci0 [00:00:00:00:00:00]: [Errno 110] Operation timed out
2024-02-26 21:27:16.646 WARNING (MainThread) [bluetooth_auto_recovery.recover] Bluetooth management socket connection lost: [Errno 22] Invalid argument
2024-02-26 21:27:21.646 WARNING (MainThread) [bluetooth_auto_recovery.recover] Bluetooth adapter hci0 [00:00:00:00:00:00] could not be reset: 
2024-02-26 21:27:21.650 WARNING (MainThread) [bluetooth_auto_recovery.recover] Closing Bluetooth adapter hci0 [00:00:00:00:00:00] failed: [Errno 9] Bad file descriptor
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing that isn't shown above
```

### System information

## System Information

version | core-2024.2.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | America/Denver
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1406
Downloaded Repositories | 11
HACS Data | ok

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 21

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | March 13, 2024 at 6:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-7.ui.nabu.casa
certificate_status | ready
instance_id | f378d4cd7122419dabf44e9282f8ec67
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 28.0 GB
disk_used | 13.1 GB
healthy | true
supported | true
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.9.0), RTSP Simple Server Add-on (v0.17.6), File editor (5.8.0), Samba share (12.3.0), Home Assistant Google Drive Backup (0.112.1), Piper (1.5.0), Mosquitto broker (6.4.0), Node-RED (17.0.7), eWeLink Smart Home (1.4.3), eufy-security-ws (1.7.1), ESPHome (2024.2.1), Spotify Connect (0.13.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 2
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 21, 2024 at 10:45 PM
-- | --
current_recorder_run | February 26, 2024 at 9:26 PM
estimated_db_size | 164.03 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

sairon, 2024-02-28:
This seems to be similar to #3180, can you try if it comes back after another reboot?

dai80027, 2024-02-28:
I tried a couple of reboots, but it didn't help. Thanks for looking at this!

sairon, 2024-02-28:
Could be something else in that case. Could you send the complete `dmesg` output after a boot?

dai80027, 2024-02-28:
Attached dmesg,txt. Thanks

-- 
Dave Ireland
303.834.7827
--

[94690.353233] Bluetooth: hci0: command 0xfc18 tx timeout
[94698.449351] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[94698.456085] Bluetooth: hci0: Failed to set baudrate
[94700.465389] Bluetooth: hci0: command 0x0c03 tx timeout
[94708.689600] Bluetooth: hci0: BCM: Reset failed (-110)
[94801.363291] Bluetooth: hci0: command 0xfc18 tx timeout
[94809.555444] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[94809.562187] Bluetooth: hci0: Failed to set baudrate
[94811.571481] Bluetooth: hci0: command 0x0c03 tx timeout
[94819.795638] Bluetooth: hci0: BCM: Reset failed (-110)
[94912.501300] Bluetooth: hci0: command 0xfc18 tx timeout
[94920.661445] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[94920.668187] Bluetooth: hci0: Failed to set baudrate
[94922.677475] Bluetooth: hci0: command 0x0c03 tx timeout
[94930.901621] Bluetooth: hci0: BCM: Reset failed (-110)
[95023.767232] Bluetooth: hci0: command 0xfc18 tx timeout
[95031.767344] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95031.774089] Bluetooth: hci0: Failed to set baudrate
[95033.783381] Bluetooth: hci0: command 0x0c03 tx timeout
[95042.007540] Bluetooth: hci0: BCM: Reset failed (-110)
[95080.956890] systemd-journald[138]: Data hash table of /var/log/journal/456339261f744da1a14c35725638b684/system.journal has a fill level at 75.0 (85336 of 113777 items, 25165824 file size, 294 bytes per hash table item), suggesting rotation.
[95080.956938] systemd-journald[138]: /var/log/journal/456339261f744da1a14c35725638b684/system.journal: Journal header limits reached or header out-of-date, rotating.
[95134.713056] Bluetooth: hci0: command 0xfc18 tx timeout
[95142.873200] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95142.879940] Bluetooth: hci0: Failed to set baudrate
[95144.889224] Bluetooth: hci0: command 0x0c03 tx timeout
[95153.113367] Bluetooth: hci0: BCM: Reset failed (-110)
[95245.786866] Bluetooth: hci0: command 0xfc18 tx timeout
[95253.978989] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95253.985741] Bluetooth: hci0: Failed to set baudrate
[95255.999010] Bluetooth: hci0: command 0x0c03 tx timeout
[95264.219155] Bluetooth: hci0: BCM: Reset failed (-110)
[95357.148634] Bluetooth: hci0: command 0xfc18 tx timeout
[95365.340744] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95365.347482] Bluetooth: hci0: Failed to set baudrate
[95367.356799] Bluetooth: hci0: command 0x0c03 tx timeout
[95375.580902] Bluetooth: hci0: BCM: Reset failed (-110)
[95468.510371] Bluetooth: hci0: command 0xfc18 tx timeout
[95476.702465] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95476.709200] Bluetooth: hci0: Failed to set baudrate
[95478.718496] Bluetooth: hci0: command 0x0c03 tx timeout
[95486.942631] Bluetooth: hci0: BCM: Reset failed (-110)
[95579.808040] Bluetooth: hci0: command 0xfc18 tx timeout
[95587.808161] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95587.814914] Bluetooth: hci0: Failed to set baudrate
[95589.824191] Bluetooth: hci0: command 0x0c03 tx timeout
[95598.048318] Bluetooth: hci0: BCM: Reset failed (-110)
[95690.849726] Bluetooth: hci0: command 0xfc18 tx timeout
[95698.913871] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95698.920623] Bluetooth: hci0: Failed to set baudrate
[95700.929917] Bluetooth: hci0: command 0x0c03 tx timeout
[95709.154079] Bluetooth: hci0: BCM: Reset failed (-110)
[95801.795362] Bluetooth: hci0: command 0xfc18 tx timeout
[95810.019480] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95810.026231] Bluetooth: hci0: Failed to set baudrate
[95812.035506] Bluetooth: hci0: command 0x0c03 tx timeout
[95820.259626] Bluetooth: hci0: BCM: Reset failed (-110)
[95913.253013] Bluetooth: hci0: command 0xfc18 tx timeout
[95921.381125] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[95921.387909] Bluetooth: hci0: Failed to set baudrate
[95923.397151] Bluetooth: hci0: command 0x0c03 tx timeout
[95931.621267] Bluetooth: hci0: BCM: Reset failed (-110)
[96024.582634] Bluetooth: hci0: command 0xfc18 tx timeout
[... 925 lines cut ...]
[115712.499925] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[115712.506745] Bluetooth: hci0: Failed to set baudrate
[115714.515958] Bluetooth: hci0: command 0x0c03 tx timeout
[115722.740089] Bluetooth: hci0: BCM: Reset failed (-110)
[115815.733394] Bluetooth: hci0: command 0xfc18 tx timeout
[115823.861609] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[115823.868433] Bluetooth: hci0: Failed to set baudrate
[115825.877530] Bluetooth: hci0: command 0x0c03 tx timeout
[115834.101646] Bluetooth: hci0: BCM: Reset failed (-110)
[115926.934955] Bluetooth: hci0: command 0xfc18 tx timeout
[115934.967069] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[115934.974050] Bluetooth: hci0: Failed to set baudrate
[115936.983097] Bluetooth: hci0: command 0x0c03 tx timeout
[115945.207219] Bluetooth: hci0: BCM: Reset failed (-110)
[116038.008516] Bluetooth: hci0: command 0xfc18 tx timeout
[116046.072624] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[116046.079443] Bluetooth: hci0: Failed to set baudrate
[116048.088657] Bluetooth: hci0: command 0x0c03 tx timeout
[116056.312775] Bluetooth: hci0: BCM: Reset failed (-110)
[116148.954070] Bluetooth: hci0: command 0xfc18 tx timeout
[116157.178182] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[116157.185005] Bluetooth: hci0: Failed to set baudrate
[116159.194217] Bluetooth: hci0: command 0x0c03 tx timeout
[116167.418320] Bluetooth: hci0: BCM: Reset failed (-110)
[116260.3

sairon, 2024-02-28:
Unfortunately, there's not the data I was looking for in the output you sent, however, it shows that your Yellow has been running at least for the past 32 hours at the time of posting. Can you please try rebooting it once again? Ideally, perform a cold boot, i.e. unplug it from the power adapter (or PoE) for a couple of seconds.

dai80027, 2024-02-28:
Jan, my apologies. I was restarting HA, not cold rebooting. That fixed my
problem.

I do appreciate your help.

Regards,

Dave

closed 2024-02-28 by dai80027