home-assistant/operating-system #3208: Backup storage not available after upgrading to HASSOS 12.0

proyleg, 2024-02-27:
### Describe the issue you are experiencing

After upgrading to 12.0, the backup storage is no longer accessible. 
In a second instance of HASS using HASSOS 11.5, this one continues to work fine

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.16-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Simply add a Backup storage  using Samba/Windows (CIFS parameters )and the connection never successfull

### Anything in the Supervisor logs that might be useful for us?

```txt
INFO (MainThread) [supervisor.mounts.manager] Creating or updating mount: HABackups
ERROR (MainThread) [supervisor.mounts.mount] Mounting HABackups did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-HABackups.mount for details.
```

### Anything in the Host logs that might be useful for us?

```txt
homeassistant systemd[1]: Mounting Supervisor cifs mount: HABackups...
homeassistant kernel: CIFS: Attempting to mount //192.168.x.x/HABackups
homeassistant kernel: CIFS: VFS: parse_reparse_point: unhandled reparse tag: 0x9000601a
homeassistant kernel: CIFS: VFS: cifs_read_super: get root inode failed
homeassistant mount[24652]: mount error(95): Operation not supported
homeassistant mount[24652]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
homeassistant systemd[1]: mnt-data-supervisor-mounts-HABackups.mount: Mount process exited, code=exited, status=32/n/a
homeassistant systemd[1]: mnt-data-supervisor-mounts-HABackups.mount: Failed with result 'exit-code'.
homeassistant systemd[1]: Failed to mount Supervisor cifs mount: HABackups.
```

### System information

## System Information

version | core-2024.2.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.6.16-haos
arch | x86_64
timezone | America/Toronto
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1476
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 12 février 2025 à 19:00
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-6.ui.nabu.casa
certificate_status | ready
instance_id | 61b613bf2e5a45cca289a576886e157e
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 234.0 GB
disk_used | 12.9 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Hydroqc Add-on (v1.2.0-1), Zigbee2MQTT (1.35.3-1), Samba share (12.3.0), rtl_433 MQTT Auto Discovery (0.8.1), Terminal & SSH (9.9.0), Advanced SSH & Web Terminal (17.1.1), Studio Code Server (5.15.0), rtl_433 (2024.2.8), Mosquitto broker (6.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 4
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 17 février 2024 à 14:25
-- | --
current_recorder_run | 26 février 2024 à 16:15
estimated_db_size | 3298.10 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

sleepy00, 2024-02-29:
I have the same issue. Please fix.

mwramos, 2024-05-28:
I have the same issue, too.
It was working and stopped.
It wasn't just the message that changed.