home-assistant/operating-system #3210: Unable to shutdown host

tomlut, 2024-02-28:
### Describe the issue you are experiencing

In preparation for some electrical work today that would require turning off my house power I attempted to shutdown home assistant like this:

[image]

However this acted like a Host restart. As soon as it shut down it started up again

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.16-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Attempt to shut down the host via the UI
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing unusual
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing unusual
```

### System information

## System Information

version | core-2024.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.6.16-haos
arch | x86_64
timezone | Australia/Hobart
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1399
Downloaded Repositories | 32

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 2 October 2024 at 10:00
relayer_connected | true
relayer_region | ap-southeast-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | ap-southeast-1-1.ui.nabu.casa
certificate_status | ready
instance_id | 7aac4f3d638c481abe5be42f20e81e7e
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 457.7 GB
disk_used | 34.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Grafana (9.1.1), InfluxDB (5.0.0), Mosquitto broker (6.4.0), Network UPS Tools (0.13.0), Samba share (12.3.0), Zigbee2MQTT (1.35.3-1), phpMyAdmin (0.9.1), MariaDB (2.6.1), Studio Code Server (5.15.0), Advanced SSH & Web Terminal (17.1.1), Samba Backup (5.2.0), Tailscale (0.16.0), Piper (1.5.0), Whisper (1.0.2), openWakeWord (1.10.0), ESPHome (2023.12.9)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 24
views | 63
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 21 February 2024 at 05:08
-- | --
current_recorder_run | 28 February 2024 at 23:38
estimated_db_size | 1932.36 MiB
database_engine | mysql
database_version | 10.6.12

</details>

sairon, 2024-02-29:
What HW is this running on? And was it working reliably in the past? You can try running `poweroff` on the host directly  (after switching to to the root shell from HA CLI using `login`) but it seems to be a quirk of the particular hardware or BIOS configuration.

tomlut, 2024-02-29:
Intel NUC11PAHI7 i7-1165G7 

Nothing special about it. 

I have not shut it down for years but do not remember having this issue in the past.

sairon, 2024-02-29:
There's plethora of "reboots instead of shutdown" issues discussed around the internet, the key to solving it might be in the BIOS config - see [this Wake on USB issue](https://www.intel.com/content/www/us/en/support/articles/000089681/intel-nuc.html) or [this related to stanby mode](https://www.intel.com/content/www/us/en/support/articles/000089009/intel-nuc.html). If you can't pinpoint where or if it ever worked correctly, we might be chasing ghosts here, as I checked the shutdown works correctly on my NUC and the Beelink mini PC :shrug: 

Running `poweroff` with an HDMI display connected could enlighten what's happening. If not, then issuing the command and disconnecting all cables and peripherals (except the power, obviously :sweat_smile:) to see if it stays powered down then.

tomlut, 2024-03-01:
I wonder if it could be the NUT addon restarting the Host?

I'll investigate further this weekend.

sairon, 2024-03-01:
Not sure if the add-on itself, but if you've got an USB-connected UPS, it might be the very same issue as the first one linked in my previous post. I'd suggest to try the poweroff with disconnected peripherals first.

tomlut, 2024-03-02:
Yeah that was it. Thanks.

closed 2024-03-02 by tomlut