home-assistant/operating-system #3211: Bluetooth integration fails on Raspberry Pi 3 HAOS greater 11.0

nicopasslick, 2024-02-29:
### Describe the issue you are experiencing

On my Raspberry Pi 3B+ with Home Assistant OS 12.0 currently installed, I unfortunately experienced that the Bluetooth integration for HAOS regularly crashes/becomes unavailable with HAOS version greater 11.0. The Bluetooth integration is usually still available when HAOS is rebooted or started, but then becomes unavailable within a maximum of one hour (usually significantly faster). This is most noticeable for me in the fact that my BTHome sensors will no longer be available from this point on.

Currently I always have to downgrade to HAOS version 11.0 to have Bluetooth consistently available again and I had hoped that this problem would be solved with version 12.0 at the latest, but the problem persists. 

I use onboard Bluetooth from my Raspberry Pi 3B+ board and no external adapter. **Updates of Home Assistant Core / Supervisor have no influence, the problem only ever appears when upgrading HAOS to >11.0. Other components are up-to-date.**

Similar issues: #3207 #3180 #3047 (solution approaches in these issues without effect)

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade HAOS to version greater 11.0 (in this case: 12.0)
2. Wait for bluetooth integration to become unavailable (noticeable for me in form of missing BTHome sensors) - max. 1 hour
3. Integration becomes unavailable...
4. Cry and downgrade to HAOS 11.0 - everything works fine now (Core and Supervisor up-to-date)

### Anything in the Supervisor logs that might be useful for us?

```txt
24-02-28 23:50:50 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
24-02-28 23:50:50 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
24-02-28 23:50:50 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2023.11.0
24-02-28 23:50:50 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
24-02-28 23:50:53 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/aarch64-hassio-cli with version 2023.11.0 - 172.30.32.5
24-02-28 23:50:53 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-dns with version 2023.06.2
24-02-28 23:50:53 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
24-02-28 23:50:56 INFO (MainThread) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/aarch64-hassio-dns with version 2023.06.2 - 172.30.32.3
24-02-28 23:50:56 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
24-02-28 23:50:56 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.12.0
24-02-28 23:50:56 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
24-02-28 23:51:01 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.12.0 - 172.30.32.4
24-02-28 23:51:01 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2023.06.0
24-02-28 23:51:01 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2023.06.2
24-02-28 23:51:01 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
24-02-28 23:51:05 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/aarch64-hassio-multicast with version 2023.06.2 - Host
24-02-28 23:51:05 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
24-02-28 23:51:05 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/raspberrypi3-64-homeassistant with version 2024.2.5
24-02-28 23:51:05 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 12.0 / BootSlot B
24-02-28 23:51:08 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
24-02-28 23:51:08 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/9074a9fa repository
24-02-28 23:51:08 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
24-02-28 23:51:08 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
24-02-28 23:51:10 INFO (MainThread) [supervisor.store] Loading add-ons from store: 78 all - 78 new - 0 remove
24-02-28 23:51:10 INFO (MainThread) [supervisor.addons.manager] Found 3 installed add-ons
24-02-28 23:51:10 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.9.0
24-02-28 23:51:10 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/brenner-tobias/cloudflared/aarch64 with version 5.1.4
24-02-28 23:51:10 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-configurator with version 5.8.0
24-02-28 23:51:10 INFO (MainThread) [supervisor.backups.manager] Found 40 backup files
24-02-28 23:51:11 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
24-02-28 23:51:11 INFO (MainThread) [supervisor.ingress] Loaded 1 ingress sessions
24-02-28 23:51:11 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup
24-02-28 23:51:11 INFO (MainThread) [supervisor.resolution.check] System checks complete
24-02-28 23:51:11 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
24-02-28 23:51:11 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-02-28 23:51:11 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup
24-02-28 23:51:11 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
24-02-28 23:51:11 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
[... 287 lines cut ...]
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "src/habluetooth/scanner.py", line 380, in habluetooth.scanner.HaScanner._async_restart_scanner
  File "src/habluetooth/scanner.py", line 294, in _async_start
habluetooth.scanner.ScannerStartError: hci0 (B8:27:EB:3C:56:D2): Timed out starting Bluetooth after 15 seconds; Try power cycling the Bluetooth hardware.

</details>

<details><summary>Logger: habluetooth.scanner, source: runner.py:188</summary>

Logger: habluetooth.scanner
Source: runner.py:188
First occurred: 12:03:16 AM (2 occurrences)
Last logged: 12:05:16 AM

hci0 (B8:27:EB:3C:56:D2): Error stopping scanner: [org.bluez.Error.InProgress] Operation already in progress

</details>

<details><summary>Full Home Assistant Core log</summary>

[home-assistant_2024-02-29T00-33-26.162Z.log](https://github.com/home-assistant/operating-system/files/14441310/home-assistant_2024-02-29T00-33-26.162Z.log)

</details>

sairon, 2024-02-29:
Similarly to #3047 it seems to me that the adapter still works but `habluetooth` ends up stuck trying to restart it - IMHO it should be more resilient to that, please create a Core issue for debugging it on that side. I don't see any apparent issues with the adapter, can you also check `dmesg | grep -i bluetooth`?

npasslick, 2024-02-29:
This is the ```dmesg``` output

[image]

bdraco, 2024-02-29:
> Similarly to #3047 it seems to me that the adapter still works but `habluetooth` ends up stuck trying to restart it - IMHO it should be more resilient to that, please create a Core issue for [...]

Do you have a suggestion on what `hablueooth` should do in this case?

bdraco, 2024-02-29:
The scanner start code is here https://github.com/Bluetooth-Devices/habluetooth/blob/71bd22a4bea50fff5ff9b8661b4c19814a512331/src/habluetooth/scanner.py#L249

bdraco, 2024-02-29:
@npasslick Does the problem go away if you turn on passive scanning

bdraco, 2024-02-29:
Its timing out on the dbus call

https://github.com/hbldh/bleak/blob/054b092e19c7a45d5ea07d8948022f54117fe5e8/bleak/backends/bluezdbus/manager.py#L408

npasslick, 2024-02-29:
> @npasslick Does the problem go away if you turn on passive scanning

I will try that

npasslick, 2024-02-29:
> @npasslick Does the problem go away if you turn on passive scanning

I have upgraded my system back to HAOS 12.0 and activated passive scanning immediately.

The activation of passive scanning seems to have actually prevented the problem _so far_, as I already mentioned, the integration usually crashes within an hour of system startup, and this period has now elapsed. Seems to be a suitable workaround for now, but of course it won't solve the problem long-term. 

I will have to wait and see if the Bluetooth integration does not crash over a longer period of time with passive scanning, but looks good so far.

bdraco, 2024-02-29:
```
[bluetooth]# scan off
Discovery stopped
[bluetooth]# scan on
Discovery started
[bluetooth]# scan on
Failed to start discovery: org.bluez.Error.InProgress
[bluetooth]#
```

bdraco, 2024-02-29:
If it happens again in advice mode, can you try turning the `scan off` with `bluetoothctl`, pasting the output here, and seeing if it can recover?

npasslick, 2024-02-29:
> If it happens again in advice mode, can you try turning the `scan off` with `bluetoothctl`, pasting the output here, and seeing if it can recover?

Sure thing, I guess with advice mode you mean active / non-passive mode?

bdraco, 2024-02-29:
Sorry, flipping between too many things this morning: should have wrote: active mode

npasslick, 2024-02-29:
> Sorry, flipping between too many things this morning: should have wrote: active mode

No worries, i will try that ^^

npasslick, 2024-02-29:
> If it happens again in ~advice~active mode, can you try turning the `scan off` with `bluetoothctl`, pasting the output here, and seeing if it can recover?

Unsurprisingly, the bluetooth integration failed again in active mode. It does not recover after `scan off` or `scan on`

```scan off``` --> ```org.bluez.Error.Failed```
```scan on``` ---> ```scan off``` --> ```org.bluez.Error.InProgress```

<details><summary>bluetoothctl output</summary>

[image]

</details>

<details><summary>Full bluetooth debug log</summary>

[home-assistant_bluetooth_2024-02-29T22-09-21.741Z.log](https://github.com/home-assistant/operating-system/files/14454082/home-assistant_bluetooth_2024-02-29T22-09-21.741Z.log)

</details>

bdraco, 2024-02-29:
It seems like there isn't much we can do in `habluetooth` if its failing in `bluetoothctl` its too far down the stack that any retries are going to work.

I expect passive is going to work fine because it doesn't hit the path where you get the bluez errors

npasslick, 2024-02-29:
> It seems like there isn't much we can do in `habluetooth` if its failing in `bluetoothctl` which means its too far down the stack for any retries to work.
> 
> I expect passive is going to work fine because it doesn't hit the path where you get the bluez errors

Hmm I see, then something underlying must have changed after HAOS 11.0, but thanks for the tip with the passive mode, for my purposes this should be sufficient, but others will certainly have to deal with this problem. Maybe someone else can identify the underlying problem.

npasslick, 2024-02-29:
@bdraco Thank you much for your effort so far, at least I can use the current version of HAOS now ^^
Keep up the good work, I will make a small donation to your Food Bank as a gesture of my appreciation (y)

CasperNaudts, 2024-03-08:
Having the same issue on my Raspberry Pi 4. Passive mode fixes the initialisation issue for the bluetooth integration but breaks  my switchbot bots. Switchbot curtains do work in passive mode. If there is some additional stuff, I can test for you guys feel free to let me know and I will give it a shot.

JesusSanchezLopez, 2024-03-12:
I upgraded HA Core from 2023.12.3 to 2024.3.0 and experienced the similar issue. My Bluetooth sensors would periodically disconnect then reconnect sporadically. I downgraded to 2023.12.3 and Bluetooth devices work again:

[image]

CasperNaudts, 2024-03-22:
Just updated to the latest version of HA core and everything seems to work again.

JesusSanchezLopez, 2024-03-22:
I also upgraded to HA Core 2024.3.2 and within an hour I started getting sensor unavailable messages.
Reverting back to 2024.1.5

npasslick, 2024-03-22:
Likewise, no improvement for me with the latest version of HA Core in Bluetooth active mode

npasslick, 2024-03-26:
Well, now I'm in a situation where I've got a new device and passive scanning is simply not enough to ensure reliable operation. I hope someone can take care of this soon, unfortunately my skills are currently not sufficient for hardware-oriented programming :(

apetryk2, 2024-04-07:
On my end the bluetooth issue with raspberry Pi popped up in 12.1.  Downgrading to 12.0 fixes the problem.  In case others are searching for how to downgrade the OS version issue the following command in an SSH session:

`ha os update --version 12.0`

CasperNaudts, 2024-04-08:
I found a weird way to temporarily fix the issue. Bluetooth isn't willing to initialize in active scanning after a software reboot. When I unplug the power and replug the power Bluetooth initialises with no issues in active mode until the next software reboot.

apetryk2, 2024-04-11:
I just tried 12.2 and the problem may be fixed.

CasperNaudts, 2024-04-12:
Problem also is fixed here with the new 12.2 update.

frli4797, 2024-04-17:
I'm on 12.2 with a Homeassistant Yellow (CM4). Problem still persists.

frli4797, 2024-04-22:
Seems like this problem has been pending for a bit. Any progress?

bdraco, 2024-04-22:
>  I found a weird way to temporarily fix the issue. Bluetooth isn't willing to initialize in active mode after a software reboot. When I unplug the power and replug the power Bluetooth initialises with no issues in active mode until the next software reboot.

Sadly its pretty common that these chips will stop responding and need to be powered down to get them to work again.

The upcoming HAOS 12.3 uses the 6.6.x linux kernels which, at least for me, seems to be a bit more stable with the RPi Bluetooth.

auditdata, 2024-04-30:
I have a rpi4-64. What gets me is 4 weeks ago and for at least a year before it all worked perfectly. Nothing in the hardware has changed. So it must be HAOS. What changed?
Someone must know.
I have tried the scan thing but makes no difference and tried the power off/power on thing and again no difference. The only thing that works is turning the mains power off and on again. This is not a solution in the long term. 
```
[bluetooth]# scan off
Failed to stop discovery: org.bluez.Error.Failed
[bluetooth]# scan on
[bluetooth]# 
```
The only thing that works is turning the mains power off and on again. This is not a solution in the long term.

auditdata, 2024-04-30:
[image]
Sad life now :-) :-) :-) :-(

StahlTim, 2024-05-08:
> I have a rpi4-64. What gets me is 4 weeks ago and for at least a year before it all worked perfectly. Nothing in the hardware has changed. So it must be HAOS. What changed?
> Someone must know.
> I have [...]

Same for me on a rpi3. Only power cycling (manually taking away the power for some time) fixes the issue for a couple of hours/days.

agners, 2024-05-13:
@auditdata @StahlTim did you try HAOS 12.3? It comes with and updated Linux kernel which might change things. Also do you use active or passive mode?

npasslick, 2024-05-13:
Seems like HAOS 12.3 has finally fixed this problem at least for me. The Bluetooth integration has been working in active mode since this morning without crashing. We'll see how it works in the long run and what other affected users running HAOS report, but I have a good feeling.

auditdata, 2024-05-13:
> @auditdata @StahlTim did you try HAOS 12.3? It comes with and updated Linux kernel which might change things. Also do you use active or passive mode?

Not sue as I cut my losses and bought an external Bluetooth adapter and all worked again. I guess I could unplug it and have a look but might wait for others to confirm now working again.

npasslick, 2024-05-13:
> > @auditdata @StahlTim did you try HAOS 12.3? It comes with and updated Linux kernel which might change things. Also do you use active or passive mode?
> 
> I am on 12.3 since its release and bluetooth is still not working in active mode (only for a couple of hours/days after power toggle).
> 
> Should I switch to passive mode?

Passive mode should generally work even with the bug (at least that was my workaround until the recent HAOS update), but keep in mind that you will lose functionalities that require active Bluetooth scanning if you have devices that provide those

StahlTim, 2024-05-15:
@agners @npasslick My sensors require me to stick with active mode. However, did another power-unplug 2d ago (I think the first time after upgrading to 12.3).

Maybe upgrading and rebooting to 12.3 alone did not fix the issue, but an additional power cycle (actual removing power physically) helped to complete the process.

Will monitor further and report if failing/dropping again.

StahlTim, 2024-05-27:
Update: Was on vacation - Bluetooth devices did disconnect again during my absence. Issue still present.

apetryk2, 2024-05-27:
Huh.  I have two servers in two locations both running on Raspberry Pi.  In one location the issue is fixed.  In the other location it is not.  Both on the same versions of HA OS and HA.

JiDoan, 2024-09-04:
Unfortunately, my Raspberry Pi 4 has very unstable Bluetooth even on the latest version of Home Assistant OS. Anything we can do to help debug the issue?

[image]

FlareFlo, 2024-09-04:
For me, this issue only went away after upgrading to a Pi 5 4gb from a Pi3 B 1gb

Xavantex, 2024-09-23:
Had similiar issues, but I upgraded from 11.5. I finally got mad enough to do a full backup and rebuild the SD card with a new image, the bluetooth reappeared without issue. EMQX on the other hand took a beating, but that was quickly solved :)

andrewquamme, 2024-10-13:
I just found out today that after working flawlessly for almost 9 months. all my Bluetooth sensors apparently dropped on Oct 3 and are currently 'Unavailable'

I ran all pending updates (it had been a while...) and still nothing

Logs show repeated Successful fall-back to passive scanning mode after active scanning failed

se-o-matic, 2024-11-13:
Hi,
i just ran into this problem, too.
Two days ago, I switched from an old installation of HA in a docker to a fresh installation of the HA operating system.
Installed via the rpi imaging tool on an RPI3+
And as described here, my one and only bluetooth sensor (GOVEE temperature) fails periodically.

pehovorka, 2024-11-13:
Hi, I have the same issue. I'm running HAOS on RPi4 and have to reboot it every few days because the BT connection to my Xiaomi Temperature and Humidity sensors drops. It's not ideal since I use the readings from the sensors to control heating.

I'm currently at HAOS 13.2 and Core 2024.11.1, but the issue has unfortunately been present for many months. 

Is there anything we can do to help this issue move forward?

Safislab777, 2024-11-19:
Any updates on this? Seems as though this is creating issues for many.

andrewquamme, 2025-02-19:
This has not been resolved

FlyMyPG, 2025-02-21:
This has not been resolved.

adjscent, 2025-03-31:
This still has not been resolved

FlyMyPG, 2025-06-30:
This issue is still not resolved.

VicenteVeiga89, 2025-07-12:
This issue still happens on the latest version of HAOS

FlyMyPG, 2025-07-13:
It's been a year since this broke. Why hasn't a regression been done?

garibarba, 2025-11-09:
```
[bluetoothctl]> scan off
[bluetoothctl]> scan on
```
fixed it for me. It is somewhat unstable but it is working so far.

deviantintegral, 2025-12-29:
A repair for "Bluetooth adapter requires system power cycle" started surfacing for me on my Pi 4 within the last week. Did Home Assistant recently add this as a repair? I'm on HAOS 16.3.

Luckily I don't have any Bluetooth devices set up at this location so I can ignore it. Of course, there is a ton of bluetooth LE traffic that `bluetoothctl` picks up, but toggling scanning doesn't seem to do anything to clear the repair or allow active mode to turn back on.

Is there any chance this is related to setting `dtoverlay=miniuart-bt`?

sxdjt, 2025-12-30:
This is has been happening at least weekly for me the last ~4 weeks.

damozaz, 2026-01-05:
I have the same issue on latest HA on a Raspberry Pi 3b.  The on board BT suddenly stopped working recently and won't restart.  I tried the bluetoothctl scan off/on but this fails because the bluetooth hardware isn't initialised so the command won't run.  I then tried a BT dongle which initially worked fine but then after rebooting this also go stuck initialising and so far I haven't managed to get ti to restart.  It's a real pain because one of my main uses for HA is to monitor temperatures with several Govee BT sensors

k1asa, 2026-02-12:
Same here on my rpi5-64.

Installation method: Home Assistant OS
Core: 2026.2.1
Supervisor: 2026.02.1
Operating System: 17.0
Frontend: 20260107.2

deviantintegral, 2026-02-13:
Is there any chance this is triggered by specific BLE advertisements? I have one install where I haven’t been there lately, and I haven’t seen a Bluetooth error since leaving. The biggest Bluetooth change since would be the removal of our Oral-B toothbrushes from range.

68040, 2026-08-18:
The same permanent problem with Blutooth on a RPi 3.  

Install Home Assistant OS
Core 2026.8.2
Supervisor 2026.07.5
Operating System 18.2
Frontend 20260729.7

machine: raspberrypi3-64

[1 more reply]