home-assistant/operating-system #3213: Generic image cannot be flashed on usb stick more then 45%.

AditaNi, 2024-02-29:
### Describe the issue you are experiencing

I cannot Flash an usb stick with Generic Image 12 more then 45% with balena etcher.

I tryed in linux and also in windows to flash the usb stick.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Hy Guys,

1. Download generic image x86-64 12.0
2. Download balena etcher
3. plug in the usb stick onto your laptop
4. start the flashing process.

Everything is working fine until 45%.

In windows 10 ...I receive a long error.
In linux is just hangs at 45 %

Regards
Adrian

### Anything in the Supervisor logs that might be useful for us?

```txt
nope...because nothing is installed.
```

### Anything in the Host logs that might be useful for us?

```txt
also not
```

### System information

Thinkpad edge E145 laptop

sairon, 2024-03-01:
Sorry for the trouble, however it's almost surely not an issue of the OS image itself - in the end it's just a bunch of binary data that needs to be written on the disk. I suggest filing an issue in the Balena Etcher support channels, or trying a different tool for the job.

closed 2024-03-01 by sairon

AditaNi, 2024-03-01:
Hi,

have you retested and you cannot confirm it?

Regards
Adrian

sairon, 2024-03-01:
Yeah, here's a screenshot of the verification step just a second before it finishes. I tried a random USB drive I had around and latest x86 image (compressed, to leave unpacking on the Etcher):

[image]

Not sure if you tried, but maybe check it with a different drive first. If it doesn't help, refer to Balena Echer support: https://github.com/balena-io/etcher/blob/master/docs/SUPPORT.md

AditaNi, 2024-03-01:
i have tried with 5 different usb sticks and two different pc's....but same windows 10 version.
Maybe i have bad luck...who knows.