home-assistant/operating-system #3214: Raspberry Pi 5 with HassOS booted from nvme SSD shows cma_alloc errors in dmesg at startup

jbouwh, 2024-03-01:
### Describe the issue you are experiencing

I have installed Home Assistant OS 12 on an NMVE SSD.
Hardware:
- Raspberry PI 5
- Geekworm X1003 M2 hat: https://wiki.geekworm.com/X1003
- SAMSUNG MZAL4256HBJD-00BL2: https://www.amazon.nl/dp/B08K79T9G5

To avoid pcie errors I added the settings below to `config.txt`:

```log
# Enable the PCIe External connector.
dtparam=pciex1
# This line is an alias for above (you can use either/or to enable the port).
dtparam=nvme
dtparam=pciex1_gen=3
```
I know that the `dtparam=pciex1_gen=3` is not officially supported, but without it I have a lot of corrected `pcie` errors in the logs.

My `dmesg` looks good overall except for some `cma_alloc` errors. I am not sure if they are related to the NVME storage.

```log
...
[    0.514198] nvme nvme0: Shutdown timeout set to 8 seconds
[    0.516601] nvme nvme0: allocated 64 MiB host memory buffer.
[    0.519481] cma: cma_alloc: linux,cma: alloc failed, req-size: 2 pages, ret: -12
[    0.519494] cma: cma_alloc: linux,cma: alloc failed, req-size: 8 pages, ret: -12
[    0.519506] cma: cma_alloc: linux,cma: alloc failed, req-size: 2 pages, ret: -12
[    0.519513] cma: cma_alloc: linux,cma: alloc failed, req-size: 8 pages, ret: -12
[    0.519523] cma: cma_alloc: linux,cma: alloc failed, req-size: 2 pages, ret: -12
[    0.519530] cma: cma_alloc: linux,cma: alloc failed, req-size: 8 pages, ret: -12
[    0.519539] cma: cma_alloc: linux,cma: alloc failed, req-size: 2 pages, ret: -12
[    0.519546] cma: cma_alloc: linux,cma: alloc failed, req-size: 8 pages, ret: -12
[    0.519735] nvme nvme0: 4/0/0 default/read/poll queues
[    0.525784] nvme nvme0: Ignoring bogus Namespace Identifiers
[    0.531980]  nvme0n1: p1 p2 p3 p4 p5 p6 p7 p8
[    0.519735] nvme nvme0: 4/0/0 default/read/poll queues
[    0.525784] nvme nvme0: Ignoring bogus Namespace Identifiers
[    0.531980]  nvme0n1: p1 p2 p3 p4 p5 p6 p7 p8
[    0.619473] brcm-pcie 1000120000.pcie: link up, 5.0 GT/s PCIe x4 (!SSC)
[    0.619497] pci 0001:01:00.0: [1de4:0001] type 00 class 0x020000
[    0.619514] pci 0001:01:00.0: reg 0x10: [mem 0xffffc000-0xffffffff]
[    0.619524] pci 0001:01:00.0: reg 0x14: [mem 0xffc00000-0xffffffff]
[    0.619533] pci 0001:01:00.0: reg 0x18: [mem 0xffff0000-0xffffffff]
[    0.619605] pci 0001:01:00.0: supports D1
[    0.619609] pci 0001:01:00.0: PME# supported from D0 D1 D3hot D3cold
[    0.631479] pci_bus 0001:01: busn_res: [bus 01-ff] end is updated to 01
[    0.631491] pci 0001:00:00.0: BAR 8: assigned [mem 0x1f00000000-0x1f005fffff]
[    0.631497] pci 0001:01:00.0: BAR 1: assigned [mem 0x1f00000000-0x1f003fffff]
[    0.631504] pci 0001:01:00.0: BAR 2: assigned [mem 0x1f00400000-0x1f0040ffff]
[    0.631511] pci 0001:01:00.0: BAR 0: assigned [mem 0x1f00410000-0x1f00413fff]
[    0.631518] pci 0001:00:00.0: PCI bridge to [bus 01]
[    0.631522] pci 0001:00:00.0:   bridge window [mem 0x1f00000000-0x1f005fffff]
[    0.631528] pci 0001:00:00.0: Max Payload Size set to  256/ 512 (was  128), Max Read Rq  512
[    0.631539] pci 0001:01:00.0: Max Payload Size set to  256/ 256 (was  128), Max Read Rq  512
[    0.631598] pcieport 0001:00:00.0: enabling device (0000 -> 0002)
[    0.631631] pcieport 0001:00:00.0: PME: Signaling with IRQ 42
[    0.631684] pcieport 0001:00:00.0: AER: enabled with IRQ 42
...
```

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start the PI with Raspberry PI OS with additional parameters in `config.txt`
2. Flash Home Assistant OS with imager to the SSD
3. Change the RPI boot order (I placed SD card before SSD to allow easy maintanance).
4. Adjust the `config.txt` on the on the flashed image and add the additional parmeters:
```bash
mkdir boot
mount -t vfat /dev/nvme0n1p1 ~/boot
cd boot
vi config.txt
```
5. Boot Hass OS from the SSD
6. Install Home Assistant OS (Or restore)
7. Use the HA CLI to consult DMESG

### Anything in the Supervisor logs that might be useful for us?

```txt
No errors in the supervisor logs
```

### Anything in the Host logs that might be useful for us?

```txt
No errors in the supervisor logs
```

### System information

## System Information

version | core-2024.3.0b3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4999
Installed Version | 1.34.0
Stage | running
Available Repositories | 1466
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | beta
supervisor_version | supervisor-2024.02.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 234.0 GB
disk_used | 44.3 GB
healthy | true
supported | true
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.9.0), InfluxDB (5.0.0), Studio Code Server (5.15.0), Portainer (2.19.4-3), MariaDB (2.6.1), Silicon Labs Multiprotocol (2.4.4), Matter Server (5.3.0), ESPHome (2024.2.1), Mosquitto broker (6.4.0), Piper (1.5.0), Whisper (1.0.2), Filebrowser (2.23.0_12), Samba share (12.3.0), Log Viewer (0.17.0), Grafana (9.1.3), openWakeWord (1.10.0), AppDaemon (0.16.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 9
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 20 februari 2024 om 19:00
-- | --
current_recorder_run | 1 maart 2024 om 08:20
estimated_db_size | 2098.02 MiB
database_engine | mysql
database_version | 10.6.12

</details>

agners, 2024-03-01:
Hm it seems that the NVMe you are using uses a feature named HMB (Host Memory Buffer), which essentially borrows some memory from the host system to operate the NVMe (see also this [Gigabyte sales page](https://www.gigabyte.com/FileUpload/Global/KeyFeature/1282/index.html)).

Now it seems that the kernel tries to allocate this from the CMA region, which is exhausted in that particular kernel config.

We largely rely on the Raspberry Pi kernel and kernel configuration. Can you reproduce this with Raspberry Pi OS as well (with the same hardware configuration?). If so, I'd recommend open an issue in the Raspberry Pi kernel tracker at https://github.com/raspberrypi/linux/.

jbouwh, 2024-03-01:
I can try to reproduce this but in that case i'll be booting from SD card with the NVMe installed. I'd rather not need to reflash the NMVe with Raspberry Pi OS as I choose tho use it as my production device now. Not use if this issue will impact though as the system runs fine without ant errors with this config.

jbouwh, 2024-03-01:
Trying to boot from SD card with raspberry Pi kernel gives a lot of corrected `BadDLLP` errors. But this causes that I miss the first part of the `dmesg` boot log. Any idea how to get this log?

jbouwh, 2024-03-01:
On `journalctl` on the RPI os I am not seeing CM related stuff. Let me know if I can do more.

jbouwh, 2024-03-01:
I have tried to increase monitor or increase the cma alloction. It seems the default is 64M for the Raspberry Pi 5.

```
Use `ha` to access the Home Assistant CLI.
# cat /proc/meminfo | grep -i cma
CmaTotal:          65536 kB
CmaFree:               0 kB
```

All the cma memory is used.

If I try to increase the to a higher value, e.g. then there is some free cma memory, but the kernel gets tainted and logs cma errors.

realhagie, 2024-03-01:
Hi,
I also did a HAOS install on a rpi5 with an NVME . I have different hardware but same errors :

Many of tehese errors : 
`[ 1823.692652] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[ 1823.692667] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[ 1823.692671] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00001000/00002000
[ 1823.692674] pcieport 0000:00:00.0:    [12] Timeout
`
and much less of these : 
`[ 1828.834881] nvme 0000:01:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[ 1828.834883] nvme 0000:01:00.0:   device [126f:2263] error status/mask=00009040/0000e000
[ 1828.834886] nvme 0000:01:00.0:    [ 6] BadTLP
[ 1828.834889] nvme 0000:01:00.0:    [12] Timeout
`
My Hardware is a Argon NEO 5 M.2 NVME PCIE Case with a Patriot M.2 P300 256GB

vin100ross, 2024-03-01:
@realhagie, these errors are different than those reported by the OP. I have a different NVME, but I reduced my "BadTLP" and "BadDLLP" by 90% by disabling my Bluetooth integration (possibly radio interference, I use proxies anyway). I eliminated all remaining "Timeout" and BadTLP errors by adding pcie_aspm=off to cmdline.txt. I now get a BadDLLP every few days at most. I'm Hoping a shielded FFC cable I'm waiting for will eliminate these completely. YMMV. See #3188.

jbouwh, 2024-03-02:
Please do not use this ticket for a discussion about another issue!

jbouwh, 2024-03-02:
Dropping down to gen 2 introduces the errors mentioned. What is interesting is what you see in `dmesg` Apparently my SSD features the host memory buffer feature like @agners mentioned. This claims s continues memory bloc, but that failed. If I increase the CMA memory block, then this allocation probably succeeds, but then my logs are flooded with Kerbal tainting errors, indicating there's some conflict.
Conclusion is that my system runs fine, the performance is okay, no runtime errors, but may be the performance could be even better if the host memory buffer feature would work as expected.

jbouwh, 2024-03-02:
Found a solution to correct the cma allocation. It **not** does to work to set the CMA in `cmdline.txt` BTW.

The default cma allocation of 64M seems not to be enough, so I adjusted it to 96M.

```
#  cat /proc/meminfo | grep -i cma
CmaTotal:          98304 kB
CmaFree:           21296 kB
```

To enable this I added `dtoverlay=cma,cma-96` to `/mnt/boot/config.txt`

So my adjusted `config.txt` now has the following additional lines.

```
dtparam=pciex1
dtparam=nvme
dtparam=pciex1_gen=3
dtoverlay=cma,cma-96
```

The errors in `dmesg` are gone now, also no pci errors in my host logs.

IMO it would be good if users could configure these things in a more accessible way.
I am closing this issue as it seems not an error with Home Assistant OS.

closed 2024-03-02 by jbouwh