home-assistant/operating-system #3216: HAOS 12: ZHA with SkyConnect multiprotocol stopped working after 2024.2.4

lufetico, 2024-03-02:
### Describe the issue you are experiencing

I have a Home Assistant Green with HAOS 12. I've just updated HA Core from 2024.2.3 to 2024.2.4 and ZHA won't start. There was NO blocking change about it and my entire Zigbee network has gone to ****.

My Zigbee hub is SkyConnect in multiprotocol mode.

I come from this other bug reported in the ZHA integration:

https://github.com/home-assistant/core/issues/111646

The team in charge of the ZHA integration says that the problem must be on the OS side as they have not made any modifications between those two versions.

### What operating system image do you use?

green (Home Assistant Green)

### What version of Home Assistant Operating System is installed?

12

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Install HAOS 12.
2. Update HA Core to 2024.2.4 or 2024.2.5.
3. ZHA integration start falling. 

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: homeassistant.config_entries
Source: config_entries.py:444
First occurred: 16:27:26 (3 occurrences)
Last logged: 16:46:29

Error setting up entry SkyConnect Multi-PAN for zha: Incorrect firmware installed
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/zha/init.py", line 163, in async_setup_entry
zha_gateway = await ZHAGateway.async_from_config(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 196, in async_from_config
await instance.async_initialize()
File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 214, in async_initialize
await app.startup(auto_form=True)
File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 225, in startup
await self.connect()
File "/usr/local/lib/python3.12/site-packages/bellows/zigbee/application.py", line 148, in connect
await ezsp.startup_reset()
File "/usr/local/lib/python3.12/site-packages/bellows/ezsp/init.py", line 125, in startup_reset
await self.reset()
File "/usr/local/lib/python3.12/site-packages/bellows/ezsp/init.py", line 151, in reset
await self._gw.reset()
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/zha/init.py", line 191, in async_setup_entry
raise ConfigEntryError("Incorrect firmware installed") from exc
homeassistant.exceptions.ConfigEntryError: Incorrect firmware installed
```

### Anything in the Host logs that might be useful for us?

```txt
Logger: homeassistant.config_entries
Source: config_entries.py:444
First occurred: 16:27:26 (3 occurrences)
Last logged: 16:46:29

Error setting up entry SkyConnect Multi-PAN for zha: Incorrect firmware installed
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/zha/init.py", line 163, in async_setup_entry
zha_gateway = await ZHAGateway.async_from_config(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 196, in async_from_config
await instance.async_initialize()
File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 214, in async_initialize
await app.startup(auto_form=True)
File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 225, in startup
await self.connect()
File "/usr/local/lib/python3.12/site-packages/bellows/zigbee/application.py", line 148, in connect
await ezsp.startup_reset()
File "/usr/local/lib/python3.12/site-packages/bellows/ezsp/init.py", line 125, in startup_reset
await self.reset()
File "/usr/local/lib/python3.12/site-packages/bellows/ezsp/init.py", line 151, in reset
await self._gw.reset()
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/zha/init.py", line 191, in async_setup_entry
raise ConfigEntryError("Incorrect firmware installed") from exc
homeassistant.exceptions.ConfigEntryError: Incorrect firmware installed
```

### Additional information

[image]

agners, 2024-03-04:
Between Operating System 11.5 and 12.0 the Linux kernel changed so this could lead into a different enumeration of USB devices etc.

From what I can tell, ZHA suddenly started to access the SkyConnect directly instead of going through the Mutiprotocol add-on :thinking: 

Did the downgrade of OS as suggested by puddly changed/helped in any case (https://github.com/home-assistant/core/issues/111646#issuecomment-1973248653)?

agners, 2024-03-04:
FWIW, I think it is best if this issue is continued to be handled in the Core issue for now. If logs can show that the OS is now misbehaving in some shape or form, we can investigate from OS side.

fmms, 2024-03-05:
Same issue here.