home-assistant/operating-system #3222: No Shutdown on Intel NUC PC, Homeassistant reboots

muhkuh666, 2024-03-04:
### Describe the issue you are experiencing

I have the issue that since 12.0 my Homeassistant doesnt shutdown when I use shutdown procedure. It simply reboots! in 11.5 this shutdown works without a problem, I just tried.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Choose 11.5 while start up tried to shutdown and it works, with 12 it reboots, doesnt turn off
2.
3.

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m24-03-04 20:10:01 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager[0m
[32m24-03-04 20:10:01 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc[0m
[32m24-03-04 20:10:01 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1[0m
[32m24-03-04 20:10:01 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1[0m
[32m24-03-04 20:10:01 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.host.services] Updating service information[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information[0m
[33m24-03-04 20:10:02 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.host.network] Updating local network information[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.mounts.manager] Initializing all user-configured mounts[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2024.02.1[0m
[32m24-03-04 20:10:02 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin[0m
[32m24-03-04 20:10:02 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning hassio_cli application[0m
[32m24-03-04 20:10:03 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2024.02.1 - 172.30.32.5[0m
[32m24-03-04 20:10:03 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2023.06.2[0m
[32m24-03-04 20:10:03 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin[0m
[32m24-03-04 20:10:03 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning hassio_dns application[0m
[32m24-03-04 20:10:04 INFO (MainThread) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2023.06.2 - 172.30.32.3[0m
[32m24-03-04 20:10:04 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf[0m
[32m24-03-04 20:10:04 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0[0m
[32m24-03-04 20:10:04 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin[0m
[32m24-03-04 20:10:04 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning hassio_audio application[0m
[32m24-03-04 20:10:04 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0 - 172.30.32.4[0m
[32m24-03-04 20:10:04 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2023.06.0[0m
[32m24-03-04 20:10:04 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-multicast with version 2023.06.2[0m
[32m24-03-04 20:10:04 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin[0m
[32m24-03-04 20:10:04 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning hassio_multicast application[0m
[32m24-03-04 20:10:05 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2023.06.2 - Host[0m
[32m24-03-04 20:10:05 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 10 Home Assistant secrets[0m
[32m24-03-04 20:10:05 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/generic-x86-64-homeassistant with version 2024.2.5[0m
[32m24-03-04 20:10:05 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 12.0 / BootSlot B[0m
[32m24-03-04 20:10:06 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository[0m
[32m24-03-04 20:10:06 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository[0m
[32m24-03-04 20:10:06 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/8ce4abae repository[0m
[32m24-03-04 20:10:06 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository[0m
[32m24-03-04 20:10:07 INFO (MainThread) [supervisor.store] Loading add-ons from store: 80 all - 80 new - 0 remove[0m
[32m24-03-04 20:10:07 INFO (MainThread) [supervisor.addons.manager] Found 4 installed add-ons[0m
[32m24-03-04 20:10:07 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-ssh with version 9.9.0[0m
[32m24-03-04 20:10:07 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.8.0[0m
[32m24-03-04 20:10:07 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mosquitto with version 6.4.0[0m
[32m24-03-04 20:10:07 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/egguy/grott-beta/amd64 with version 0.1.5[0m
[32m24-03-04 20:10:07 INFO (MainThread) [supervisor.backups.manager] Found 75 backup files[0m
[... 37 lines cut ...]
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.host.services] Updating service information[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.host.network] Updating local network information[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information[0m
[32m24-03-04 20:12:24 INFO (MainThread) [supervisor.host.manager] Host information reload completed[0m
```

### Anything in the Host logs that might be useful for us?

```txt
11.5 shutdown works with no issues, NUC turns of
```

jpearn, 2024-03-05:
I have  similar yet different scenario with DN2820FYKH whereby when a full reboot is issued, the PC halts but doesn't shutdown or reboot. Fine on 11.5 but 12.0 has the issue

sairon, 2024-03-05:
Do you have any USB peripherals connected? See #3210 for a very similar case.

muhkuh666, 2024-03-05:
Removed USB, still not working

muhkuh666, 2024-03-07:
Will there be a fix for this issues?

bharvey88, 2024-03-09:
additional info: reboot does not fully reboot and come back online. still able to ping and able to access my SMB share so the samba addon is running and supervisor is running.

sairon, 2024-03-11:
@muhkuh666 So far we were not able to reproduce the issue and your report lacks any information that could be useful for tracing it. What hardware are you running on? It's also worth checking all the "wake on ..." and idle state features in BIOS settings, as suggested in the help articles link in the other issue.

muhkuh666, 2024-03-15:
It is an AWOW AL34 Mini PC.

helppp, 2024-05-10:
Did you check BIOS settings? Maybe reboot after power failure is enabled and causes this

closed 2024-08-17 by github-actions[bot]