home-assistant/operating-system #3227: Odroid-N2+ Does Not Boot With 11.5 or 12.0

scoobydrvr, 2024-03-06:
### Describe the issue you are experiencing

Clean install by writing image to eMMC card. Device does not boot to HA OS with version 11.5 or 12.0. If 11.4 is flashed, device will boot and behave as expected. Updating to 11.5 causes device to fail to boot. Further information - https://github.com/home-assistant/operating-system/discussions/3226

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Flash 11.5 or 12.0 image to eMMC and attempt to boot
OR
1. Flash 11.4 image to eMMC and allow HA OS to update to 11.5

### Anything in the Supervisor logs that might be useful for us?

```txt
Unable to access after update
```

### Anything in the Host logs that might be useful for us?

```txt
Unable to access after update
```

### Additional information

Noticed that there is a kernel change between 11.4 and 11.5. Could this be the root cause?

agners, 2024-03-07:
With HAOS 11.5 we updated the U-Boot bootloader to 2024.01, so this is probably bootloader related.

FWIW, my ODROID-N2 test device here boots fine with HAOS 11.5 and 12.0, so this must be related to your particular unit.

Do you have access to the serial console (e.g. using the [USB-UART adapter from Hardkernel](https://www.hardkernel.com/shop/usb-uart-2-module-kit-copy/)) to observe the output at boot time?

scoobydrvr, 2024-03-07:
Unfortunately, no. I feel like I was lucky to have bought the eMMC writer just in case. For what it's worth, this is a brand new unit and it sounds like Hardkernel has changed something in their file loads at the factory. Would there be anything I could help provide by flashing back to Ubuntu and stopping during Petitboot or files I could share?

scoobydrvr, 2024-03-14:
@agners 
I purchased a USB serial module and got the following the terminal

```
U-Boot 2024.01 (Mar 13 2024 - 12:36:37 +0000) odroid-n2/n2-plus

Model: Hardkernel ODROID-N2
SoC:   Amlogic Meson G12B (S922X) Revision 29:c (40:2)
DRAM:  1 GiB (effective 3.8 GiB)
Core:  400 devices, 30 uclasses, devicetree: separate
MMC:   sd@ffe05000: 0, mmc@ffe07000: 1
Loading Environment from nowhere... OK
In:    usbkbd,serial
Out:   vidconsole,serial
Err:   vidconsole,serial
Board variant: n2-plus
Net:   eth0: ethernet@ff3f0000
Card did not respond to voltage select! : -110
unable to select a mode : -5
MMC Device 2 not found
no mmc device at slot 2
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... 3 USB Device(s) found
usb - USB sub-system
```
However, moving back to 11.4 the following is spat out
```
U-Boot 2023.01 (Jan 09 2024 - 11:00:06 +0000) odroid-n2/n2-plus

Model: Hardkernel ODROID-N2
SoC:   Amlogic Meson G12B (S922X) Revision 29:c (40:2)
DRAM:  1 GiB (effective 3.8 GiB)
Core:  387 devices, 27 uclasses, devicetree: separate
MMC:   sd@ffe05000: 0, mmc@ffe07000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Board variant: n2-plus
Net:   eth0: ethernet@ff3f0000
Card did not respond to voltage select! : -110
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
3491 bytes read in 2 ms (1.7 MiB/s)
## Executing script at 08000000
switch to partitions #0, OK
mmc1(part 0) is current device
loading env...

```

scoobydrvr, 2024-03-22:
Looks like there is a similar issue with 12.1 maybe

https://github.com/home-assistant/operating-system/issues/3250

piercy, 2024-03-25:
@scoobydrvr I see you linked my post, I actually got a similar output as you today. Ive been trying all sorts to get it working but having no luck. going to try 11.4 now and see if it works.

Here's what I  managed to get on output , looks very similar to what you are seeing. 

[image]

scoobydrvr, 2024-03-25:
What size is your MMC? I remembered I have a 16Gb card from an old Odroid device. I just threw 12.1 on it an, low and behold, the thing boots up! I had hoped that society had moved past storage device limits but it seems to somehow have an effect here...

piercy, 2024-03-25:
> What size is your MMC? I remembered I have a 16Gb card from an old Odroid device. I just threw 12.1 on it an, low and behold, the thing boots up! I had hoped that society had moved past storage device limits but it seems to somehow have an effect here...

It's 32Gb, I just found another thread with a similar issue.  

What I noticed in your logs was that when its successful its booting from MMC 1, but when it fails its booting from MMC 2. Mine always seems to try boot from MMC 2 (hence its always failing).

I don't know if there is a way to force this, but feels like its trying to boot the wrong device.

piercy, 2024-03-25:
Not sure it matters but I also noticed my mmc card revision is 0.5, a lot of the pictures ive seen show 0.4.  So wonder if the new revision is having issues or something.   

I might have to install with SD card as right now I have no automation running, so my house is extra dumb.

scoobydrvr, 2024-03-25:
> Not sure it matters but I also noticed my mmc card revision is 0.5, a lot of the pictures ive seen show 0.4. So wonder if the new revision is having issues or something.
> 
> I might have to install with SD card as right now I have no automation running, so my house is extra dumb.

My "non-functional" 32Gb is also a rev 0.5 hmmm...

scoobydrvr, 2024-03-25:
Maybe someone with more intimate knowledge can sort out how v0.4 and v0.5 differentiate and why v0.5 doesn't seem to be working?

https://wiki.odroid.com/accessory/emmc/reference_chart#orange_emmc_module

agners, 2024-04-02:
I have at least one eMMC module with version v0.5 and it seems to work here, so I don't think it is the PCB revision.

What PCB colors is yours? it seems that PCB color gives some indication what eMMC chip is on the PCB. I do have orange PCBs here.

piercy, 2024-04-02:
@agners Mine is an orange module

scoobydrvr, 2024-04-02:
> I have at least one eMMC module with version v0.5 and it seems to work here, so I don't think it is the PCB revision.
> 
> What PCB colors is yours? it seems that PCB color gives some indication what eMMC chip is on the PCB. I do have orange PCBs here.

Both of mine, the v0.4 16Gb and v0.5 32Gb, are orange cards

peter-dolkens, 2024-04-04:
I actually have an old model Home Assistant Blue which I based my recommendation to piercy on when he was selecting hardware.

I went through the fresh install process just days before piercy and had no issues

It's currently deployed 2 hours away, so I'll have to wait a few days before I can get any covers off to get photos / part numbers for comparison

BarBaar44, 2024-04-08:
Happy to see this one, as there's nothing wrong my my board and/or eMMC after all :)

https://github.com/home-assistant/operating-system/issues/3150#issuecomment-2031611953

I can confirm I was using an orange 32gb v0.5 eMMc

DavidRSullivan, 2024-04-10:
> Since it took about a week and a half to get the delivery from Korea, would something like [this work](https://a.co/d/0cZK4fN), provided the jumpers can be made to fit inside the header (and the appropriate TTL level is selected)?

Did this work?

TibsD, 2024-04-14:
Any solution for this? I am having the same problem wit my N2+

I had a corrupted install of HA OS on an 32Gb eMMC card and tried reflashing the HA OS to the eMMC but do not see a blue light during boot.

Is it possible that the partitions created by the flashing of HA create a problem? How can I "reset" the eMMC card so it is completely empty and has no partitions?

DavidRSullivan, 2024-04-16:
My problem was the SD card, and after purchasing a New ODROID N2+, everything is fine.  I am using a 256GB eMMC version .5 without any issue.

TibsD, 2024-04-16:
Weird issue. 
Mine just booted up without any problems after leaving it disconnected from power for a full night.
Connected it up this morning and it booted up right away.

piercy, 2024-04-16:
> Weird issue. Mine just booted up without any problems after leaving it disconnected from power for a full night. Connected it up this morning and it booted up right away.

I had mine boot once too, figured it had been resolved by an update (i'd tried a different versions). When i then shut it down to move it to another room, it didn't boot again.  

It certainly is a weird issue, seems like its starting to happen more often now though.  Hopefully more examples can help diagnose and resolve it .

piercy, 2024-04-16:
> My problem was the SD card, and after purchasing a New ODROID N2+, everything is fine. I am using a 256GB eMMC version .5 without any issue.

While i could potentially do this, the fact I can boot into ubuntu fine suggest thats it isn't a problem with my eMMC or the N2+. I am currently using SD card rather that eMMC but hoping a solution comes along soon.

TibsD, 2024-04-16:
> I had mine boot once too, figured it had been resolved by an update (i'd tried a different versions). When i then shut it down to move it to another room, it didn't boot again.
>
> It certainly is a [...]

And if you wait for a night or so and then try starting it up?

agners, 2024-04-23:
I think with #3319 we might have a solution. I've created a development build with that change specifically, it can be downloaded from this GitHub Action run: https://github.com/home-assistant/operating-system/actions/runs/8741265657 (image named `haos_odroid-n2-12.3.dev1713456781.img.xz`)

Can someone affected by the problem test that build and report if it indeed fixes the problem for you?

piercy, 2024-04-23:
@agners yes, will test it shortly 👍

agners, 2024-04-23:
@piercy can you access/download the artifacts? I just realized, at least when logged out, it is not possible. Not sure if this is tied to my account.

piercy, 2024-04-23:
@agners yes already downloaded it :)

agners, 2024-04-23:
Probably obvious, but just to be sure: The artifacts are zipped again (.zip). Make sure to unzip first, to get to the tar.xz (which depending on the tool, might need another unxz, as usual).

piercy, 2024-04-23:
@agners sorry took a bit longer than expected.. work got in the way.

Anyway, it booted up first time.  Blue boot lights straight away and minute or so later I had the setup screen. Attempting to restore my backup now, and then will cycle it a few times to see how it does

sairon, 2024-06-18:
I'm reopening this issue because now I got my hands on a Kingston eMMC and I realized we might have two different issues:

1. Kingston eMMC modules that never worked correctly (the card correctly initializes only one in many boots), until we introduced a patch in #3319.
2. However, that might have triggered #3351, where the faster frequency causes another regression, or actually increses its incidence. The root cause seems to be the update of U-Boot. As a workarond we have downgraded U-Boot to pre-11.5 state in #3409 but that's not a long-term solution, as it also breaks Kingston again. If true, it means that the above dev build, and the 12.3 release could never work correctly for @scoobydrvr and others that **do not** have a Kingston eMMC.

I'd like to ask everyone here to provide us with as many details about the modules used as possible (ideally with the photos of the eMMC module with the sticker removed). Eventually, if you're located in the EU, I could arrange shipping you a new eMMC module and sourcing the faulty one from you - please get in touch with me if you are willing to help with that.

piercy, 2024-06-18:
@sairon See below. Since the update above I have been working completely fine, very happy with everything.

```
Kingston
2401076-001.A006-A  (it could be 00G-A at the end, hard to tell)
2338  4014187
EMMC32G-PJ30

4014187.09
Taiwan
````

[image]

sairon, 2024-06-18:
@piercy Thanks, that's the same model as I have here. Unless I'm missing something very obvious, that **only works** with the 12.3 release (which we pulled back shortly after publishing it). Very likely upgrading to 12.4 release will soft-brick it. If you're seeing something different, let's keep it in #3250.

In any case, be warned about the next release, we currently had to treat Kingston eMMCs as unsupported as they're not that widespread any only a few of you managed to install HAOS on them.

roblamoreaux, 2024-06-18:
I just saw this thread. 

A couple weeks ago my N2+ stopped working after booting to the point is says "Loading Kernel..." on the hdmi screen, Is this the same symptom?

I flashed a new image on the spare eMMC and it did the same thing. I didn't have time to deal with it and wanted more RAM so I replaced it with a Raspberry pi 5 8GB. I have the backup files for it on my NAS and can try to pull the log files off it if it will help. 
I will try to flash the spare emmc with the latest image containing OS 12.4 to see if it fixes the issue for me. While I am enjoying the better [performance of the Pi 5 with NVME drive it would be nice to have the Odroid available as a backup I could get going quickly.

Rob

arjoh, 2024-06-19:
I have noticed that updating to certain versions renders the eMMC unbootable as well.
Also:
- Petitboot menu crashes with a "bad" flashed eMMC
- Sometimes (but not all the time) a longer power cycle helps. So leave it of the power for about a minute it will boot again.
- It seems to matter if I use balenaEtcher or passmark ImageUSB to flash. The latter seems to yield better results.
- Flashing an older image and restoring a full backup works. But I'm reluctant now to update OS, because this is my "production" unit.

Having tried so many things, it's hard to give a coherent summary of everything I've done and what works and what doesn't.
Flashing an older image and restoring a full backup always works, so I'm certainly willing to try out a few things and report back if that helps fixing this issue. Just tell me what would be helpful.

smarthomefamilyverrips, 2024-06-21:
@all Anyone successful did update to 12.4 on Odroid N2+?

arjoh, 2024-06-21:
> @all Anyone successful did update to 12.4 on Odroid N2+? I am on 12.2 and honestly am a bit afraid to do the update to 12.4 😳

Would not boot for me.

smarthomefamilyverrips, 2024-06-21:
> > @ALL Anyone successful did update to 12.4 on Odroid N2+? I am on 12.2 and honestly am a bit afraid to do the update to 12.4 😳
> 
> Would not boot for me.

@arjoh you where already on 12.2?

piercy, 2024-06-21:
> @ALL Anyone successful did update to 12.4 on Odroid N2+? I am on 12.2 and honestly am a bit afraid to do the update to 12.4 😳

I haven't tried yet. Based on sairons comments my emmc isn't supported. Which sucks because I don't know how we're supposed to tell when we just order a emmc and get what we get. Without lifting the label , I didn't even know it was a Kingston one. 

At the moment I'll be staying on 12.3 dev build ,  but it is quite worrying

arjoh, 2024-06-21:
> > > @ALL Anyone successful did update to 12.4 on Odroid N2+? I am on 12.2 and honestly am a bit afraid to do the update to 12.4 😳
> 
> > 
> 
> > Would not boot for me.
> 
> 
> 
> @arjoh you where already on 12.2?

I have been restoring a lot lately so I'm not 100% sure but I think I recall updating from 12.2 to 12.4 yes.

smarthomefamilyverrips, 2024-06-21:
> > > > @ALL Anyone successful did update to 12.4 on Odroid N2+? I am on 12.2 and honestly am a bit afraid to do the update to 12.4 😳
> > 
> > 
> > > 
> > 
> > 
> > > Would not boot for me.
> > 
> > 
> > @arjoh you where already on 12.2?
> 
> I have been restoring a lot lately so I'm not 100% sure but I think I recall updating from 12.2 to 12.4 yes.

@arjoh you used the install function in UI when it showed you update was available? Or you flashing manually? I am on a HA Blue and never did do a OS update manually and would have no idea what to do if not boots up anymore, so that is why I am holding back to do update. 🫣

smarthomefamilyverrips, 2024-06-21:
> I haven't tried yet. Based on sairons comments my emmc isn't supported. Which sucks because I don't know how we're supposed to tell when we just order a emmc and get what we get. Without lifting the [...]

@piercy I also really have no idea what I should do, as stated in my other comment I normally just chose install in UI whenever I got prompted a update is available, 12.3 was already pulled again before I got the chance to push install, what feels like I was lucky, but now as said I not know if should push install on the 12.4 install, I really would not know what to do when not will boot as said I have the HA blue what already came pre-installed with HA so I never loaded/flashed myself anything. 🙄

smarthomefamilyverrips, 2024-06-21:
Is it difficult to restore in case my HA blue will not boot anymore?

peter-dolkens, 2024-06-22:
If you bought HA Blue, I believe you'll have older stock of the eMMC module
and should be ok

smarthomefamilyverrips, 2024-06-22:
> If you bought HA Blue, I believe you'll have older stock of the eMMC module and should be ok
> […](#)

@peter-dolkens I have the HA blue for about 2.5 years and came with 128 GB eMMC, so I also assume it is not the newest type of eMMC.... I think just will need to gather some courage, cross my fingers and bite the bullet in the end, normally I like to keep my install up to date.

smarthomefamilyverrips, 2024-06-24:
@all just to share, I managed to successful update my Home Assistant Blue from 12.2 to 12.4, all seems to work fine. I not did flash the eMCC , I just used the install option on the update notification within the HA UI.

ajfriesen, 2024-06-24:
> I'm reopening this issue because now I got my hands on a Kingston eMMC and I realized we might have two different issues:
> 
> 1. Kingston eMMC modules that never worked correctly (the card correctly initializes only one in many boots), until we introduced a patch in [ODROID-N2/C2/C4: Force 40MHz instead of 24MHz as eMMC clock #3319](https://github.com/home-assistant/operating-system/pull/3319). This is tracked in [ODROID-N2 doesn't boot from Kingston eMMC modules #3250](https://github.com/home-assistant/operating-system/issues/3250).
> 2. However, that might have triggered [Upgrade Odroid N2+ to 12.3, now unable to boot. #3351](https://github.com/home-assistant/operating-system/issues/3351), where the faster frequency causes another regression, or actually increses its incidence. The root cause seems to be the update of U-Boot. As a workarond we have downgraded U-Boot to pre-11.5 state in [Temporary revert ODROID-N2 U-Boot to 2023.01 to fix eMMC issues #3409](https://github.com/home-assistant/operating-system/pull/3409) but that's not a long-term solution, as it also breaks Kingston again. If true, it means that the above dev build, and the 12.3 release could never work correctly for @scoobydrvr and others that **do not** have a Kingston eMMC.
> 
> I'd like to ask everyone here to provide us with as many details about the modules used as possible (ideally with the photos of the eMMC module with the sticker removed). Eventually, if you're located in the EU, I could arrange shipping you a new eMMC module and sourcing the faulty one from you - please get in touch with me if you are willing to help with that.

@sairon 
I have a Home Assistant Blue like forever. I did not update yet. Still on 12.2 because I do not have a second device I could run Home Assistant right now without much hassle.

Would those emmcs be affected by this bug?
Would those be helpful for your testing?

[image]

[image]

TibsD, 2024-06-25:
Updating my Odroid n2+ with Hardkernel eMMC caused it to fail on boot. I will try to disconnect the Odroid and let it reboot this evening to, hopefully, solve the issue.

cvasantrao, 2024-06-26:
Hi
I am a similar issue but booting from SDCard after a fresh install. Attaching the log with errors during setup.
https://dpaste.org/YCfqv

TibsD, 2024-06-26:
I reverted to 12.3 and booted without a problem. The problem is apparently related to the 12.4 update.

dparksfl, 2024-07-10:
Just wanted to add my 2 cents...

SD card failed in my oDroid-N2+ and I replaced it with a eMMC from Amazon...
https://www.amazon.com/gp/product/B0C9GRHRT1
[image]

Downloaded and flashed 12.4 with no issue.  Booted, restored from backup.  Everything was fine.  Did some updates and rebooted.
That's when I got the "Card did not respond to a voltage select! : -110" 
Multiple reboots and nothing.

I found the thread that suggested downgrading to 12.2.  Flashed, restored, did updates (except for the 12.4 update), and rebooted.  Got the same error.  Unplugged it 10 minutes while I flashed a SD card - planning to abandon this eMMC.  When I got back to the oDroid, powered it up for kicks and it booted fine.

I know I'm not really providing any additional info here, but thought I'd share what happened, as well as the pic of my eMMC.  

-David

hannesbe, 2024-07-13:
> I'd like to ask everyone here to provide us with as many details about the modules used as possible (ideally with the photos of the eMMC module with the sticker removed). Eventually, if you're [...]

Experiencing this issue with 2 newly purchased eMMCs with any version of HAOS.

[image]
[image]

lugaru90, 2024-08-08:
Hi, ive got a Kingston with the same Problem:
2401122-001 A00G-A
24Q4 4059794
@sairon If your offer is still available, i would be interested in exchanging cards. Im based in Germany.
[image]

smarthomefamilyverrips, 2024-08-14:
#3508 

@sairon will the above mentioned bump affect or break Home Assistant Blue Devices? Or where HA Blue Devices never sold with incompatible eMMC's and this latest update from 12.4 to 13.0 should just work fine?

smarthomefamilyverrips, 2024-08-14:
> @lugaru90 There's another issue for Kingston modules, unfortunately there's no resolution for that yet, as it's probably a SoC issue that it's not able to work with them reliably.
> 
> @smarthomefamilyverrips No, Blue should be fine. The issues with other eMMCs started to appear after we changed the clock rate to 40 MHz in order to fix compatibility with Kingston. That's now reverted as well.

@sairon OK thank you for your quick response, then I know I can safely update to 13.0 👍

lugaru90, 2024-08-14:
> @lugaru90 There's another issue for Kingston modules, unfortunately there's no resolution for that yet, as it's probably a SoC issue that it's not able to work with them reliably. [...]

Thanks, any idea where i could get a working Chip from a vendor based in Germny (or at least Europe)?

smarthomefamilyverrips, 2024-08-14:
@lugaru90 or this one

https://www.ebay.com/itm/115087306854?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=1sxocBENSii&sssrc=4429486&ssuid=4d9238EiSrW&var=&widget_ver=artemis&media=COPY

piercy, 2024-08-14:
@smarthomefamilyverrips the problem is that the pictures do not match what you get, and there is no option to chose your brand on most of these websites.  The picture of the one I ordered isn't Kingston, but I still received a Kingston.  Upon contacting the company they gave me some nonsense about they test everything before shipped etc etc.  Which doesn't mean its tested with HAOS compatibility, just that they ran some tests. I wanted them to guarantee they wouldn't send me another Kingston, but they wouldn't guarantee it.

At this point I might as well just name them, it was odroid.co.uk.    They have since added this little bracket next to it, so looks like theyre doubling down on it.  Such a crap experience as everyone is just blaming each other. 

[image]

piercy, 2024-08-14:
I would encourage you guys to use this issue for kingston issues, as maybe once enough people have the problem we will start to get somewhere.   Currently we haven't reached critical mass, but I suspect this emmc's aren't going away, so its only a matter of time before it becomes more widespread 

https://github.com/home-assistant/operating-system/issues/3250

smarthomefamilyverrips, 2024-08-14:
@piercy maybe you can try the ebay one, my experience with ebay sellers are that they are willing to send you pictures of the actual product you receive. I also can not guarantee nothing, just trying to help out :)

piercy, 2024-08-14:
> EDIT: maybe also is a idea to move to a other platform as Odroid N2, or take a plug&play HA Green

@smarthomefamilyverrips Yeah realistically, I don't have much faith or trust in HA at the moment. I realize it isn't entirely down to them, but I am stuck between everyone blaming each other.  Odroid seemed like a sensible purchase as it follows HA Blue but now I have this issue.   So purchasing another HA device (ie, HA Green), feels like a "fool me once" style thing.   I'd basically be spending another £86 on top of the ~£100 ive already spent.

smarthomefamilyverrips, 2024-08-14:
> @smarthomefamilyverrips Yeah realistically, I don't have much faith or trust in HA at the moment. I realize it isn't entirely down to them, but I am stuck between everyone blaming each other. Odroid [...]

@piercy sure I kind of understand you. I purchased a HA Blue few years ago and so far in general all worked for me as expected. Sure now and then needed to tinker a bit or skip a update now and then because it would break a integration or something but think partly you are unlucky getting a Kingston eMMC and partly the Kingston problems are not documented good enough. 

But putting all this aside HA is still a great product to my opinion and if you still have the energy and money for it I would say not give up yet... to my opinion as said being you looked at the HA Blue the HA Green is a good out of the box alternative, if I ever will need to replace my HA blue surely I will go for whatever plug&play solution that is offered by the HA-team at that moment.

[8 more replies]

commit eab292f4 (2024-04-18): DROID-N2/C2/C4: Force 40MHz instead of 24MHz as eMMC clock

It seems that forcing 24MHz clocks is problematic for newer 32GB
Kingston based eMMC modules on ODROID-N2(+). Use what downstream
U-Boot is using as f_max, which is 40MHz.

Fixes: #3227
commit 865d6404 (2024-04-18): ODROID-N2/C2/C4: Force 40MHz instead of 24MHz as eMMC clock

It seems that forcing 24MHz clocks is problematic for newer 32GB
Kingston based eMMC modules on ODROID-N2(+). Use what downstream
U-Boot is using as f_max, which is 40MHz.

Fixes: #3227
PR #3319 (2024-04-24): ODROID-N2/C2/C4: Force 40MHz instead of 24MHz as eMMC clock
It seems that forcing 24MHz clocks is problematic for newer 32GB Kingston based eMMC modules on ODROID-N2(+). Use what downstream U-Boot is using as f_max, which is 40MHz.

Fixes: #3227
commit e1b11540 (2024-04-24): ODROID-N2/C2/C4: Force 40MHz instead of 24MHz as eMMC clock (#3319)

It seems that forcing 24MHz clocks is problematic for newer 32GB
Kingston based eMMC modules on ODROID-N2(+). Use what downstream
U-Boot is using as f_max, which is 40MHz.

Fixes: #3227
closed 2024-11-22 by sairon