home-assistant/operating-system #3228: Rasberry Pi 5 HAOS 12.0 NVMe drive

JaCh0o, 2024-03-06:
### Describe the issue you are experiencing

System is stucking in boot loop after reboot. It works properly if system is shut down/powered off then cold booted only. Already tried to flash HAOS thru rpi imager and as a disk iamge downloaded directly from github.  It doesnt matter is there added in config file "dtparam=pciex1" and "dtparam=pciex1_gen=3" or not (however it enables faster disk transfers - up to 850+ MB/s). Rpi5 firmware is up to date - February 16th, 2024.

When rpi is turned of or power cable is taken off then it is plugged back/turned on system boots properly.
When under Settings is forced hard reboot (not HAOS only) it stucks in pre system screen. Same thing happens when reboot command is called from SSH & Terminal Addon.

Boot loops look like this:
PCIe timeout: 0x0001e08f
Failed to open device: 'nvme'
PCIe timeout: 0x0001e08f
Failed to open device: 'nvme'
Boot mode: USB-MSD (04) order f1
USB MSD stopped. Timeout: 25 seconds
Boot mode: SD (01) order f
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: RESTART (0f) order 0
Boot mode NVME (06) order f14
PCIe timeout: 0x0001e08f
...

My set up:
1) Rasberry Pi 5 8GB
2) Genuine Rasbbery Pi power adapter
3) Genuine Rasberry Case
4) Genuine Rasberry Active Cooler
5) Pineberry HatDrive! Top
6) WD SN530 NVMe Gen3 
7) No SD card

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Flash NVMe drive with current HAOS version (in my case plugged as USB drive into Windows/Linux
2. Insert drive to HatDrive and connect it to rpi
3. Plug monitor, ethernet and power cable to rpi
4. Wait until system fully boots
5. Enter inital setup thru app/website
6. Enter Settings - three dots symbol - reboot home assistant - advanced - reboot whole system

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[23:44:34] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
24-03-06 23:44:35 INFO (MainThread) [__main__] Initializing Supervisor setup
24-03-07 00:44:35 INFO (MainThread) [supervisor.bootstrap] Setting up coresys for machine: raspberrypi5-64
24-03-07 00:44:35 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/aarch64-hassio-supervisor with version 2024.02.1
24-03-07 00:44:35 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
24-03-07 00:44:35 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-03-07 00:44:35 INFO (MainThread) [__main__] Setting up Supervisor
24-03-07 00:44:35 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
24-03-07 00:44:35 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
[... 235 lines cut ...]
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 6 marca 2024 23:09
-- | --
current_recorder_run | 7 marca 2024 00:44
estimated_db_size | 0.24 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

agners, 2024-03-07:
This looks like the Raspberry Pi EEPROM is not able to enumerate your NVMe on reboot.
```
Boot loops look like this:
PCIe timeout: 0x0001e08f
Failed to open device: 'nvme'
PCIe timeout: 0x0001e08f
Failed to open device: 'nvme'
Boot mode: USB-MSD (04) order f1
USB MSD stopped. Timeout: 25 seconds
Boot mode: SD (01) order f
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: RESTART (0f) order 0
Boot mode NVME (06) order f14
PCIe timeout: 0x0001e08f
```

What Pi Hat do you use for the NVMe? Can you reproduce this with Raspberry Pi OS?

I don't think we can do much here from Home Assistant OS side, as the boot is stuck in the Raspberry Pi firmware. I'd suggest to test Raspberry Pi OS just to confirm if this is indeed a firmware issue and report this to the Raspberry Pi folks at https://github.com/raspberrypi/firmware.

JaCh0o, 2024-03-13:
Sorry for not responding with additional data but for now I've changed case and hat from official to argonem one V3 nvme and after update to 2024.3.0 this issue seems to be resolved.

closed 2024-03-13 by JaCh0o