home-assistant/operating-system #3229: Rasberry Pi 5 HAOS 12.0 - Unable to start system without monitor plugged

JaCh0o, 2024-03-06:
### Describe the issue you are experiencing

System won't boot without any display/monitor plugged to rpi. After proper boot monitor can be unplugged.

My set up:
Rasberry Pi 5 8GB
Genuine Rasbbery Pi power adapter
Genuine Rasberry Case
Genuine Rasberry Active Cooler
Pineberry HatDrive! Top
WD SN530 NVMe Gen3
No SD card

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Flash NVMe drive with current HAOS version (in my case plugged as USB drive into Windows/Linux
2. Insert drive to HatDrive and connect it to rpi
3. Plug monitor, ethernet and power cable to rpi
4. Wait until system fully boots
5. Enter inital setup thru app/website
6. Shut down thru Settings - three dots symbol - reboot home assistant - advanced - turn off or unplug power cable
7. Unplug monitor
8. Power on unit

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[23:44:34] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
24-03-06 23:44:35 INFO (MainThread) [__main__] Initializing Supervisor setup
24-03-07 00:44:35 INFO (MainThread) [supervisor.bootstrap] Setting up coresys for machine: raspberrypi5-64
24-03-07 00:44:35 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/aarch64-hassio-supervisor with version 2024.02.1
24-03-07 00:44:35 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
24-03-07 00:44:35 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-03-07 00:44:35 INFO (MainThread) [__main__] Setting up Supervisor
24-03-07 00:44:35 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
24-03-07 00:44:35 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
24-03-07 00:44:35 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
24-03-07 00:44:35 INFO (MainThread) [supervisor.host.services] Updating service information
24-03-07 00:44:35 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
24-03-07 00:44:35 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
24-03-07 00:44:35 INFO (MainThread) [supervisor.host.network] Updating local network information
24-03-07 00:44:35 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
24-03-07 00:44:35 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
24-03-07 00:44:35 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
24-03-07 00:44:36 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2024.02.1
24-03-07 00:44:36 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
[... 219 lines cut ...]
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 6 marca 2024 23:09
-- | --
current_recorder_run | 7 marca 2024 00:44
estimated_db_size | 0.24 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

agners, 2024-03-07:
I actually use the Pineberry HatDrive! Bottom on my end with the Raspberry Pi 5. However, my unit works just fine without monitor plugged in.

How do you determine "not booting"? What do the LEDs show?

Do you use an official power supply? Backfeeding through HDMI might change the behavior power wise...

JaCh0o, 2024-03-07:
Yes, as I stated in setup list I use official raspberry power adapter. To be honest I determine this thru that after cold start official active cooler goes loud = 100% and as soon as system start booting it goes almost silent. When I unplug monitor and then cold boot rpi it just goes loud and never goes silent, haos is not loading, there is no response from local website. Same as when there is an error with boot order (on pre system boot screen) = fan is loud and system doesn't start.

agners, 2024-03-07:
Hm, I wonder where in the boot process the system hangs exactly. What happens when you plug in the monitor in that state, does the system continue to boot? If not, what is the screen showing?

Can you reproduce this when booting from SD card? How about Raspberry Pi OS?

JaCh0o, 2024-03-07:
No screen after pluging monitor. So I don't know what exactly it show up is there any place where logs from this state lands?
About raspberry os from Sdcard will check it tomorrow probably.

closed 2024-03-13 by JaCh0o