home-assistant/operating-system #3230: Can't recover from backup

LordArrin, 2024-03-07:
### Describe the issue you are experiencing

Регистратор: homeassistant.util.package
Источник: util/package.py:122
Первое сообщение: 10:39:30 (2 сообщений)
Последнее сообщение: 10:40:46

Unable to install package tinytuya==1.13.1: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245b67680>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245bb0440>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245bb05f0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245bb0800>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245bb0a10>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb247f7d6a0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245b70b60>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245b670b0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245cd7d40>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fb245bb0740>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ ERROR: Could not find a version that satisfies the requirement tinytuya==1.13.1 (from versions: none) ERROR: No matching distribution found for tinytuya==1.13.1
Unable to install package tinytuya==1.13.1: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfeced1d30>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfec240620>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfec2407d0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfec240950>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfec2409b0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfecfbaba0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfeced1d30>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfec3d8fb0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfec234980>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fbfec240c20>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ ERROR: Could not find a version that satisfies the requirement tinytuya==1.13.1 (from versions: none) ERROR: No matching distribution found for tinytuya==1.13.1

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
24-03-07 10:34:13 WARNING (MainThread) [supervisor.homeassistant.websocket] Connection is closed
24-03-07 10:34:27 INFO (MainThread) [supervisor.addons.manager] Phase 'services' stopping 0 add-ons
24-03-07 10:34:27 INFO (MainThread) [supervisor.addons.manager] Phase 'system' stopping 0 add-ons
24-03-07 10:34:27 INFO (MainThread) [supervisor.addons.manager] Phase 'initialize' stopping 0 add-ons
24-03-07 10:34:27 INFO (MainThread) [supervisor.backups.manager] Restore 5f2d4979 starting stage docker_config
24-03-07 10:34:27 INFO (MainThread) [supervisor.backups.manager] Restore 5f2d4979 starting stage folders
24-03-07 10:34:27 INFO (SyncWorker_2) [supervisor.backups.backup] Restore folder share
24-03-07 10:34:27 INFO (SyncWorker_2) [supervisor.backups.backup] Restore folder share done
24-03-07 10:34:27 INFO (SyncWorker_0) [supervisor.backups.backup] Restore folder addons/local
24-03-07 10:34:27 INFO (SyncWorker_0) [supervisor.backups.backup] Restore folder addons/local done
24-03-07 10:34:27 INFO (SyncWorker_2) [supervisor.backups.backup] Restore folder ssl
24-03-07 10:34:27 INFO (SyncWorker_2) [supervisor.backups.backup] Restore folder ssl done
24-03-07 10:34:27 INFO (SyncWorker_1) [supervisor.backups.backup] Restore folder media
24-03-07 10:34:27 INFO (SyncWorker_1) [supervisor.backups.backup] Restore folder media done
24-03-07 10:34:27 INFO (MainThread) [supervisor.backups.manager] Restore 5f2d4979 starting stage home_assistant
24-03-07 10:34:28 INFO (MainThread) [supervisor.homeassistant.module] Restore Home Assistant Core config folder
[... cut ...]
24-03-07 10:44:18 INFO (MainThread) [supervisor.backups.manager] Full-Restore 5f2d4979 done
```

### Anything in the Host logs that might be useful for us?

```txt
Регистратор: homeassistant.components.homeassistant_alerts
Источник: helpers/update_coordinator.py:331
интеграция: Home Assistant Alerts (документация, проблемы)
Первое сообщение: 10:44:18 (1 сообщений)
Последнее сообщение: 10:44:18

Error requesting homeassistant_alerts data: Cannot connect to host alerts.home-assistant.io:443 ssl:default [Try again]
```

agners, 2024-03-07:
It does seem that the restore succeeded? :thinking: 

```
24-03-07 10:44:18 INFO (MainThread) [supervisor.backups.manager] Full-Restore 5f2d4979 done
```

What is the problem exactly? The errors from the Core log look more Core related.

LordArrin, 2024-03-07:
[supervisor_2024-03-07T13-33-25.212Z.log](https://github.com/home-assistant/operating-system/files/14524636/supervisor_2024-03-07T13-33-25.212Z.log)
[home-assistant_2024-03-07T13-33-19.440Z.log.txt](https://github.com/home-assistant/operating-system/files/14524638/home-assistant_2024-03-07T13-33-19.440Z.log.txt)

LordArrin, 2024-03-07:
Plz help, 😭, whole my home Assistant can't boot, all my backups won't work, even oldest one

LordArrin, 2024-03-07:
Регистратор: aiohttp.server
Источник: /usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py:421
Первое сообщение: 17:38:51 (4 сообщений)
Последнее сообщение: 17:48:28

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 350, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 557, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message:
  Invalid method encountered:

    b'\x16\x03\x01\x02'
      ^

LordArrin, 2024-03-07:
Регистратор: homeassistant.components.homeassistant_alerts
Источник: helpers/update_coordinator.py:331
интеграция: Home Assistant Alerts (документация, проблемы)
Первое сообщение: 17:32:44 (1 сообщений)
Последнее сообщение: 17:32:44

Error requesting homeassistant_alerts data: Cannot connect to host alerts.home-assistant.io:443 ssl:default [Try again]

LordArrin, 2024-03-07:
Регистратор: homeassistant.util.package
Источник: util/package.py:122
Первое сообщение: 18:42:40 (5 сообщений)
Последнее сообщение: 18:47:43

Unable to install package tinytuya==1.13.1: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a6f17ec0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a71b55e0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a6374470>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a63747a0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a63749b0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a67de360>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a6f17ec0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a64851f0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a64a0740>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f47a63748f0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ ERROR: Could not find a version that satisfies the requirement tinytuya==1.13.1 (from versions: none) ERROR: No matching distribution found for tinytuya==1.13.1
Unable to install package tinytuya==1.13.1: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fced7ee6900>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fced875cef0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fced73ebfe0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fced7400200>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fced7400470>: Failed to establish a new connection: [Errno -3] Try again')': /simple/tinytuya/ WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fced7f50140>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fced82d7170>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/tinytuya/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=N [... cut ...] ndex/openrgb-python/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8b054013d0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/openrgb-python/ ERROR: Could not find a version that satisfies the requirement openrgb-python==0.2.10 (from versions: none) ERROR: No matching distribution found for openrgb-python==0.2.10

LordArrin, 2024-03-07:
Регистратор: homeassistant.util.package
Источник: util/package.py:122
Первое сообщение: 23:10:38 (1 сообщений)
Последнее сообщение: 23:10:38

Unable to install package aioblescan>=0.2.14: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8258884740>: Failed to establish a new connection: [Errno -3] Try again')': /simple/aioblescan/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82576f40e0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/aioblescan/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82576f4470>: Failed to establish a new connection: [Errno -3] Try again')': /simple/aioblescan/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82576f46e0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/aioblescan/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82576f48f0>: Failed to establish a new connection: [Errno -3] Try again')': /simple/aioblescan/ WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82578dd310>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/aioblescan/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82577e30b0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/aioblescan/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82576d3bf0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/aioblescan/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82576c76b0>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/aioblescan/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f82576f4950>: Failed to establish a new connection: [Errno -3] Try again')': /musllinux-index/aioblescan/ ERROR: Could not find a version that satisfies the requirement aioblescan>=0.2.14 (from versions: none) ERROR: No matching distribution found for aioblescan>=0.2.14

LordArrin, 2024-03-09:
My bad, it's Russia internet related errors.

closed 2024-03-09 by LordArrin

agners, 2024-03-09:
Thanks for the update.