home-assistant/operating-system #3245: HA fresh install freezes on new Raspberri Pi 8G

JohannYellow, 2024-03-13:
### Describe the issue you are experiencing

I have the following versions on a new Pi5 8G with 128G SD card.
Core 2024.3.0, Supervisor 2024.03.0, Operating System 12.0, Frontend 20240306.0

Initially, I just restored my whole HA from a Pi3 1G but it would freeze after about a day or less.  I decided to rather do a clean install and test it without installing any addons or setting up any devices.  
To my surprise, it still freezes but just after a few days this time instead of just 1 day. I looked at the log (attached) but cannot see anything that causes the freeze.

Is this a known issue on the Pi5 - since it is not an official release yet?  How can I help to debug this?
PS: I reported this on the HA forum and was requested to rather report it here.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.1.73-haos-raspi

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Do a clean install of HA on a Pi5 8G on a new SD card.
2. Let it run for a few days until the green LED goes red.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Supervisor
24-03-13 13:29:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-03-13 13:59:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-03-13 14:27:12 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.check] System checks complete
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
24-03-13 14:27:12 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
24-03-13 14:27:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
24-03-13 14:27:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
24-03-13 14:27:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
24-03-13 14:27:15 INFO (MainThread) [supervisor.store] Loading add-ons from store: 77 all - 0 new - 0 remove
24-03-13 14:27:15 INFO (MainThread) [supervisor.store] Loading add-ons from store: 77 all - 0 new - 0 remove
24-03-13 14:29:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-03-13 14:47:12 INFO (MainThread) [supervisor.host.info] Updating local host information
24-03-13 14:47:13 INFO (MainThread) [supervisor.host.services] Updating service information
24-03-13 14:47:13 INFO (MainThread) [supervisor.host.network] Updating local network information
24-03-13 14:47:13 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
24-03-13 14:47:13 INFO (MainThread) [supervisor.host.manager] Host information reload completed
24-03-13 14:59:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-03-13 15:27:12 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
24-03-13 15:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
24-03-13 15:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
24-03-13 15:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
24-03-13 15:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
24-03-13 15:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
24-03-13 15:27:12 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
[... 219 lines cut ...]

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 6, 2024 at 12:11 PM
-- | --
current_recorder_run | March 13, 2024 at 8:27 AM
estimated_db_size | 9.46 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

I'm powering it with a 40W laptop USB-C power supply to ensure current is not an issue.
The SD card is new but I also tried with a different (64G) one and had the same issue.

Landsailor, 2024-03-13:
Are you _sure_ the Pi 5 is seeing full power from your power supply? Connect a monitor and try booting from a standard RPi OS and see if you get the "not capable" warning on boot. 

I've been running HA since it was officially released for the Pi 5, but that was _after_ I got the official power supply and it's been rock solid. However, I started with Pi OS before it was released and had already updated the firmware. 

Hope that helps.

sairon, 2024-03-14:
> I looked at the log (attached) but cannot see anything that causes the freeze.

Actually, there are undervoltage warnings all over the place: `kernel: hwmon hwmon3: Undervoltage detected!`

As Landsailor correctly pointed out above, it's very likely a power supply issue. What adapter are you using to power the RPi?

JohannYellow, 2024-03-14:
I'm using a "Treqa CH-637".  Its specs say it provides 45W PD on the USB-C output (DC 5V 3A, 9V 3A, 12V 3A, 15V 3A, 20V 2.25A).
I saw those "undervoltage" messages but thought they were due to me resetting (power-cycle) the device.  
Why would the power consistently and repeatedly drop when I upgrade from Core12.0 to Core12.1?
I don't have a Pi5 PSU but I have a Pi3 PSU that I can use to test.  I'm not hopeful that it will make a difference though, because I used it initially and also got freeze issues.

JohannYellow, 2024-03-14:
Follow-up note:  I connected it to my Pi3 (5V @ 2.5A) PSU and the update went though.  I did not think that a 45W PSU that can change my laptop will be insufficient for the Pi5, so I humbly apologise.  I will get a Pi5 PSU before continuing with testing.  (Seems that not all USB-PD supplies are created equal.)

sairon, 2024-03-14:
> I'm using a "Treqa CH-637". Its specs say it provides 45W PD on the USB-C output (DC 5V 3A, 9V 3A, 12V 3A, 15V 3A, 20V 2.25A).

Actually, RPi  5 can only use the 5V voltage out of those. If it can't deliver high currents reliably, or because the voltage drop is too high at the end of the cable (e.g. because of thin wires), it can lead to undervoltage for the end device. I see that charger doesn't have a cable permanently attached, maybe you can try using a different higher-quality cable?

>  I saw those "undervoltage" messages but thought they were due to me resetting (power-cycle) the device.

Not sure what you mean by that. It means the voltage is dropping below a certain limit while the device is running, which can lead to unexpected behavior.

>  Why would the power consistently and repeatedly drop when I upgrade from Core12.0 to Core12.1?

IMO it's just a coincidence, there are no changes in the Raspberry Pi kernel between those two versions that could make a difference.

Landsailor, 2024-03-14:
> DC 5V **3A**

There's your problem, @JohannYellow .

Your power supply cannot deliver the 5.1 volts at 5 amps that the Pi 5 requires. I don't know how many peripherals you have hooked up, but my best guess is that your transceivers or additional USB hardware is actually pulling close enough to 600mA to be an issue. (I think that's the default USB power availability when using an "unknown" power supply.)

Get one of the official power supplies and your problems (with this issue, anyway) will disappear.

https://www.pishop.co.za/store/rpi-power/raspberry-pi-official-27w-usb-c-power-supply-black-51v-5a-psu-type-eufor-pi5

JohannYellow, 2024-03-15:
Thank you @Landsailor and @sairon for your feedback and thank you for your work on HA.  I've ordered that PSU and will see if I get any further problems. 
Just an idea:  It would be nice if there was some "self-test" image for the Pi to strain-test all aspects of it, including RAM, SD card, etc.  It could, for example, put the CPU under max load and see if a PSU dip occurs.

robertpaananen, 2024-04-06:
@JohannYellow Was that the problem? I am running my Pi 5 with 5v 4.5A and get some warnings about undervoltage and also experience the same "crashes" where after a day or so the led turns red. Just wondering if I should invest in a better power supply or if it is something else.

JohannYellow, 2024-04-06:
@robertpaananen Yes - I bought an official Pi5 PSU and the problem went away and never occurred again since.  And this is 3 weeks later.

closed 2024-07-13 by github-actions[bot]