home-assistant/operating-system #3247: external data disk gets disabled on HAOS update in UEFI (arm64) VM - machine_id found empty

claplace, 2024-03-13:
### Describe the issue you are experiencing

After the recent update to 12.1, the system got stuck during boot. This time I took time to fetch the journalctl and dmesg outputs.

My external data disk was labelled "hassos-data-dis". I was able to change it to "hassos-data" and reboot successfully.

Looking at existing issues, I thing the root cause is that the boot immediatly after the operating system update got an empty machine_id value:

`homeassistant kernel: Kernel command line: BOOT_IMAGE=(hd0,gpt2)/Image root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id= fsck.repair=yes systemd.condition-first-boot=true console=tty1 console=ttyS0 rauc.slot=A`

This caused the `haos-data-disk-detach` script to run:

`homeassistant systemd[1]: Starting HAOS data disk detach...`

and that got the data disk disabled.

It's the second time it happens (since I moved the data on a second disk), and might be related to the update procedure for UEFI systems. I see the new machine_id in the `/mnt/boot/efi/boot/grubenv`.

[journalctl.txt](https://github.com/home-assistant/operating-system/files/14594796/journalctl.txt)
[dmesg.txt](https://github.com/home-assistant/operating-system/files/14594797/dmesg.txt)

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Started 12.0 -> 12.1 update.

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

## System Information

version | core-2024.3.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20-haos
arch | aarch64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4937
Installed Version | 1.34.0
Stage | running
Available Repositories | 1411
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | November 27, 2024 at 7:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | us-east-1-4.ui.nabu.casa
certificate_status | ready
instance_id | d135c865502f446fa7746e274daf1f76
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 125.9 GB
disk_used | 23.7 GB
healthy | true
supported | true
board | generic-aarch64
supervisor_api | ok
version_api | ok
installed_addons | ZeroTier One (0.18.0), Terminal & SSH (9.10.0), File editor (5.8.0), ESPHome (2024.2.2), Z-Wave JS (0.4.5), Matter Server (5.4.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 3
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 6, 2024 at 11:33 PM
-- | --
current_recorder_run | March 13, 2024 at 7:01 PM
estimated_db_size | 268.23 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

chris0607, 2024-03-14:
I suspect that I have the same bug on the RPi5 with SSD, but I can no longer access any logs. In my case, only a reinstallation with the implementation of the backup helped.

agners, 2024-04-02:
@claplace I agree with your analysis, it seems that first boot got detected again. 

What usually is the cause is if a internal SD card as well as a external SD card/disk have a full HAOS installation (with the boot partition etc.) on it. Then the system might write the boot information on a different disk than the boot loader reads things from.

Can you run `lsblk` from the terminal to check what disks and partitions there are?

claplace, 2024-04-02:
Sure :)

```
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda      8:0    0   32G  0 disk 
|-sda1   8:1    0   32M  0 part /mnt/boot
|-sda2   8:2    0   24M  0 part 
|-sda3   8:3    0  256M  0 part /
|-sda4   8:4    0   24M  0 part 
|-sda5   8:5    0  256M  0 part 
|-sda6   8:6    0    8M  0 part 
|-sda7   8:7    0   96M  0 part /var/lib/systemd
|                               /var/lib/bluetooth
|                               /var/lib/NetworkManager
|                               /root/.ssh
|                               /root/.docker
|                               /etc/udev/rules.d
|                               /etc/systemd/timesyncd.conf
|                               /etc/modules-load.d
|                               /etc/modprobe.d
|                               /etc/hosts
|                               /etc/hostname
|                               /etc/dropbear
|                               /etc/NetworkManager/system-connections
|                               /mnt/overlay
`-sda8   8:8    0 31.3G  0 part 
sdb      8:16   0  128G  0 disk 
`-sdb1   8:17   0  128G  0 part /var/log/journal
                                /var/lib/docker
                                /mnt/data
zram0  253:0    0    0B  0 disk 
zram1  253:1    0   32M  0 disk 
zram2  253:2    0   16M  0 disk /tmp
```

claplace, 2024-04-12:
Hi, just a note to tell I've hit the issue again with 12.2 update.

fixed with:
```
e2label /dev/sdb1 hassos-data
reboot
```

chris0607, 2024-04-12:
The update to 12.2 worked on the Pi 5 with PCIe SSD.

agners, 2024-04-12:
> Hi, just a note to tell I've hit the issue again with 12.2 update.

I've tried to reproduce this on generic-aarch64, but wasn't able to: For me the machine ID got saved, and on successive boot it was present in the cmdline.

@claplace can you check the logs of hassos-persists?

```
journalctl -u hassos-persists
```

And check if the GRUB environment is ok?
```
grub-editenv /mnt/boot/EFI/BOOT/grubenv list
```

claplace, 2024-05-09:
There is a new HAOS update! I used the commands above, the `hassos-persists` journal is empty, and as I don't know how to properly ssh into haos, here's a screenshot for the current grub env, before update:

[image]

claplace, 2024-05-09:
I started the update, and went back print the grub env again and again... and suddenly it went empty:

[image]

[image]

claplace, 2024-06-28:
It happened again. So I went to understand how the HAOS update worked, and here's what I think I have understood.

HAOS is using RAUC, and the update is done from a rauc bundle. e.g. https://github.com/home-assistant/operating-system/releases/download/12.4/haos_generic-aarch64-12.4.raucb for the latest release.

Here's the bundle content:
```
$ ls -l
total 227656
-rw-r--r-- 1 cyprien cyprien  33554432 Jun 18 09:57 boot.vfat
-rwxr-xr-x 1 cyprien cyprien      4935 Jun 18 09:57 hook
-rw-r--r-- 1 cyprien cyprien  20287488 Jun 18 09:57 kernel.img
-rw-r--r-- 1 cyprien cyprien       521 Jun 18 09:57 manifest.raucm
-rw-r--r-- 1 cyprien cyprien 211886080 Jun 18 09:57 rootfs.img
```

boot.vfat contains the boot files that will *replace* the existing ones:
```
$ find boot
boot
boot/EFI
boot/EFI/BOOT
boot/EFI/BOOT/grub.cfg
boot/EFI/BOOT/bootaa64.efi
boot/EFI/BOOT/grubenv
boot/cmdline.txt
$ grub-editenv boot/EFI/BOOT/grubenv list
$
```

The grub environment is empty there. Now the bundle `hook` file contains a `install_boot()` function that replaces the existing boot files with the new one, making sure the `*.txt` files are restored (but I don't see any...).

```
        # Backup boot config
        cp -f "${BOOT_MNT}"/*.txt "${BOOT_TMP}/" || true

        cp -rf "${BOOT_NEW}"/* "${BOOT_MNT}/"

        # Restore boot config
        cp -f "${BOOT_TMP}"/*.txt "${BOOT_MNT}/" || true
```

Then I believe rauc is rewriting the grubenv with the new boot order, but it does not know about `MACHINE_ID` key, that does not appear in `https://github.com/rauc/rauc/blob/master/src/bootchooser.c`.

After manually downloading and installing the update bundle `rauc install haos_generic-aarch64-12.4.raucb`, I see that the `grubenv` does not have the `MACHINE_ID` entry:

[image]

A simple line added in the `hook` file could restore it:

[image]

PR #3451 (2024-07-04): Fix rauc hook to restore machine-id in new grub environment
Without a machine-id, the next boot will be considered as a first boot, and any external data disk will be disabled.

Fixes #3247.

[image]

After installing my raucb update bundle and a reboot:

```
Jun 29 16:34:38 homeassistant kernel: Kernel command line: BOOT_IMAGE=(hd0,gpt4)/Image root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id=6c631b4986cb426f9b7860df4ce10629 fsck.repair=yes console=ttyS0 console=tty0 rauc.slot=B
```

The machine-id has been conserved 👍
commit 33d8b538 (2024-07-04): Fix rauc hook to save and restore the grub environment (#3451)

Without a machine-id, the next boot will be considered as a
first boot, and any external data disk will be disabled.

Fixes #3247.
closed 2024-07-04 by sairon