home-assistant/operating-system #3251: QEMU agent command 'guest-fsfreeze-freeze' issue with HAOS 12.1

GrimD, 2024-03-14:
### Describe the issue you are experiencing

With HAOS 12.1 when trying to snapshot the VM running HAOS for backup it failed with the error; "unable to execute QEMU agent command 'guest-fsfreeze-freeze': fsfreeze hook has failed with status 1". Tried shutting down the VM, starting it and trying again but got the same error. The QEMU agent is running correctly in general. after restoring a backup of the VM so it is back to HAOS 12.0 the snapshot and therefore back up run fine again.

The hypervisor is KVM with in the Unraid OS.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Upgrade to 12.1
2.Try to to back up the VM via the VM backup plugin in Unraid
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
I don't think so but here is the out put only a minute or so after the issue occurs;

24-03-14 19:34:02 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
24-03-14 19:34:02 WARNING (MainThread) [supervisor.host.network] Requested to update interface enp1s0 which does not exist or is disabled.
24-03-14 19:34:02 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
24-03-14 19:34:02 INFO (MainThread) [supervisor.mounts.manager] Initializing all user-configured mounts
24-03-14 19:34:02 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
24-03-14 19:34:02 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
24-03-14 19:34:03 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2024.03.1
24-03-14 19:34:03 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
24-03-14 19:34:03 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2024.03.1 - 172.30.32.5
24-03-14 19:34:03 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2024.03.0
24-03-14 19:34:03 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
24-03-14 19:34:03 INFO (MainThread) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2024.03.0 - 172.30.32.3
24-03-14 19:34:03 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
24-03-14 19:34:03 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0
24-03-14 19:34:03 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
24-03-14 19:34:04 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0 - 172.30.32.4
24-03-14 19:34:04 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2023.06.0
24-03-14 19:34:04 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0
24-03-14 19:34:04 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
24-03-14 19:34:04 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0 - Host
24-03-14 19:34:04 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
24-03-14 19:34:04 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/qemux86-64-homeassistant with version 2024.3.1
24-03-14 19:34:04 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 12.1 / BootSlot A
24-03-14 19:34:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
24-03-14 19:34:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
24-03-14 19:34:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/f4f71350 repository
24-03-14 19:34:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
24-03-14 19:34:05 INFO (MainThread) [supervisor.store] Loading add-ons from store: 78 all - 78 new - 0 remove
24-03-14 19:34:05 INFO (MainThread) [supervisor.addons.manager] Found 6 installed add-ons
24-03-14 19:34:05 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/node-red/amd64 with version 17.0.9
24-03-14 19:34:05 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-ssh with version 9.10.0
24-03-14 19:34:05 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.15.0
24-03-14 19:34:05 INFO (MainThread) [supervisor.docker.interface] Attaching to f4f71350/amd64-addon-ewelink_smart_home_slug with version 1.4.3
24-03-14 19:34:05 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio with version 2024.2.2
24-03-14 19:34:05 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/zwave-js-ui/amd64 with version 3.4.1
24-03-14 19:34:05 INFO (MainThread) [supervisor.backups.manager] Found 46 backup files
24-03-14 19:34:06 INFO (MainThread) [supervisor.discovery] Loaded 2 messages
24-03-14 19:34:06 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
24-03-14 19:34:06 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup
24-03-14 19:34:06 INFO (MainThread) [supervisor.resolution.check] System checks complete
[... 265 lines cut ...]
Here is the relevant section from the VM Backup plugin log;
2024-03-14 19:39:54 information: DBHA01 can be found on the system. attempting backup.
2024-03-14 19:39:54 information: creating local DBHA01.xml to work with during backup.
2024-03-14 19:39:54 information: /mnt/user/Backups/Unraid/VMs/DBHA01 exists. continuing.
2024-03-14 19:39:54 information: skip_vm_shutdown is false and use_snapshots is 1. skipping vm shutdown procedure. DBHA01 is running. can_backup_vm set to y.
2024-03-14 19:39:54 information: actually_copy_files is 1.
2024-03-14 19:39:54 information: can_backup_vm flag is y. starting backup of DBHA01 configuration, nvram, and vdisk(s).
sending incremental file list
DBHA01.xml

sent 6,401 bytes received 35 bytes 12,872.00 bytes/sec
total size is 6,294 speedup is 0.98
2024-03-14 19:39:55 information: copy of DBHA01.xml to /mnt/user/Backups/Unraid/VMs/DBHA01/20240314_1936_DBHA01.xml complete.
sending incremental file list
43c3c14f-bbb0-e3b1-af5f-fc81c78df426_VARS-pure-efi-tpm.fd

sent 540,951 bytes received 35 bytes 1,081,972.00 bytes/sec
total size is 540,672 speedup is 1.00
2024-03-14 19:39:55 information: copy of /etc/libvirt/qemu/nvram/43c3c14f-bbb0-e3b1-af5f-fc81c78df426_VARS-pure-efi-tpm.fd to /mnt/user/Backups/Unraid/VMs/DBHA01/20240314_1936_43c3c14f-bbb0-e3b1-af5f-fc81c78df426_VARS-pure-efi-tpm.fd complete.
2024-03-14 19:39:55 information: able to perform snapshot for disk /mnt/user/domains/DBHA01/vdisk1.img on DBHA01. use_snapshots is 1. vm_state is running. vdisk_type is qcow2
2024-03-14 19:39:55 information: qemu agent found. enabling quiesce on snapshot.
error: internal error: unable to execute QEMU agent command 'guest-fsfreeze-freeze': fsfreeze hook has failed with status 1

2024-03-14 19:39:55 failure: snapshot command failed on vdisk1.snap for DBHA01.
2024-03-14 19:39:57 failure: snapshot_fallback is 0. skipping backup for DBHA01 to prevent data loss. no cleanup will be performed for this vm.

artifactdev, 2024-03-15:
Since the new Version I also receive this error in connection to Unraid with VM Backup Plugin for my HAOS VM

Soleima77, 2024-03-15:
After upgrading til 12.1 full snapshot on Synology Virtual Machine Manager is not possible. It can only make  "Crash Consistant" snapshot.

Baxxy13, 2024-03-15:
No problems "snapshotting" the running HA-OS-VM (12.1) with proxmox. 

<details>

<summary>vzdump - log</summary>

```
INFO: starting new backup job: vzdump 105 --node PVE-N100-229 --storage ssd-512gb --compress zstd --notes-template '{{guestname}}' --mode snapshot --remove 0 --notification-mode auto
INFO: Starting Backup of VM 105 (qemu)
INFO: Backup started at 2024-03-15 12:54:08
INFO: status = running
INFO: VM Name: HAOS-Live-VM-211
INFO: include disk 'scsi0' 'local-lvm:vm-105-disk-1' 32G
INFO: include disk 'efidisk0' 'local-lvm:vm-105-disk-0' 4M
INFO: backup mode: snapshot
INFO: ionice priority: 7
INFO: creating vzdump archive '/mnt/pve/ssd-512gb/dump/vzdump-qemu-105-2024_03_15-12_54_08.vma.zst'
INFO: issuing guest-agent 'fs-freeze' command
INFO: issuing guest-agent 'fs-thaw' command
INFO: started backup task 'b19bd217-9adf-4823-96d5-6f9b2180543d'
INFO: resuming VM again
INFO:   2% (982.4 MiB of 32.0 GiB) in 3s, read: 327.5 MiB/s, write: 232.8 MiB/s
INFO:   6% (2.1 GiB of 32.0 GiB) in 6s, read: 373.6 MiB/s, write: 258.8 MiB/s
INFO:  12% (4.0 GiB of 32.0 GiB) in 9s, read: 673.6 MiB/s, write: 200.4 MiB/s
INFO:  16% (5.2 GiB of 32.0 GiB) in 12s, read: 390.6 MiB/s, write: 177.5 MiB/s
INFO:  17% (5.5 GiB of 32.0 GiB) in 15s, read: 113.6 MiB/s, write: 107.3 MiB/s
INFO:  20% (6.5 GiB of 32.0 GiB) in 18s, read: 323.8 MiB/s, write: 130.0 MiB/s
INFO:  21% (7.0 GiB of 32.0 GiB) in 21s, read: 177.6 MiB/s, write: 152.2 MiB/s
INFO:  22% (7.3 GiB of 32.0 GiB) in 24s, read: 122.0 MiB/s, write: 120.3 MiB/s
INFO:  28% (9.0 GiB of 32.0 GiB) in 27s, read: 564.3 MiB/s, write: 107.8 MiB/s
INFO:  29% (9.3 GiB of 32.0 GiB) in 30s, read: 122.0 MiB/s, write: 121.0 MiB/s
INFO:  31% (9.9 GiB of 32.0 GiB) in 33s, read: 200.6 MiB/s, write: 114.4 MiB/s
INFO:  34% (11.0 GiB of 32.0 GiB) in 36s, read: 368.5 MiB/s, write: 99.1 MiB/s
INFO:  37% (12.1 GiB of 32.0 GiB) in 39s, read: 379.0 MiB/s, write: 95.2 MiB/s
INFO:  40% (13.1 GiB of 32.0 GiB) in 42s, read: 318.2 MiB/s, write: 139.4 MiB/s
INFO:  42% (13.5 GiB of 32.0 GiB) in 45s, read: 155.6 MiB/s, write: 136.0 MiB/s
INFO:  43% (14.0 GiB of 32.0 GiB) in 48s, read: 179.1 MiB/s, write: 147.0 MiB/s
INFO:  45% (14.6 GiB of 32.0 GiB) in 51s, read: 182.5 MiB/s, write: 114.3 MiB/s
INFO:  46% (15.0 GiB of 32.0 GiB) in 54s, read: 147.4 MiB/s, write: 120.4 MiB/s
INFO:  48% (15.6 GiB of 32.0 GiB) in 57s, read: 193.0 MiB/s, write: 122.3 MiB/s
INFO:  50% (16.1 GiB of 32.0 GiB) in 1m, read: 173.2 MiB/s, write: 170.1 MiB/s
INFO:  54% (17.5 GiB of 32.0 GiB) in 1m 3s, read: 496.6 MiB/s, write: 126.2 MiB/s
INFO:  56% (17.9 GiB of 32.0 GiB) in 1m 6s, read: 135.3 MiB/s, write: 124.9 MiB/s
INFO:  57% (18.3 GiB of 32.0 GiB) in 1m 10s, read: 92.4 MiB/s, write: 91.8 MiB/s
INFO:  58% (18.7 GiB of 32.0 GiB) in 1m 14s, read: 97.0 MiB/s, write: 94.4 MiB/s
INFO:  59% (19.1 GiB of 32.0 GiB) in 1m 17s, read: 142.6 MiB/s, write: 104.0 MiB/s
INFO:  61% (19.6 GiB of 32.0 GiB) in 1m 20s, read: 187.3 MiB/s, write: 149.2 MiB/s
INFO:  63% (20.5 GiB of 32.0 GiB) in 1m 23s, read: 285.2 MiB/s, write: 120.2 MiB/s
INFO:  65% (20.9 GiB of 32.0 GiB) in 1m 26s, read: 149.8 MiB/s, write: 119.4 MiB/s
INFO:  66% (21.4 GiB of 32.0 GiB) in 1m 29s, read: 166.8 MiB/s, write: 163.0 MiB/s
INFO:  68% (21.9 GiB of 32.0 GiB) in 1m 32s, read: 169.2 MiB/s, write: 165.0 MiB/s
INFO:  71% (23.0 GiB of 32.0 GiB) in 1m 35s, read: 368.5 MiB/s, write: 106.2 MiB/s
INFO:  72% (23.4 GiB of 32.0 GiB) in 1m 38s, read: 130.5 MiB/s, write: 111.9 MiB/s
INFO:  74% (23.9 GiB of 32.0 GiB) in 1m 41s, read: 182.2 MiB/s, write: 110.2 MiB/s
INFO:  76% (24.6 GiB of 32.0 GiB) in 1m 44s, read: 228.0 MiB/s, write: 171.7 MiB/s
INFO:  78% (25.0 GiB of 32.0 GiB) in 1m 47s, read: 154.0 MiB/s, write: 124.6 MiB/s
INFO:  79% (25.3 GiB of 32.0 GiB) in 1m 50s, read: 110.0 MiB/s, write: 108.6 MiB/s
INFO:  80% (25.7 GiB of 32.0 GiB) in 1m 53s, read: 123.2 MiB/s, write: 123.1 MiB/s
INFO:  81% (26.2 GiB of 32.0 GiB) in 1m 56s, read: 186.1 MiB/s, write: 185.7 MiB/s
INFO:  83% (26.8 GiB of 32.0 GiB) in 1m 59s, read: 201.8 MiB/s, write: 140.0 MiB/s
INFO:  91% (29.4 GiB of 32.0 GiB) in 2m 2s, read: 880.9 MiB/s, write: 41.6 MiB/s
INFO: 100% (32.0 GiB of 32.0 GiB) in 2m 5s, read: 886.4 MiB/s, write: 42.3 MiB/s
INFO: backup is sparse: 16.03 GiB (50%) total zero data
INFO: transferred 32.00 GiB in 125 seconds (262.1 MiB/s)
INFO: archive file size: 7.67GB
INFO: adding notes to backup
INFO: Finished Backup of VM 105 (00:02:06)
INFO: Backup finished at 2024-03-15 12:56:14
INFO: Backup job finished successfully
INFO: notified via target `mail-to-root`
TASK OK
```

</details>

felge20000, 2024-03-15:
> After upgrading til 12.1 full snapshot on Synology Virtual Machine Manager is not possible. It can only make "Crash Consistant" snapshot.

Same here, guest tools are running.
filesystem-consistent snapshots aber broken on my Synology VMM since last HAOS Update.
As I am reading Synology also uses the QEMU agent

felge20000, 2024-03-15:
@agners 
Thanks for providing the commands, I'd be at a loss here

so the output on my system is exactly as you posted, but the time between freezing and thawing is reduced to 0-1s instead of the 10-15s while it was working.
So the service gets called and seems to think everything's ok?
(let's see if those screenshots work)

Working:
[image]

After 12.1 Update, not working
[image]

felge20000, 2024-03-15:
From the VMM log:
Warning,2024/03/15 16:09:31,USER ,Took a snapshot [GMT-2024.03.15-15.09.23] from virtual machine [Home Assistant] by [USER] without filesystem consistency. Reason: [Filesystem failed to freeze because the virtual machine is busy]

agners, 2024-03-15:
> So the service gets called and seems to think everything's ok?

Hm, yes this looks as if all is good. You can also call the service calls explicitly using:

```
/usr/libexec/haos-freeze-hook freeze; echo $?
/usr/libexec/haos-freeze-hook thaw; echo $?
```

(call both, otherwise the system will stay in freeze.

agners, 2024-03-15:
> Warning,2024/03/15 16:09:31,USER ,Took a snapshot [GMT-2024.03.15-15.09.23] from virtual machine [Home Assistant] by [USER] without filesystem consistency. Reason: [Filesystem failed to freeze because the virtual machine is busy]

@felge20000 hm, I don't think that your case is related to the snapshot freeze feature really. This seems to be a Synology/Hypervisor related issue to me.

felge20000, 2024-03-15:
Yepp, freezing and thawing manually also worked.
So yeah, seems like the hypervisor and the vm don't like to talk to each other any more since the update.
But I guess that's really a separated issue from the OP's. I was just "relieved" to see from @Soleima77 I'm not the only one with synology and snapshot problems :)
So sorry for taking over this topic @GrimD

GrimD, 2024-03-15:
> Yepp, freezing and thawing manually also worked. So yeah, seems like the hypervisor and the vm don't like to talk to each other any more since the update. But I guess that's really a separated issue [...]

@felge20000, Not sure sounds simular to me but kind of out of my depth as not much of a Linux tech. All I know is upgrading to HAOS 12.1 breaks it but it works fine with 12. After initially restoring to 12.0 which fixed it I then allowed it to upgrade again to double-check it broke (it did) and to get all the logs etc to log this. I then restored the VM back to before the 12.1 update again (so back to 12.0) and its happy once more. 

@agners I'll need to find some time to install 12.1 again to try what you have asked but as I have upgraded it to 12.1 twice and it breaks then restoring the whole VM disk back to before the update it certainly seems to be something in the update that's killing it. Will try to test ASAP

GrimD, 2024-03-15:
Well I ended up just doing it now as thinking about it, it doesn't take long to update and then restore it again so here is the result:
[image]

Seems like it issues the freeze and then that's it, left it about 10 mins and it still shows the same. You can see the successful freeze\thaw from my backup last night on version 12.0

genezig, 2024-03-23:
Same problem here with HAOS 12.1. Debian bookworm host, QEMU/KVM virtualization. Trying to create snapshot of vm with `virsh snapshot-create-as --quiesce --disk-only` results in `error: internal error: unable to execute QEMU agent command 'guest-fsfreeze-freeze': fsfreeze hook has failed with status 1` from virsh. `journalctl -u qemu-agent` output on HAOS guest console quite similar to the above quoted by [felge20000] (thaw only 1 sec after freeze, too fast). Tried changing virtual disk attachment to vm from sata to virtio - no help. According HAOS changelog last qemu guest agent update (to 8.0.5) happened in 11.1 so this should not be the cause as 12.0 is not throwing this error. Maybe a problem related to kernel 6.6.20, should not be limited to HAOS guests then - but I did not find anything related on the net.

joggs, 2024-03-24:
Same here with 12.1

TheFitzZZ, 2024-04-08:
Same issue with 12.1 running on unraid kvm

samcrang, 2024-04-11:
I upgraded to 12.2 and my backups ran successfully.

Thanks for fixing this.

GrimD, 2024-04-12:
All good here too on 12.2

Thank you.

felge20000, 2024-04-12:
My Synology vm ist also snapshotting happily since 12.2
Thanks to everyone involved <3

VNRARA, 2024-07-28:
I've got this same issue, but with 12.4 for some reason with HA as a proxmox VM.

commit 0f44ae69 (2024-04-08): Only run HA CLI interactively if stdout is a terminal

Flags for running HA CLI commands in an interactive shell added in #3238
cause the command to fail if the process is not running in a terminal.
This is needed for example for the fsfreeze hook, otherwise the command
fails, as seen in this trace when the hook is executed:

-----------
+ '[' thaw '=' freeze ]
+ '[' thaw '=' thaw ]
+ echo 'File system thaw requested, thawing Home Assistant'
File system thaw requested, thawing Home Assistant
+ ha backups thaw
the input device is not a TTY
------------

However, for example on Proxmox this mess [... cut]
commit 3da2e2e9 (2024-04-08): Only run HA CLI interactively if stdout is a terminal

Flags for running HA CLI commands in an interactive shell added in #3238
cause the command to fail if the process is not running in a terminal.
This is needed for example for the fsfreeze hook, otherwise the command
fails, as seen in this trace when the hook is executed:

-----------
+ '[' thaw '=' freeze ]
+ '[' thaw '=' thaw ]
+ echo 'File system thaw requested, thawing Home Assistant'
File system thaw requested, thawing Home Assistant
+ ha backups thaw
the input device is not a TTY
------------

However, for example on Proxmox this mess [... cut]
commit 4edaefd0 (2024-04-08): Only run HA CLI interactively if stdout is a terminal

Flags for running HA CLI commands in an interactive shell added in #3238
cause the command to fail if the process is not running in a terminal.
This is needed for example for the fsfreeze hook, otherwise the command
fails, as seen in this trace when the hook is executed:

-----------
+ '[' thaw '=' freeze ]
+ '[' thaw '=' thaw ]
+ echo 'File system thaw requested, thawing Home Assistant'
File system thaw requested, thawing Home Assistant
+ ha backups thaw
the input device is not a TTY
------------

However, for example on Proxmox this mess [... cut]
PR #3292 (2024-04-08): Only run HA CLI interactively if stdout is a terminal
Flags for running HA CLI commands in an interactive shell added in #3238 cause the command to fail if the process is not running in a terminal. This is needed for example for the fsfreeze hook, otherwise the command fails, as seen in this trace when the hook is executed:

```
+ '[' thaw '=' freeze ]
+ '[' thaw '=' thaw ]
+ echo 'File system thaw requested, thawing Home Assistant'
File system thaw requested, thawing Home Assistant
+ ha backups thaw
the input device is not a TTY
```

However, for example on Proxmox this message is not logged  [... cut]