home-assistant/operating-system #3252: (x86_64) HA OS doesn't boot when updating from 11.5 to 12.0 or 12.1

Milo721, 2024-03-14:
### Describe the issue you are experiencing

When i upgrade from 11.5 to 12.0 or 12.1 the OS throws out these errors:

ACPI ERROR: no handler or method for gpe 0b

or

ACPI ERROR: no installed handler for fixed event

and it doesn't boot.

The computer i'm using for hosting HA is an ASUS Vivobook E402NA and currently i'm using the B slot for booting into HA OS 11.5.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Upgrade to any 12 version
2. Wait the PC to reboot
3. HA OS doesn't boot

### Anything in the Supervisor logs that might be useful for us?

```txt
I don't know how to see the logs from one slot to the other
```

### Anything in the Host logs that might be useful for us?

```txt
I don't know how to see the logs from one slot to the other
```

### System information

ASUS Vivobook E402NA
4GB RAM
32GB Emmc

DanDon, 2024-03-14:
Same issue. Was the same with 12.0 and hasn't been fixed in 12.1 so had to revert to 11.5 so get a working system again.

agners, 2024-03-15:
> It got closed here:
> 
> #3195
> 
> But it isn't resolved at all

Did you read the original issue in #3195? It was about a error on the console, compleeeetly different from what your reported. The original poster also confirmed that the issue he was seeing is resolved, see https://github.com/home-assistant/operating-system/issues/3195#issuecomment-1965894777.

Looking at the screenshot you posted in https://github.com/home-assistant/operating-system/issues/3195#issuecomment-1964629609, the errors are also different to this issue.

So again: Your issue manifests different (other error messages) and therefor (most likely) different root cause.

Instead of hijacking other people's report, please open a issue report on your own.

agners, 2024-03-15:
@Milo721 hm, this is most likely related to the Linux 6.6 upgrade. It looks similar to [this report on Arch Linux](https://bbs.archlinux.org/viewtopic.php?id=289284).

From the system console (keyboard/terminal), can you enter the OS shell using `login`, then enter the following command:

```
sed -i "1 s/$/ pci=nommconf/" /mnt/boot/cmdline.txt
```

Then try another update and see if that works.

Milo721, 2024-03-15:
Still the same problem

closed 2024-06-22 by github-actions[bot]