home-assistant/operating-system #3253: Upgrade from 12.0 to 12.1 kills DisplayPort output

humantex, 2024-03-14:
### Describe the issue you are experiencing

Logged in today after a fresh/new install and initial setup on x86 box (HP ProDesk 400 G2) day before yesterday, and followed both prompts to upgrade HA core and HAOS. Core update went smoothly, with a successful reboot - attached monitor showed onscreen progress and running state before/after completing. OS update showed a series of notifications through web panel, but showed nothing on the monitor after hardware reboot. System is up and running in the browser, but there is zero video output to the monitor. Input selector on the monitor does show it as an active input (electrically, as 'connected'), but won't push any video.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Login to 12.0
2. Update Core
3. System reboots/runs fine
4. Update OS from 12.0 to 12.1
5. System restarts - but - attached monitor is blank
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs of any kind are showing
```

### Anything in the Host logs that might be useful for us?

```txt
Again - no logs available
```

### System information

## System Information

version | core-2024.3.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20-haos
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 219.4 GB
disk_used | 5.9 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), VLC (0.3.0), motionEye (0.20.0), Log Viewer (0.17.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 12, 2024 at 10:02 PM
-- | --
current_recorder_run | March 14, 2024 at 5:59 PM
estimated_db_size | 0.91 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

sairon, 2024-03-15:
The only change that could potentially cause it is the upstream kernel update. You say the system is running, can you get the host logs from the machine? Also can you check if it gets resolved by reverting to the previous version using `ha os update --version 12.0`, without unplugging anything or cold-booting in between?

humantex, 2024-03-15:
After posting this issue, this is all there is in the 'Host' log:

```
Mar 15 10:35:07 homeassistant kernel: audit: type=1334 audit(1710498907.518:331): prog-id=108 op=UNLOAD
Mar 15 10:35:07 homeassistant kernel: audit: type=1334 audit(1710498907.518:332): prog-id=107 op=UNLOAD
Mar 15 12:41:17 homeassistant kernel: audit: type=1334 audit(1710506477.544:333): prog-id=110 op=LOAD
Mar 15 12:41:17 homeassistant kernel: audit: type=1334 audit(1710506477.544:334): prog-id=111 op=LOAD
Mar 15 12:41:17 homeassistant kernel: audit: type=1334 audit(1710506477.544:335): prog-id=112 op=LOAD
Mar 15 12:41:17 homeassistant systemd[1]: Starting Hostname Service...
Mar 15 12:41:17 homeassistant systemd[1]: Started Hostname Service.
Mar 15 12:41:17 homeassistant kernel: audit: type=1334 audit(1710506477.688:336): prog-id=113 op=LOAD
Mar 15 12:41:17 homeassistant kernel: audit: type=1334 audit(1710506477.688:337): prog-id=114 op=LOAD
Mar 15 12:41:17 homeassistant kernel: audit: type=1334 audit(1710506477.688:338): prog-id=115 op=LOAD
Mar 15 12:41:17 homeassistant systemd[1]: Starting Time & Date Service...
Mar 15 12:41:17 homeassistant systemd[1]: Started Time & Date Service.
Mar 15 12:41:47 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Mar 15 12:41:47 homeassistant kernel: audit: type=1334 audit(1710506507.760:339): prog-id=112 op=UNLOAD
Mar 15 12:41:47 homeassistant kernel: audit: type=1334 audit(1710506507.760:340): prog-id=111 op=UNLOAD
Mar 15 12:41:47 homeassistant kernel: audit: type=1334 audit(1710506507.760:341): prog-id=110 op=UNLOAD
Mar 15 12:41:47 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
Mar 15 12:41:47 homeassistant kernel: audit: type=1334 audit(1710506507.830:342): prog-id=115 op=UNLOAD
Mar 15 12:41:47 homeassistant kernel: audit: type=1334 audit(1710506507.830:343): prog-id=114 op=UNLOAD
Mar 15 12:41:47 homeassistant kernel: audit: type=1334 audit(1710506507.830:344): prog-id=113 op=UNLOAD
Mar 15 14:47:57 homeassistant kernel: audit: type=1334 audit(1710514077.881:345): prog-id=116 op=LOAD
Mar 15 14:47:57 homeassistant kernel: audit: type=1334 audit(1710514077.882:346): prog-id=117 op=LOAD
Mar 15 14:47:57 homeassistant kernel: audit: type=1334 audit(1710514077.882:347): prog-id=118 op=LOAD
Mar 15 14:47:57 homeassistant systemd[1]: Starting Hostname Service...
Mar 15 14:47:58 homeassistant systemd[1]: Started Hostname Service.
Mar 15 14:47:58 homeassistant kernel: audit: type=1334 audit(1710514078.029:348): prog-id=119 op=LOAD
Mar 15 14:47:58 homeassistant kernel: audit: type=1334 audit(1710514078.030:349): prog-id=120 op=LOAD
Mar 15 14:47:58 homeassistant kernel: audit: type=1334 audit(1710514078.030:350): prog-id=121 op=LOAD
Mar 15 14:47:58 homeassistant systemd[1]: Starting Time & Date Service...
Mar 15 14:47:58 homeassistant systemd[1]: Started Time & Date Service.
Mar 15 14:48:28 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Mar 15 14:48:28 homeassistant kernel: audit: type=1334 audit(1710514108.098:351): prog-id=118 op=UNLOAD
Mar 15 14:48:28 homeassistant kernel: audit: type=1334 audit(1710514108.098:352): prog-id=117 op=UNLOAD
Mar 15 14:48:28 homeassistant kernel: audit: type=1334 audit(1710514108.098:353): prog-id=116 op=UNLOAD
Mar 15 14:48:28 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
Mar 15 14:48:28 homeassistant kernel: audit: type=1334 audit(1710514108.170:354): prog-id=121 op=UNLOAD
Mar 15 14:48:28 homeassistant kernel: audit: type=1334 audit(1710514108.170:355): prog-id=120 op=UNLOAD
Mar 15 14:48:28 homeassistant kernel: audit: type=1334 audit(1710514108.170:356): prog-id=119 op=UNLOAD
Mar 15 16:54:38 homeassistant kernel: audit: type=1334 audit(1710521678.228:357): prog-id=122 op=LOAD
Mar 15 16:54:38 homeassistant kernel: audit: type=1334 audit(1710521678.228:358): prog-id=123 op=LOAD
Mar 15 16:54:38 homeassistant kernel: audit: type=1334 audit(1710521678.228:359): prog-id=124 op=LOAD
Mar 15 16:54:38 homeassistant systemd[1]: Starting Hostname Service...
Mar 15 16:54:38 homeassistant systemd[1]: Started Hostname Service.
Mar 15 16:54:38 homeassistant kernel: audit: type=1334 audit(1710521678.370:360): prog-id=125 op=LOAD
Mar 15 16:54:38 homeassistant kernel: audit: type=1334 audit(1710521678.370:361): prog-id=126 op=LOAD
Mar 15 16:54:38 homeassistant kernel: audit: type=1334 audit(1710521678.370:362): prog-id=127 op=LOAD
Mar 15 16:54:38 homeassistant systemd[1]: Starting Time & Date Service...
Mar 15 16:54:38 homeassistant systemd[1]: Started Time & Date Service.
Mar 15 16:55:08 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Mar 15 16:55:08 homeassistant kernel: audit: type=1334 audit(1710521708.449:363): prog-id=124 op=UNLOAD
Mar 15 16:55:08 homeassistant kernel: audit: type=1334 audit(1710521708.449:364): prog-id=123 op=UNLOAD
Mar 15 16:55:08 homeassistant kernel: audit: type=1334 audit(1710521708.449:365): prog-id=122 op=UNLOAD
Mar 15 16:55:08 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
Mar 15 16:55:08 homeassistant kernel: audit: type=1334 audit(1710521708.543:366): prog-id=127 op=UNLOAD
Mar 15 16:55:08 homeassistant kernel: audit: type=1334 audit(1710521708.543:367): prog-id=126 op=UNLOAD
Mar 15 16:55:08 homeassistant kernel: audit: type=1334 audit(1710521708.543:368): prog-id=125 op=UNLOAD
[... cut ...] t's probably all I'll see if it does anything at all. Is there a command prompt I can use inside the browser?

If there's no way to do it there, can I accomplish the same thing by restoring a backup, There is one backup listed that references the auto backup done before Core was updated... and that was done before I updated the OS. If that won't accomplish the same thing, I can only flash everything back to the original image version I started with, which was was: haos_generic-x86-64-12.0.img.xz

humantex, 2024-03-31:
A quick update...

The PC this runs on came with a standard hard drive with Win 10 pre-loaded. I removed the SSD flashed with HA and after swapping back to the original drive, everything worked as expected. That at least means it wasn't a hardware related problem that just happened to coincide with the timing of the software/OS update.

As this was only a test run of HA on dedicated hardware for me, this isn't a deal breaker, but obviously there are issues somewhere after the 12.0 to 12.1 update. If I re-flash the SSD, I may just leave it at 12.0 for now. Hopefully there will be some progress with an upcoming update.

closed 2024-07-08 by github-actions[bot]