home-assistant/operating-system #3255: RTL8821CU still doesn’t work after #3225

aambroze, 2024-03-18:
### Describe the issue you are experiencing

I previously rolled back update from 12.0 to 11.5 due to issue allegedly fixed later in #3225.

After updating to 12.1 adapter still doesn’t work.

I tried enabling the adapter but I have 500 error from HA.
[image]
[image]
[image]

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Have HA 11.5 up and running with ODROID WLAN dongle.
2. Update to 12.0 (not sure if necessary)
3. Revert to 11.5 (not sure if necessary) from CLI
4. Update to 12.1

### Anything in the Supervisor logs that might be useful for us?

```txt
File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 189, in block_bad_requests
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 205, in system_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 272, in token_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 283, in core_proxy
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/utils.py", line 63, in wrap_api
    answer = await method(api, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/network.py", line 222, in interface_update
    await asyncio.shield(self.sys_host.network.apply_changes(interface))
  File "/usr/src/supervisor/supervisor/host/network.py", line 221, in apply_changes
    settings = get_connection_from_interface(interface, self.sys_dbus.network)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/dbus/network/setting/generate.py", line 141, in get_connection_from_interface
    "ssid": Variant("ay", interface.wifi.ssid.encode("UTF-8")),
```

### Anything in the Host logs that might be useful for us?

```txt
Mar 18 07:08:07 homeassistant kernel: RTW: wlan0- hw port(0) mac_addr =e0:e1:a9:38:66:d7
```

aambroze, 2024-03-18:
I don’t have access to Supervisor logs anymore but I see they were cut.

Log was usual Python error stating about NoneType not having „ssid” attribute.

I reverted system to 11.5 so I can’t reproduce and provide full log, but in case more details will be needed I could try reproducing.

Saugerpaul, 2024-03-18:
Same thing here under 12 and 12.1, Bluetooth no longer works... back to 11.5 no problems.

sairon, 2024-03-19:
The internal server error comes from the Supervisor, please report is as a bug there with complete logs, as it might be unrelated to OS. Ideally please try to obtain **complete** Supervisor and Host logs.