home-assistant/operating-system #3262: HaOS info messaags from supervisor show up as error in systemd journal

rct, 2024-03-21:
### Describe the issue you are experiencing

If I try to use `journalctl -p err` to look for errors on the OS, hardware, etc., I get a lot of lines from supervisor that are tagged `INFO`. This makes it difficult to see if any actual problems have occurred.

  

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. On the host console, or through the host ssh (-p 22222), run `journalctl -p err`
2. Notice how many supervisor messages show up, with are tagged as `INFO` 
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
The same messages as shown in green via the UI as `journalctl -p err` categorizes as errors.
```

### Anything in the Host logs that might be useful for us?

```txt
# journalctl --since '2 days ago' -p err
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.check] System checks complete
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
Mar 21 15:19:09 hasstst hassio_supervisor[407]: 24-03-21 11:19:09 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
Mar 21 15:19:38 hasstst hassio_supervisor[407]: 24-03-21 11:19:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository
Mar 21 15:19:38 hasstst hassio_supervisor[407]: 24-03-21 11:19:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/rct/Enphase-Envoy-mqtt-json repository
Mar 21 15:19:38 hasstst hassio_supervisor[407]: 24-03-21 11:19:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/bropat/hassio-eufy-security-ws repository
Mar 21 15:19:38 hasstst hassio_supervisor[407]: 24-03-21 11:19:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
Mar 21 15:19:38 hasstst hassio_supervisor[407]: 24-03-21 11:19:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
Mar 21 15:19:38 hasstst hassio_supervisor[407]: 24-03-21 11:19:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
Mar 21 15:19:39 hasstst hassio_supervisor[407]: 24-03-21 11:19:39 INFO (MainThread) [supervisor.store] Loading add-ons from store: 82 all - 0 new - 0 remove
Mar 21 15:19:39 hasstst hassio_supervisor[407]: 24-03-21 11:19:39 INFO (MainThread) [supervisor.store] Loading add-ons from store: 82 all - 0 new - 0 remove
Mar 21 15:19:48 hasstst hassio_supervisor[407]: 24-03-21 11:19:48 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
Mar 21 15:49:48 hasstst hassio_supervisor[407]: 24-03-21 11:49:48 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
Mar 21 15:59:26 hasstst hassio_supervisor[407]: 24-03-21 11:59:26 INFO (MainThread) [supervisor.host.info] Updating local host information
Mar 21 15:59:27 hasstst hassio_supervisor[407]: 24-03-21 11:59:27 INFO (MainThread) [supervisor.host.services] Updating service information
Mar 21 15:59:27 hasstst hassio_supervisor[407]: 24-03-21 11:59:27 INFO (MainThread) [supervisor.host.network] Updating local network information
Mar 21 15:59:27 hasstst hassio_supervisor[407]: 24-03-21 11:59:27 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
Mar 21 15:59:27 hasstst hassio_supervisor[407]: 24-03-21 11:59:27 INFO (MainThread) [supervisor.host.manager] Host information reload completed
Mar 21 16:19:09 hasstst hassio_supervisor[407]: 24-03-21 12:19:09 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
Mar 21 16:19:09 hasstst hassio_supervisor[407]: 24-03-21 12:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
Mar 21 16:19:09 hasstst hassio_supervisor[407]: 24-03-21 12:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
Mar 21 16:19:09 hasstst hassio_supervisor[407]: 24-03-21 12:19:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
[... 74 lines cut ...]
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.4.0), Advanced SSH & Web Terminal (17.2.0), UniFi Network Application (3.0.3), Studio Code Server (5.15.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 1
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 14, 2024 at 7:03 PM
-- | --
current_recorder_run | March 18, 2024 at 2:19 PM
estimated_db_size | 7.20 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

sairon, 2024-03-25:
This is basically a duplicate of the issue discussed here: https://github.com/home-assistant/supervisor/issues/4948

Sorry to hear about the stability issues you have. With all the clues, everything points to the hypervisor now since it's not an issue that appears here widely but I don't have any better tips what to try next :(

closed 2024-03-25 by sairon

rct, 2024-04-01:
Yes, basically the same though that was for a supervised install vs. Home Assistant OS.

That issue has also been closed.

It is somewhat disappointing that for Home Assistant OS that so much supervisor log output is logged as an error, including periodic checks like this:

```
2024-04-01 13:14:24.720 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2024-04-01 13:14:24.730 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2024-04-01 13:14:24.876 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2024-04-01 13:14:24.876 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2024-04-01 13:14:24.877 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-04-01 13:14:24.877 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2024-04-01 13:14:24.880 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-04-01 13:14:24.880 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2024-04-01 13:14:24.881 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2024-04-01 13:14:24.881 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2024-04-01 13:14:24.882 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-04-01 13:14:24.882 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-04-01 13:14:24.943 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-04-01 13:14:24.943 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-04-01 13:14:24.943 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
```

So I guess that isn't accepted as a defect. Is there any point to making this an RFE in the community forum?