home-assistant/operating-system #3267: USB can't set config error -32 Bluetooth dongle [Proxmox 8.1.5 - USB Passthrough]

smtac, 2024-03-26:
### Describe the issue you are experiencing

Bluetooth usb dongle regurlary lost after randomly time (minute, hours or days) on HOAS VM hosting by Proxmox 8.1.5.
Error message on VM screen is "usb *-*: can't set config #1, error -32"
[image]

VM configuration on Proxmox 8.1.5 : 
[image]
[image]

USB dugle are configured in "use uUSB Vendor/Device ID". But same issue when I configure "Use USB Port".
[image]

3 others USB devices are configured for other VM. No issue on theses other usb devices.

Bluetooth is only use for Integration "Xiaomi BLE" with 8 entities

Test make with bluetooth entities configutation "Passive scan" :
------ : Device 1 - Device 2
Test 1 : Enable - Enable
Test 2 : Disable - Enable
Test 3 : Enable - Disable
Test 4 : Disable - Disable 
Issue persiste

Reboot system resolve issue for few times. But issue comme back after randomly time (minutes, hours or days)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.20-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Boot HAOS

- Bluetooth integration work normaly

3. Wait few time (Minutes, Hours or days)
4. Issue append

- Bluetooth entities are lost
- Error Message on VM screen

### Anything in the Supervisor logs that might be useful for us?

```txt
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "src/habluetooth/scanner.py", line 381, in habluetooth.scanner.HaScanner._async_restart_scanner
  File "src/habluetooth/scanner.py", line 333, in _async_start
habluetooth.scanner.ScannerStartError: hci0 (F4:4E:FC:27:E6:55): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "RegisterMonitor" with signature "o" on interface "org.bluez.AdvertisementMonitorManager1" doesn't exist
; Try power cycling the Bluetooth hardware.
2024-03-26 10:31:35.430 ERROR (MainThread) [habluetooth.scanner] hci1 (8C:88:2B:04:91:93): Failed to restart Bluetooth scanner: hci1 (8C:88:2B:04:91:93): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
; Try power cycling the Bluetooth hardware.
Traceback (most recent call last):
  File "src/habluetooth/scanner.py", line 249, in habluetooth.scanner.HaScanner._async_start
  File "src/habluetooth/scanner.py", line 250, in habluetooth.scanner.HaScanner._async_start
  File "/usr/local/lib/python3.12/site-packages/bleak/__init__.py", line 198, in start
    await self._backend.start()
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 185, in start
    self._stop = await manager.active_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 405, in active_scan
    assert_reply(reply)
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "src/habluetooth/scanner.py", line 381, in habluetooth.scanner.HaScanner._async_restart_scanner
  File "src/habluetooth/scanner.py", line 333, in _async_start
habluetooth.scanner.ScannerStartError: hci1 (8C:88:2B:04:91:93): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
; Try power cycling the Bluetooth hardware.
```

### Anything in the Host logs that might be useful for us?

```txt
[ 3443.198457] usb 2-3: can't set config #1, error -32
[ 3507.154133] usb 2-2 :can't set config #1, error -32
```

### System information

## System Information

version | core-2024.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20-haos
arch | x86_64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 30.8 GB
disk_used | 22.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Cloudflared (5.1.7), Git pull (7.14.0), Let's Encrypt (5.0.15), ESPHome (2024.3.0), Terminal & SSH (9.10.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 21 novembre 2023 à 21:21
-- | --
current_recorder_run | 25 mars 2024 à 19:23
estimated_db_size | 2149.32 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

denymen, 2024-04-05:
Hello, The same issue happening with VirtualBox environment. Is it have any timeline when the fix will be ready?

closed 2024-04-10 by sairon

denymen, 2024-04-10:
Hello,

I have the ASUS USB-BT500 (RTL8761BU), which is on the support list if I am not wrong.
I just rebooted my system, so I cannot provide details, but after it will occur again, I will collect some data.