home-assistant/operating-system #3277: hdmi_cec component fails to load

inytar, 2024-03-28:
### Describe the issue you are experiencing

After upgrading to version 12.2.rc1 there's an error message for the hdmi_cec component that it fails to load. Reverting back to version 12.1 (same HA core version) fixes the issue. 

See logs:
```
Logger: homeassistant.setup
Source: setup.py:398
First occurred: 20:44:55 (1 occurrences)
Last logged: 20:44:55

Error during setup of component hdmi_cec
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 398, in _async_setup_component
    result = await task
             ^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hdmi_cec/__init__.py", line 207, in setup
    hdmi_network = HDMINetwork(adapter, loop=loop)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pycec/network.py", line 309, in __init__
    self._command_queue = Queue()
                          ^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 103, in Queue
    return Queue(maxsize, ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/queues.py", line 43, in __init__
    self._rlock = ctx.Lock()
                  ^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 68, in Lock
    return Lock(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 169, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 57, in __init__
    sl = self._semlock = _multiprocessing.SemLock(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system
```

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

12.2.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Make sure hdmi_cec component is setup
2. Upgrade to version 12.2.rc1
3.  Error message will show up

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m2024-03-28 20:44:28.031 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor[0m
[32m2024-03-28 20:44:28.032 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/beta.json[0m
[32m2024-03-28 20:44:28.144 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2024.03.1[0m
[32m2024-03-28 20:44:28.156 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin[0m
[32m2024-03-28 20:44:28.667 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/aarch64-hassio-cli with version 2024.03.1 - 172.30.32.5[0m
[32m2024-03-28 20:44:28.695 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-dns with version 2024.03.0[0m
[32m2024-03-28 20:44:28.713 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin[0m
[... 275 lines cut ...]
dashboards | 3
-- | --
resources | 22
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 23 February 2024 at 18:55
-- | --
current_recorder_run | 28 March 2024 at 20:12
estimated_db_size | 2173.47 MiB
database_engine | postgresql
database_version | 16.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

sairon, 2024-04-02:
Thanks for the report, we identified the issue to be a result of change in Docker 25 that was upgraded in HAOS 12.2.rc1. This needs to be handled in Supervisor, where we already have [an issue](https://github.com/home-assistant/supervisor/issues/4996) describing it bit more in depth. We plan to release new Supervisor version to the beta channel with a fix ASAP (i.e. it should be fixed in Supervisor >2024.03.1), in the meantime the only workaround is to downgrade to the previous OS version (using `ha os update --version 12.1`).

closed 2024-04-02 by sairon

sairon, 2024-04-03:
The issue is fixed in Supervisor 2024.04.0 available on the beta channel now, so it should be safe to upgrade back to 12.2.rc1.