home-assistant/operating-system #3278: I can't login via ssh anymore:

rononline, 2024-03-28:
I can't login via ssh anymore:

```
❯ ssh hassio@homeassistant.local
sudo: unable to allocate pty: Read-only file system
Connection to homeassistant.local closed.
```

_Originally posted by @rononline in https://github.com/home-assistant/operating-system/discussions/3276#discussioncomment-8945694_

maddinhome, 2024-03-29:
Same here with a custom integration:

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.wemportal
Quelle: custom_components/wemportal/wemportalapi.py:122
Integration: Weishaupt WEM Portal (Dokumentation, Probleme)
Erstmals aufgetreten: 07:42:42 (2 Vorkommnisse)
Zuletzt protokolliert: 08:40:12

[Errno 30] Read-only file system

dcmeglio, 2024-03-31:
Thought it was just me, having the same issue since upgrading. Port 22222 SSH works, but 22 does not.

Anto79-ops, 2024-04-01:
im also getting the same [issue](https://github.com/home-assistant/operating-system/issues/3282) `[Errno 30] Read-only file system` for my InfluxDB integration on 12.2rc1

sairon, 2024-04-02:
Thanks for the report, we identified the issue to be a result of change in Docker 25 that was upgraded in HAOS 12.2.rc1. This needs to be handled in Supervisor, where we already have [an issue](https://github.com/home-assistant/supervisor/issues/4996) describing it bit more in depth. We plan to release new Supervisor version to the beta channel with a fix ASAP (i.e. it should be fixed in Supervisor >2024.03.1), in the meantime the only workaround is to downgrade to the previous OS version (using `ha os update --version 12.1`).

closed 2024-04-02 by sairon

sairon, 2024-04-03:
The issue is fixed in Supervisor 2024.04.0 available on the beta channel now, so it should be safe to upgrade back to 12.2.rc1.