home-assistant/operating-system #3279: CORAL TPU M.2 B+M MODULE ([212-G650-04686-01]) M2 slot not detected by HAOS

seblang, 2024-03-30:
### Describe the issue you are experiencing

CORAL TPU M.2 B+M MODULE ([212-G650-04686-01]) M2 slot not detected by HAOS..........;;it's appear that normally I have to find "Global Unichip Corp" in hadware list when detected.
If I check in the bios it's appear to be detectected by the bios

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...
just added the coral but not detected un HA

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing
```

### Anything in the Host logs that might be useful for us?

```txt
nothing
```

### System information

## System Information

version | core-2024.2.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.6.20-haos
arch | x86_64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1473
Downloaded Repositories | 26
HACS Data | ok

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 46

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 21 janvier 2025 à 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-17.ui.nabu.casa
certificate_status | ready
instance_id | b326ba0173a84fceb0a63e3c30dfe62d
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 116.7 GB
disk_used | 39.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Google Assistant SDK (2.5.0), Advanced SSH & Web Terminal (17.2.0), Terminal & SSH (9.10.0), File editor (5.8.0), Glances (0.21.1), MQTT Server & Web client (1.2.0), Node-RED (17.0.10), Bluetooth Presence Monitor (1.0.0), Z-Wave JS (0.4.5), Samba share (12.3.1), Home Assistant Google Drive Backup (0.112.1), SQLite Web (4.1.2), Custom deps deployment (1.3.4), Mosquitto broker (6.4.0), Filebrowser (2.23.0_12), ESPHome (2024.2.2), TileBoard (1.3.8), NGINX Home Assistant SSL proxy (3.9.0), Studio Code Server (5.15.0), Matter Server (5.4.1), openWakeWord (1.10.0), Network UPS Tools (0.13.0), go2rtc (1.8.5), Frigate (0.13.2), Double Take (1.13.1), OneDrive Backup (2.3.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 11
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 29 mars 2024 à 16:14
-- | --
current_recorder_run | 29 mars 2024 à 17:37
estimated_db_size | 673.75 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
remaining_requests | 5
rooftop_site_count | 1
polling_hours | 6 -> 20 and once at midnight

</details>

sairon, 2024-04-08:
Can you please post the output of `lspci`, and ideally also `dmesg` shortly after the boot?

Zaleo80, 2024-06-03:
+1

### Describe the issue you are experiencing
When using a Coral M.2 Accelerator A+E Key Frigate Add-On does not start anymore.

### What operating system image do you use?
rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?
6.6.28-haos-raspi

### Did you upgrade the Operating System.
Yes

### Steps to reproduce the issue

1. A working HomeAssistant with Frigate Add-On and intergration.
2. Installed a PINEBOARDS Hat AI! Dual Edge Coral TPU Bundle for Raspberry Pi 5
3. Checked under Settings -> hardware -> Show All -> Search for apex (It does showup, see image below)
[image]

4. Replaced the CPU detector for a pci one (Let's start with one) in frigate.yaml.
```
detectors:
#  cpu1:
#    type: cpu
#    num_threads: 2
  coral1:
    type: edgetpu
    device: pci:0
```

5. Restarted Frigate

### Anything in the Supervisor logs that might be useful for us?

```
# lspci
03:00.0 Class 0880: 1ac1:089a
02:07.0 Class 0604: 1b21:1182
00:00.0 Class 0604: 14e4:2712
02:03.0 Class 0604: 1b21:1182
01:00.0 Class 0604: 1b21:1182
00:00.0 Class 0604: 14e4:2712
04:00.0 Class 0108: 1987:5013
01:00.0 Class 0200: 1de4:0001
```

```
# dmesg | grep apex
[255773.884961] apex 0000:03:00.0: Couldn't reinit interrupts: -28
[255773.884996] apex 0000:03:00.0: Permission checking failed.
```

### Anything in the Host logs that might be useful for us?

```
Process detector:coral1:
[2024-06-03 11:27:45] frigate.detectors.plugins.edgetpu_tfl ERROR   : No EdgeTPU was detected. If you do not have a Coral device yet, you must configure CPU detectors.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/frigate/frigate/object_detection.py", line 102, in run_detector
    object_detector = LocalObjectDetector(detector_config=detector_config)
  File "/opt/frigate/frigate/object_detection.py", line 53, in __init__
    self.detect_api = create_detector(detector_config)
  File "/opt/frigate/frigate/detectors/__init__.py", line 18, in create_detector
    return api(detector_config)
  File "/opt/frigate/frigate/detectors/plugins/edgetpu_tfl.py", line 41, in __init__
    edge_tpu_delegate = load_delegate("libedgetpu.so.1.0", device_config)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1.0
```

### System information

version | core-2024.5.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.28-haos-raspi
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1393
Downloaded Repositories | 22

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 15 april 2025 om 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-10.ui.nabu.casa
certificate_status | ready
instance_id | **********************************
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.05.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 228.5 GB
disk_used | 64.0 GB
healthy | true
supported | true
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.15.0), Terminal & SSH (9.14.0), Samba share (12.3.1), Home Assistant Google Drive Backup (0.112.1), eufy-security-ws (1.8.0-2), Mosquitto broker (6.4.1), ESPHome (2024.5.4), HassOS SSH port 22222 Configurator (0.9.3), Whisper (2.1.0), Piper (1.5.0), openWakeWord (1.10.0), Music Assistant (2.0.4), Frigate (Full Access) (0.13.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 10
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 28 mei 2024 om 10:58
-- | --
current_recorder_run | 31 mei 2024 om 12:30
estimated_db_size | 2314.16 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

closed 2024-09-09 by github-actions[bot]