home-assistant/operating-system #3282: Cannot connect to InfluxDB after updating to 12.2.rc1

Anto79-ops, 2024-04-01:
### Describe the issue you are experiencing

hello,

After updating to 12.2.rc1, my InfluxDB is not working anymore. Here's my InfluxDB setup via config yaml:

```
api_version: 2
ssl: false
host: 192.168.1.154
port: 8086
token: [redacted]
organization: [redacted]
bucket: Home Assistant
tags:
  source: HA
tags_attributes:
  - friendly_name
default_measurement: units
max_retries: 3
exclude:
  entities:
    - zone.home
  domains:
    - persistent_notification
    - person
    - alarm_control_panel
    - automation 
    - input_boolean
    - remote 
    - script
    - climate
    - cover
    - device_tracker
    - fan
    - light
    - lock
    - media_player
    - switch
include:
  domains:
    - sensor
    - binary_sensor
  entities:
    - weather.home
```

and upon every reboot, I get this error message:

```
Logger: homeassistant.components.influxdb
Source: components/influxdb/__init__.py:591
integration: InfluxDB (documentation, issues)
First occurred: 7:14:34 PM (1 occurrences)
Last logged: 7:14:34 PM

Cannot connect to InfluxDB due to '[Errno 30] Read-only file system'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible.
```
and then every 2 minutes or so, I get these logs:

```
Logger: homeassistant.components.influxdb
Source: components/influxdb/__init__.py:567
integration: InfluxDB (documentation, issues)
First occurred: 7:15:34 PM (4 occurrences)
Last logged: 7:18:34 PM

Catching up, dropped 2020 old events.
Catching up, dropped 3883 old events.
Catching up, dropped 338 old events.
Catching up, dropped 260 old events.
```

My HAOS system is running on a RPi4-4GB/SDD and my InfluxDB server is running on the same network on a Ubuntu Computer. The InfluxDB server is running normally and is accessible. The only thing that changed was updating from HAOS 12.1 to 12.2rc1. I'm also on HA core beta 2024.4.0b3

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.2.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. On 12.2rc1, restart Core and/or Host
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
logs available if required
```

### Anything in the Host logs that might be useful for us?

```txt
logs available if required
```

### System information

## System Information

version | core-2024.4.0b3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | America/Edmonton
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4752
Installed Version | 1.34.0
Stage | running
Available Repositories | 1469
Downloaded Repositories | 22

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | August 21, 2024 at 6:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | us-east-1-7.ui.nabu.casa
certificate_status | ready
instance_id | a1e06eefeedc498b850f3606439247a3
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2.rc1
-- | --
update_channel | beta
supervisor_version | supervisor-2024.03.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 457.7 GB
disk_used | 33.8 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (17.2.0), Home Assistant Google Drive Backup (0.112.1), File editor (5.8.0), Samba share (12.3.1), AdGuard Home (5.0.5), Z-Wave JS UI (3.4.1), Glances (0.21.1), AppDaemon (0.16.4), Grafana (9.2.1), ESPHome (2024.3.1), Whisper (2.0.0), Piper (1.5.0), SQLite Web (4.1.2), Matter Server (5.4.1), openWakeWord (1.10.0), porcupine1 (1.2.0), Silicon Labs Multiprotocol (2.4.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 14
-- | --
resources | 13
views | 25
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 22, 2024 at 12:05 PM
-- | --
current_recorder_run | March 31, 2024 at 7:13 PM
estimated_db_size | 1882.00 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2024-04-02:
There is also issue https://github.com/home-assistant/core/issues/108810 which reports the same problem.

Anto79-ops, 2024-04-02:
@agners oh so interesting! that issue you found is months old, but this only happened to me recently when updating to to 12.2rc1.

Did you want me to roll back to 12.1 and see if that fixes the issue to confirm its related to 12.2-rc1? If so, can you please tell how? thank you!

sairon, 2024-04-02:
Thanks for the report, we identified the issue to be a result of change in Docker 25 that was upgraded in HAOS 12.2.rc1. This needs to be handled in Supervisor, where we already have [an issue](https://github.com/home-assistant/supervisor/issues/4996) describing it bit more in depth. We plan to release new Supervisor version to the beta channel with a fix ASAP (i.e. it should be fixed in Supervisor >2024.03.1), in the meantime the only workaround is to downgrade to the previous OS version (using `ha os update --version 12.1`).

For the Core issue being few months old already - some Supervised installation moved to the newer Docker version sooner, so it manifested earlier there (again, see the linked issue for more details).

closed 2024-04-02 by sairon

Anto79-ops, 2024-04-02:
thanks @sairon. Just to confirm, downgrading to 12.1 it solved the issue.

sairon, 2024-04-03:
The issue is fixed in Supervisor 2024.04.0 available on the beta channel now, so it should be safe to upgrade back to 12.2.rc1.

Anto79-ops, 2024-04-03:
Thanks @sairon! I just updated to Supervisor 2024.4.0, and then updated to OS 12.2rc1 (from 12.1), but I'm sorry the problem came back for my InfluxDB.  Same  error messages as above. 

Do you require some logs?

sairon, 2024-04-03:
@Anto79-ops Unless you also upgraded HA Core after upgrading the Supervisor, you must run `ha core rebuild`, as mentioned in the linked Supervisor issue. If the problem persists even then, please use that ticket for further reports.