home-assistant/operating-system #3284: AX101 WiFi 6 module issue

AtomRoman, 2024-04-01:
### Describe the issue you are experiencing

I have a Beelink MINI-S12 Mini PC with Alder Lake-N (N95/N100) with the AX101 WiFi 6 module  This is not working in Home Assistant yet.  Note: I have installed the full HAOS software (Generic x86-64).

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
2024-03-31 02:08:05.085 homeassistant kernel: iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
2024-03-31 02:08:05.085 homeassistant kernel: iwlwifi 0000:00:14.3: Transport status: 0x0000004B, valid: 7
2024-03-31 02:08:05.085 homeassistant kernel: iwlwifi 0000:00:14.3: 0x20101A0D | ADVANCED_SYSASSERT
2024-03-31 02:08:05.085 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000000 | umac branchlink1
2024-03-31 02:08:05.086 homeassistant kernel: iwlwifi 0000:00:14.3: 0x8045E934 | umac branchlink2
2024-03-31 02:08:05.087 homeassistant kernel: iwlwifi 0000:00:14.3: 0x0106B050 | umac interruptlink1
2024-03-31 02:08:05.088 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000000 | umac interruptlink2
2024-03-31 02:08:05.089 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000006 | umac data1
2024-03-31 02:08:05.091 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000001 | umac data2
2024-03-31 02:08:05.091 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00FF0FFF | umac data3
2024-03-31 02:08:05.091 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000048 | umac major
2024-03-31 02:08:05.092 homeassistant kernel: iwlwifi 0000:00:14.3: 0xDAA05125 | umac minor
2024-03-31 02:08:05.093 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00998088 | frame pointer
2024-03-31 02:08:05.094 homeassistant kernel: iwlwifi 0000:00:14.3: 0xC0886BBC | stack pointer
2024-03-31 02:08:05.096 homeassistant kernel: iwlwifi 0000:00:14.3: 0x0039050F | last host cmd
2024-03-31 02:08:05.096 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000000 | isr status reg
2024-03-31 02:08:05.097 homeassistant kernel: iwlwifi 0000:00:14.3: IML/ROM dump:
2024-03-31 02:08:05.097 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000B03 | IML/ROM error/state
2024-03-31 02:08:05.099 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00004F80 | IML/ROM data1
2024-03-31 02:08:05.099 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000090 | IML/ROM WFPM_AUTH_KEY_0
2024-03-31 02:08:05.102 homeassistant kernel: iwlwifi 0000:00:14.3: Fseq Registers:
2024-03-31 02:08:05.102 homeassistant kernel: iwlwifi 0000:00:14.3: 0x60000000 | FSEQ_ERROR_CODE
2024-03-31 02:08:05.102 homeassistant kernel: iwlwifi 0000:00:14.3: 0x80350002 | FSEQ_TOP_INIT_VERSION
2024-03-31 02:08:05.103 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00150000 | FSEQ_CNVIO_INIT_VERSION
2024-03-31 02:08:05.103 homeassistant kernel: iwlwifi 0000:00:14.3: 0x0000A482 | FSEQ_OTP_VERSION
2024-03-31 02:08:05.105 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000003 | FSEQ_TOP_CONTENT_VERSION
2024-03-31 02:08:05.105 homeassistant kernel: iwlwifi 0000:00:14.3: 0x4552414E | FSEQ_ALIVE_TOKEN
2024-03-31 02:08:05.107 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00080400 | FSEQ_CNVI_ID
2024-03-31 02:08:05.107 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000501 | FSEQ_CNVR_ID
2024-03-31 02:08:05.108 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00080400 | CNVI_AUX_MISC_CHIP
2024-03-31 02:08:05.109 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00000501 | CNVR_AUX_MISC_CHIP
2024-03-31 02:08:05.110 homeassistant kernel: iwlwifi 0000:00:14.3: 0x05B0905B | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
2024-03-31 02:08:05.110 homeassistant kernel: iwlwifi 0000:00:14.3: 0x0000025B | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
2024-03-31 02:08:05.112 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00150000 | FSEQ_PREV_CNVIO_INIT_VERSION
2024-03-31 02:08:05.112 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00350002 | FSEQ_WIFI_FSEQ_VERSION
2024-03-31 02:08:05.114 homeassistant kernel: iwlwifi 0000:00:14.3: 0x00350002 | FSEQ_BT_FSEQ_VERSION
2024-03-31 02:08:05.114 homeassistant kernel: iwlwifi 0000:00:14.3: 0x000000C8 | FSEQ_CLASS_TP_VERSION
2024-03-31 02:08:05.116 homeassistant kernel: iwlwifi 0000:00:14.3: WRT: Collecting data: ini trigger 4 fired (delay=0ms).
[... 29 lines cut ...]
2024-03-31 02:08:05.117 homeassistant kernel: ret_from_fork+0x2c/0x50
2024-03-31 02:08:05.117 homeassistant kernel: ? __pfx_kthread+0x10/0x10
2024-03-31 02:08:05.117 homeassistant kernel: ret_from_fork_asm+0x1b/0x30
2024-03-31 02:08:05.117 homeassistant kernel:
2024-03-31 02:08:05.117 homeassistant kernel: wlo1: failed to move station 7c:10:c9:b3:3e:04 to desired state
2024-03-31 02:08:05.117 homeassistant kernel: iwlwifi 0000:00:14.3: Failed to trigger RX queues sync (-5)
2024-03-31 02:08:05.562 homeassistant kernel: iwlwifi 0000:00:14.3: Failed to send flush command (-5)
2024-03-31 02:08:05.562 homeassistant kernel: iwlwifi 0000:00:14.3: flush request fail
2024-03-31 02:08:05.563 homeassistant kernel: iwlwifi 0000:00:14.3: Couldn't send the SESSION_PROTECTION_CMD: -5
2024-03-31 02:08:05.564 homeassistant kernel: iwlwifi 0000:00:14.3: Failed to send MAC_CONTEXT_CMD (action:2): -5
2024-03-31 02:08:05.564 homeassistant kernel: iwlwifi 0000:00:14.3: failed to update MAC b0:dc:ef:7e:f2:ac
2024-03-31 02:08:05.565 homeassistant wpa_supplicant[407]: BSSID 7c:10:c9:b3:3e:04 ignore list count incremented to 3, ignoring for 60 seconds
2024-03-31 02:08:05.565 homeassistant wpa_supplicant[407]: wlo1: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="home_ap5" auth_failures=2 duration=20 reason=CONN_FAILED
2024-03-31 02:08:05.566 homeassistant kernel: iwlwifi 0000:00:14.3: Failed to send binding (action:3): -5
2024-03-31 02:08:05.566 homeassistant kernel: iwlwifi 0000:00:14.3: Failed to send MAC_CONTEXT_CMD (action:2): -5
2024-03-31 02:08:05.568 homeassistant kernel: iwlwifi 0000:00:14.3: failed to update MAC b0:dc:ef:7e:f2:ac
2024-03-31 02:08:05.569 homeassistant kernel: iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
2024-03-31 02:08:05.691 homeassistant kernel: iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x1f
2024-03-31 02:08:05.691 homeassistant kernel: iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
2024-03-31 02:08:05.692 homeassistant kernel: iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
2024-03-31 02:08:05.692 homeassistant kernel: iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x10
2024-03-31 02:08:05.743 homeassistant wpa_supplicant[407]: wlo1: CTRL-EVENT-REGDOM-CHANGE init=DRIVER type=COUNTRY alpha2=RU
2024-03-31 02:08:05.749 homeassistant NetworkManager[396]: [1711850885.7493] device (wlo1): supplicant interface state: associating -> disconnected
2024-03-31 02:08:05.749 homeassistant NetworkManager[396]: [1711850885.7493] device (p2p-dev-wlo1): supplicant management interface state: associating -> disconnected
```

AtomRoman, 2024-04-01:
wrong

arripeo, 2024-04-13:
I know this type of bug, i got in the past the same problem than solved by an new OS release, so when I've seen that in 12.1 release they changed the driver of Wifi I've deiced to NOT UPDATE waiting for a new suitable release. So if you want, please vote my new feature request that cloud be, on my opinion, a strong solution for all these type of problem that are very common

https://community.home-assistant.io/t/interface-by-ui-or-cli-for-installing-wifi-propitary-driver-or-linux-compatible/716527

AtomRoman, 2024-04-17:
without changes OS 12.2

AtomRoman, 2024-10-19:
System information
  IPv4 addresses for enp1s0: 
  IPv4 addresses for wlo1:  192.168.1.34/24
  IPv6 addresses for wlo1:  fe80::d71a:605b:e1f2:64cf/64

  OS Version:               Home Assistant OS 13.2
  Home Assistant Core:      2024.10.3

it worked

closed 2024-10-19 by AtomRoman