home-assistant/operating-system #3290: Vigorf.A virus detected by Windows Security in haos_generic-x86-64-12.1.img.xz

gb53smith, 2024-04-07:
### Describe the issue you are experiencing

Download of 12.1 image is deleted by Windows Security because Trojan: Win32/Vigorf.A virus is detected.
12.0 image is OK.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

unsupported

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.Download the  haos_generic-x86-64-12.1.img.xz file from github on a Windows 11 system
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### Additional information

I want to upgrade the OS from Alpine to Debian using this image.   A workaround may be to install 12.0 and update to 12.1

Impact123, 2024-04-08:
I can reproduce this on windows 10 as well. [The scanners used by VirusTotal can't find anything](https://www.virustotal.com/gui/url/e56a6910b61e0ca142cd6e22ad73207412b9a150b9bc921a71e8776c4167240a).
> I want to upgrade the OS from Alpine to Debian using this image.

 Huh? HAOS is not debian or based on it.

gb53smith, 2024-04-08:
OK, then it is Alpine to HAOS.  Trying to fix a repair notice which says my OS is unsupported.
I think I need 12.1 to support my Z-Wave USB stick.

agners, 2024-04-08:
This seems a similar case to #3249, although, there it was Trojan/Wacatac.B!ml. Similar to that case, this is almost certainly a false positive, as HAOS is not a Win32 binary/also doesn't include such binaries. Searching for Win32/Vigorf.A also suggests that other binaries which got flagged by Windows Defender turned out to be false positives.

kdelios, 2024-04-08:
Same happened to me recently when I tried to download (at windows) a snapshot. tar file from Google drive. I had to disable temporarily windows defender in order to download it.

gb53smith, 2024-04-08:
I suggest that a developer submit the file to:
https://www.microsoft.com/en-us/wdsi/filesubmission
Microsoft should be able to fix this false positive from reoccuring.
I would do so myself but the form requires submission of a file.  In my case, the file disappears
shortly after download.

wnto, 2024-04-09:
Just to record that this has happened to me today, too.

sairon, 2024-04-10:
This is indeed an obvious false positive result - something in the binary sequence triggers that particular's virus detection heuristics. Since the OS image can not be executed on Windows, it doesn't make sense it could be a threat.

Anyway, I have confirmed it is reported like that with latest Defender signatures, while 12.2.rc1 and 12.2.rc2 images are fine. Even with the virus reported, simply selecting "Allow on device" and "Start actions" makes it possible to open and flash the image from Windows, so it's not a show-stopper. Or wait for 12.2 release, which is just around the corner.

closed 2024-04-10 by sairon