home-assistant/operating-system #3295: Constant network traffic without Internet access

chrisduerr, 2024-04-08:
### Describe the issue you are experiencing

I have all my local devices setup to not have Internet access by default. Why I would have loved for HAOS to work with an offline install, I had it enabled during the installation but disabled it afterwards. However after rebooting I noticed that There was constant local network traffic on my HAOS VM. Once I enabled Internet access, the traffic dropped (see screenshot) and stayed at that level even after denying network access again.

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install HAOS
2. Deny Internet Access
3. Reboot
4. Watch network traffic

### Anything in the Supervisor logs that might be useful for us?

```txt
Only warning in the logs is this:

2024-04-08 19:31:32.920 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
```

It doesn't sound significant enough to cause this much traffic and since it goes away after temporarily giving Internet access that's probably not it.
```

### Anything in the Host logs that might be useful for us?

```txt
No relevant host log entries
```

### System information

## System Information

version | core-2024.4.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20-haos
arch | x86_64
timezone | Irrelevant
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | failed to load: unreachable
can_reach_cloud_auth | failed to load: unreachable
can_reach_cloud | failed to load: unreachable

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 9.1 GB
disk_used | 3.7 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | failed to load: unreachable
installed_addons | 

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 7, 2024 at 10:00 PM
-- | --
current_recorder_run | April 8, 2024 at 7:00 PM
estimated_db_size | 2.34 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

agners, 2024-04-08:
This might be related to https://github.com/home-assistant/plugin-dns/pull/134. The next OS release will contain a fixed version of the DNS plug-in by default.

agners, 2024-04-08:
FWIW, Supervisor should already update to it if you give it network access to do so (after a while, or at startup) .

chrisduerr, 2024-04-08:
@agners Is there a way to manually trigger it and/or validate that it has updated? I'm happy to update and check if this still is an issue.

chrisduerr, 2024-04-08:
Network traffic without Internet access seems to have calmed down. Difficult to say if this is due to the reinstall with the VM image, or the latest supervisor version, or if it just temporarily calmed down and will act up again later.

For now I'll assume the issue was fixed in the latest supervisor version and will report back if this turns out to be incorrect.

Thanks for your help @agners.

closed 2024-04-08 by chrisduerr

agners, 2024-04-08:
> Testing will probably take me a while, it seems like updates don't work with the generic image on a virtualized host, so I have to reinstall 

Uh yes, we just recently introduced that detection! Better to reinstall soon.

@chrisduerr thanks for the update!

chrisduerr, 2024-04-08:
> Uh yes, we just recently introduced that detection! Better to reinstall soon.

Honestly might be nice to detect this during install, not just during updates. I didn't have to setup much again and things are pretty simple to setup (good job), but I can imagine a lot of frustration caused by this especially when noticed after weeks/months.

agners, 2024-04-09:
This is a relatively new feature, which only got implemented recently (see https://github.com/home-assistant/supervisor/pull/4968). The Supervisor bundled in the current stable HAOS release (12.1) does not include this version, that is why this time around this message appeared late.

In the future, this will pop-up first thing after the on-boarding.