home-assistant/operating-system #3305: (x86_64) HA OS doesn't boot when updating from 12.1 to 12.2

HAPSagan, 2024-04-12:
### Describe the issue you are experiencing

I see GNU GRUB with 4 options - Slot A, Slot B, Slot A rescue shell, Slot B rescue shell.
Selecting any of them results in a message that it's unable to boot.
I dont get any CLI options.
I have used Linux Reader to download the backups from the disk and have then tried to do a fresh installation with OS12.2. The result is the same as after the update - unable to boot from any slot. After that I did a fresh install with OS 12.1 and everything started fine again.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Do a fresh install with OS 12.2 or update to OS 12.2 and it's unable to boot.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Can´t get a log.
```

### Anything in the Host logs that might be useful for us?

```txt
Can´t get a log.
```

ahmetem, 2024-04-12:
same problem . HA OS doesn't boot when updating from 12.1 to 12.2

Mpgod80, 2024-04-12:
Same issue here. Going from 12.1 to 12.2 then I can´t boot.

Onepamopa, 2024-04-12:
Same here... When the VM reboots after the update - it selects "Slot B" which has "unknown filesystem".
I have to manually select "Slot A" to boot - it boots 12.1 w/o an update.

agners, 2024-04-12:
@HAPSagan @ahmetem @Mpgod80 you all are running on native x86-64 hardware? What hardware are you using?

@Onepamopa can you open a new issue along with information of your virtualization environment? It also seems yours behaved different as the old boot slot still worked (unlike the case OP reported).

ahmetem, 2024-04-12:
Its features are below. It was working smoothly until the last update.
tried to do a fresh installation with OS12.2. but the result did not change.
Slot A, Slot B, Slot A rescue shell, Slot B rescue shell .none of them work.

MacBook Air (11-inch, Mid 2011)   a1370
64GB flash storage
1.6GHz dual-core Intel Core i5 
2GB  of 1333MHz DDR3 onboard memory
Advanced Intel HD Graphics 3000.

agners, 2024-04-12:
On a new installation, what happens exactly when you choose Slot A with HAOS 12.2? Anything written on the screen? Black screen? Reset?

Mpgod80, 2024-04-12:
@agners 
The hardware both me and HAPSagan is runing on is: 

Intel Atom CPU D2500 @ 1,86Ghz 4,0 Gb RAM 60gd SSD.
Runing HASOS img directly with an image flashed by Balena Etcher.

Onepamopa, 2024-04-12:
> @HAPSagan @ahmetem @Mpgod80 you all are running on native x86-64 hardware? What hardware are you using?
>
> @Onepamopa can you open a new issue along with information of your virtualization [...]

My environment is a bit weird... 
Proxmox 6.x (EOL), LVM storage only (no lvm-thin or disk storage).
I had to convert the qcow2 image into a "raw" LVM disk (which I also resized from 32G to 60G).

qm importdisk <VMID> <THE_DOWNLOADED_DISK>.qcow2 <Target_LVM_Storage> --format raw
qm resize <VMID> virtio1 +28G  (where virtio1 is the imported disk)

After this - the VM booted, HA utilized the whole storage perfectly.

Restored my backup, all good. All updates work fine, apart from the OS 12.2 one...

henrikcaesar, 2024-04-12:
Same here 😱 Also running an old Intel atom with cpu integrated on the motherboard. Can’t get any info out of the system. If it would help I can get a live  Linux usb.

ahmetem, 2024-04-12:
i boot ha with Super Grub2 Disk image on usb disk. show all boot menu.  and selected slot a. it open normaly.  but i dont know How do I make this permanent? or fix it.
i think re install grub.  or maybe downgrade 12.1 fix it.

coc, 2024-04-12:
Same problem on a ThinkCentre m93p.

SellSan, 2024-04-13:
I Have HP T630 and the same issue. The only I can do I can run command from this window and can edit these 4 options parameters. 
[image]

Onepamopa, 2024-04-13:
> I Have HP T630 and the same issue. The only I can do I can run command from this window and can edit these 4 options parameters. ![20240413_085029](https://private-user-images.githubusercontent.com/121897721/322184124-fb226ed9-d856-42c9-a6cf-3f15cf110c1c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI5OTE2NDIsIm5iZiI6MTcxMjk5MTM0MiwicGF0aCI6Ii8xMjE4OTc3MjEvMzIyMTg0MTI0LWZiMjI2ZWQ5LWQ4NTYtNDJjOS1hNmNmLTNmMTVjZjExMGMxYy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxM1QwNjU1NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYzUzNWZhODBkZTc0YzBhMTcxOGYxYjkxYzZhZDJmZTA0NzViZjRhZTRkMTcyMTQzMjc1YWViODYxNTc5N2I3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kDRnrF4q5RiTYr1Ks_Z708Ddr43nz3V9EgM0eOb2Tuo)

I have the same screen but when I select "Slot A" it boots into 12.1. Have you tried?

Mpgod80, 2024-04-13:
> I have the same screen but when I select "Slot A" it boots into 12.1. Have you tried?

Tried that but that wont work for me :/

SellSan, 2024-04-13:
> I have the same screen but when I select "Slot A" it boots into 12.1. Have you tried?

I tried, none of the options worked...

Omri-Kleynhans, 2024-04-13:
Even downloading haos_generic-x86-64-12.2.img and created new installation does not work. Boot not possible. Same error as afterupgrade from 12.1 to 12.2

Onepamopa, 2024-04-13:
So... they screwed up this update, bad... :) Let's see who'll take the blame.

agners, 2024-04-13:
It sounds like a GRUB2 issue of some kind. That got updated from GRUB 2.06 to 2.12 as far as I can tell.

FWIW, the rc (and release) versions have been tested and are working fine on various Intel NUC systems. It seems a particular hardware/BIOS which causes issue.

It kinda reminds me of that bug fix we added in 8.0.rc4 (https://github.com/home-assistant/operating-system/issues/1830#issuecomment-1119718325), but it seems that is applied upstream in GRUB 2.12 today, so that should no longer be the problem :thinking: 

Can you replace the U-Boot binary (on the first partition of the disk at `/EFI/BOOT/bootx64.efi`) with the version from HAOS 12.1 and see if this boots?

Omri-Kleynhans, 2024-04-13:
> It sounds like a GRUB2 issue of some kind. That got updated from GRUB 2.06 to 2.12 as far as I can tell.
> 
> FWIW, the rc (and release) versions have been tested and are working fine on various Intel NUC systems. It seems a particular hardware/BIOS which causes issue.
> 
> It kinda reminds me of that bug fix we added in 8.0.rc4 ([#1830 (comment)](https://github.com/home-assistant/operating-system/issues/1830#issuecomment-1119718325)), but it seems that is applied upstream in GRUB 2.12 today, so that should no longer be the problem 🤔
> 
> Can you replace the U-Boot binary (on the first partition of the disk at `/EFI/BOOT/bootx64.efi`) with the version from HAOS 12.1 and see if this boots?
Yes have done that, replaced all files at `/EFI/BOOT/'. It  does work. Server can boot again HAS back up. Thanks

SellSan, 2024-04-13:
> Yes have done that, replaced all files at `/EFI/BOOT/'. It does work. Server can boot again HAS back up. Thanks

Hi, could you share where from I can download that file?

pierrepaap, 2024-04-13:
same issue.
hardware :  Intel D525MW motherboard (bios from 2010)

I'm surprised though that the 'backup' entry also does not start. I would assume that the EFI file would still be from 12.1 ?

Onepamopa, 2024-04-13:
So... any viable solution to update to 12.2 ? Or we should wait for 12.3..

pierrepaap, 2024-04-14:
> So... any viable solution to update to 12.2 ? Or we should wait for 12.3..

well...
after replacing the uboot efi with the one of the 12.1 image I am not in 12.2
Obivously a hassle. So 'im going to stay like this until this bug is fixed
```
 Core 2024.4.2
Supervisor 2024.04.0
Operating System 12.2
Frontend 20240404.1
```

Onepamopa, 2024-04-14:
> well... after replacing the uboot efi with the one of the 12.1 image I am not in 12.2 Obivously a hassle. So 'im going to stay like this until this bug is fixed
>
> ```
>  Core 2024.4.2
> Supervisor [...]

Ugh.... it says you're on 12.2 ?

pierrepaap, 2024-04-14:
> Ugh.... it says you're on 12.2 ?

typo... I am NOW in 12.2

agners, 2024-04-15:
> I'm surprised though that the 'backup' entry also does not start. I would assume that the EFI file would still be from 12.1 ?

We only ship a single boot loader. In a way, we rely on the bootloader to implement the backup boot method. But if that new bootloader fails in some ways, it is essentially game over :cry: 

> Hi, could you share where from I can download that file?

These are the two U-Boot boot loader files from HAOS 12.1 (replace the existing ones in /EFI/BOOT on the first boot partition with these files):
[grub-2.06-haos-12.1.zip](https://github.com/home-assistant/operating-system/files/14975625/grub-2.06-haos-12.1.zip)

Are those affected UEFI BIOS'es maybe 32-bit BIOSes? E.g. can someone check if only replacing `bootia32.efi` helps?

agners, 2024-04-15:
> Same here... When the VM reboots after the update - it selects "Slot B" which has "unknown filesystem".
> I have to manually select "Slot A" to boot - it boots 12.1 w/o an update.

@Onepamopa your case is different. Anyone else here can't boot either slot. In your case it is not the boot loader failing, but somehow that new boot slot did not get written properly. Also your's is a VM (ova image), whereas everyone else is runnning the `generic-x86-64` image on native hardware. Again, *please* open a new issue for your case. It is very hard for us to track bugs when folks mix independent issue in a single issue report.

Onepamopa, 2024-04-15:
> @Onepamopa your case is different. Anyone else here can't boot either slot. In your case it is not the boot loader failing, but somehow that new boot slot did not get written properly. Also your's [...]

Yes, I downloaded haos_ova-12.1.qcow2.xz however the symptoms look like they come from the exact same issue - something in the bootloader of 12.2 caused this, don't you agree?

Also, considering maybe I'm the only one experiencing this *all-so-similar* issue with a VM, that wouldn't generate much attention now, would it?

agners, 2024-04-15:
> Yes, I downloaded haos_ova-12.1.qcow2.xz however the symptoms look like they come from the exact same issue - something in the bootloader of 12.2 caused this, don't you agree?

It points to 12.2, I am not sure about the bootloader part: The new boot loader seems to be still able to boot the old boot slot. So the boot loader _per-se_ is capable of booting a HAOS. This is fundamentally different from the other reports: The bootloader seems not to be able to boot even the old boot slot.

Also, when you read the other messages, people who replace just the boot loader are able to boot into 12.2 even then. So for them 12.2 works, it is just the bootloader which makes troubles.

> Also, considering maybe I'm the only one experiencing this all-so-similar issue with a VM, that wouldn't generate much attention now, would it?

I do have ideas how to debug your case, but clutter this issue with logs from a different case is just not the way it works. So I don't ask you here. This is about a different issue. If you want me seriously look into your case, then you'll have to open a new issue.

Shad0wguy, 2024-04-15:
Same issue here. Both Slot A and B show error: cannot load image. Also running on an Atom based system as others have reported.

Shad0wguy, 2024-04-15:
> > I'm surprised though that the 'backup' entry also does not start. I would assume that the EFI file would still be from 12.1 ?
> 
> We only ship a single boot loader. In a way, we rely on the bootloader to implement the backup boot method. But if that new bootloader fails in some ways, it is essentially game over 😢
> 
> > Hi, could you share where from I can download that file?
> 
> These are the two U-Boot boot loader files from HAOS 12.1 (replace the existing ones in /EFI/BOOT on the first boot partition with these files): [grub-2.06-haos-12.1.zip](https://github.com/home-assistant/operating-system/files/14975625/grub-2.06-haos-12.1.zip)
> 
> Are those affected UEFI BIOS'es maybe 32-bit BIOSes? E.g. can someone check if only replacing `bootia32.efi` helps (so maybe this is related to #1752)?

I believe my system (Atom D525) does have a 32-bit bios even though the cpu is 64-bit. Could that be the cause?

Shad0wguy, 2024-04-15:
> We only ship a single boot loader. In a way, we rely on the bootloader to implement the backup boot method. But if that new bootloader fails in some ways, it is essentially game over 😢
>
>
> These are [...]

I booted into a live image on my HAOS pc and mounted the disk. None of the partitions seem to have a /EFI/BOOT path.

sairon, 2024-04-15:
> I believe my system (Atom D525) does have a 32-bit bios even though the cpu is 64-bit. Could that be the cause?

The discrepancy itself isn't really problematic but in this particular case yes - it seems that this issue only affects systems that only have 32bit UEFI support - thus the request to check if replacing just the `bootia32.efi` helps (while `bootx64.efi` shouldn't be needed at all). 32bit UEFI isn't (thankfully) that common and there are probably more factors involved that caused the issue to be discovered only now.

> I booted into a live image on my HAOS pc and mounted the disk. None of the partitions seem to have a /EFI/BOOT path.

The partition might be hidden from file managers because of the flags it has set but it should be always present - it's always the first 32 MiB partition and you should be able to mount it with e.g. with `sudo mount /dev/sdb1 /mountpoint` (you might need to replace `sdb` with the disk's identifier and `/mnt/boot` with an empty/different folder.

Shad0wguy, 2024-04-15:
> The discrepancy itself isn't really problematic but in this particular case yes - it seems that this issue only affects systems that only have 32bit UEFI support - thus the request to check if [...]

You were right. I had to mount sda1. I tried copying just the bootia32.efi but that gave the same result. I then added bootx64.efi as well and it booted. Cant access the web gui though.

agners, 2024-04-15:
> You were right. I had to mount sda1. I tried copying just the bootia32.efi but that gave the same result. I then added bootx64.efi as well and it booted. Though I am back on 12.1.

Hm, so it seems to be the 64-bit GRUB in your case then :thinking: 

You probably can boot into 12.2 by manually select the other slot at bootup.

SellSan, 2024-04-16:
> We only ship a single boot loader. In a way, we rely on the bootloader to implement the backup boot method. But if that new bootloader fails in some ways, it is essentially game over 😢
>
>
> These are [...]

I replaced the bootx64.efi file and system started, now I have 12.1 version.

sairon, 2024-04-16:
We're trying to gather as much information as possible and get a device where the issue can be reproduced, as this one will be harder to debug remotely and simply reverting the upgrade is not viable in the long term. From the reports here, these need some clarification:

@HAPSagan @Mpgod80 - you only mentioned `Intel Atom CPU D2500` - can you please clarify what board/device is it on?

@henrikcaesar - as well, you only said it's an "old Intel Atom motherboard" - could you please be more specific?

@Omri-Kleynhans - you shared no details on the hardware used, could you do so, please?

@SellSan - your device is an odd-ball here, as it shows the same symptoms but has a different CPU than all those Intel Atom-based devices (AMD GX-420GI). Would it be possible to check if a fresh install of HAOS 12.2 doesn't really boot? Also can you check what BIOS versions is it running? I remember seeing more people running these, so I wonder there are no more reports around. Also anyone else running the same device can chime in.

@coc - you stated ThinkCentre m93p which should be an Intel i5 4th Gen. Does it indeed show the same symptoms and only replacing the EFI files helped?

@JitteryDoodle - same as the above - it's an Intel i5 and booting into the other slot should not be possible if the cause is the same as discussed in the issue. Ideally please try performing another upgrade, and if it fails again, open a new issue with logs from the previous boot (`ha host logs -b-1 -n 10000`).

henrikcaesar, 2024-04-16:
@sairon found the 📦 D525MW board with Intel Atom D525, from 2010 it seems. 

[image]

Shad0wguy, 2024-04-16:
> @sairon found the 📦 D525MW board with Intel Atom D525, from 2010 it seems.
> 
> ![IMG_8564](https://private-user-images.githubusercontent.com/1260498/322840986-a1159c87-80e3-4e2c-8fc5-606ee8422be5.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMyNzE1MTQsIm5iZiI6MTcxMzI3MTIxNCwicGF0aCI6Ii8xMjYwNDk4LzMyMjg0MDk4Ni1hMTE1OWM4Ny04MGUzLTRlMmMtOGZjNS02MDZlZTg0MjJiZTUuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE2VDEyNDAxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzMTI5YTNhNjk2NmYyODA0MzM2OGZhODM4YWMwOGI1ODc4YTNiNjE0NDA0OWY3YzBmMDVlY2U5OTY3YWY4MmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.eP0NUH--x_mRS7ynxSEoVKZO2RXgQU0J9gjgmiw5KRU)

This is the same board I have with this issue.

sairon, 2024-04-17:
@aman-sandhu That is strange, because another user in #3313 is using T620 and update to 12.2 "only" broke his WiFi.

@bearhntr, sorry for the tag here, but maybe you could check with @aman-sandhu if there are any differences between your setups? Especially the BIOS version might have the biggest impact. Or maybe the boot device used?

Mpgod80, 2024-04-17:
@sairon I have found this about the motherboard: Intel D2500CC  AAG81477-401, from  OCT 2013

sairon, 2024-04-17:
@bearhntr

>  I replaced my card with the AX210 ...

That should not make a difference here. In short - this issue is that for some people the HP T620 (and HP T630, which seems quite similar in specs) fails to boot completely. This appears to be UEFI firmware issue, that's why I am curious if we could find any differences in the BIOS firmware version or in the setup (e.g. booting from the internal vs. external drive) that would explain it.

SellSan, 2024-04-17:
@sairon 
I set up my haos and really don't want to play with it as I like when everything is fine, and I don't have time to play when it brake, so I'm glad I could replace onle one file and make it working. I have my set up done and expecting all works fine, without touching it. If it helps I took photos of the BIOS settings, maybe this will be of any help for you.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

bearhntr, 2024-04-17:
> @bearhntr
>
>
> That should not make a difference here. In short - this issue is that for some people the HP T620 (and HP T630, which seems quite similar in specs) fails to boot completely. This appears [...]

When I installed it - I went into the BIOS (I have the lasted installed sp115305.exe [version L40_0219].  I restored Factory Defaults, and then disabled SERIAL, PRINTER, SATA port not used by the M.2 SSD in there and the internal USB ports.  I also disabled anything related to LEGACY and cleared the TPM stuff and disabled SECURE BOOT.  

I installed it by using RUFUS to write the image to the SSD (while it was out of the machine).  I then plugged it in and booted it - took about 20 mins to complete and then I got the web page.  Been running this way for nearly 2 years.

aman-sandhu, 2024-04-18:
> @aman-sandhu That is strange, because another user in #3313 is using T620 and update to 12.2 "only" broke his WiFi.
>
> @bearhntr, sorry for the tag here, but maybe you could check with @aman-sandhu if [...]

I don't use Wi-Fi, and I don't have a Wi-Fi/BT card installed. I have an Ethernet connection. After updating to 12.2, it's giving me the same behavior as OP as below:
I see GNU GRUB with 4 options - Slot A, Slot B, Slot A rescue shell, Slot B rescue shell.
Selecting any of them results in a message that it's unable to boot.
I dont get any CLI options.

sairon, 2024-04-18:
@SellSan Well, the last image shows it is not HP T630 but HP T610, with an entirely different CPU. Quite a difference. However, it also shows the BIOS is at the latest version available from HP, so it's unlikely an update will fix that. The issue is that even though you replaced the file, another OS update will break it again, unless we find the root cause of the problem.

@aman-sandhu Can you also check the BIOS version in the System Information (i.e. as on last picture above)? The version should be `00.02.19` (or similar). If not, try updating to the latest BIOS first, as this one seems to work correctly.

ChernyaevAN, 2024-04-19:
I have the same problem with this motherboard and processor.
[DN2800MT_TechProdSpec05.pdf](https://github.com/home-assistant/operating-system/files/15038883/DN2800MT_TechProdSpec05.pdf)
I use mSATA SSD, but USB flash drive with HAOS have the same error.

mwevromans, 2024-04-19:
Same issue here. HP T620 wiht a AMD GX-415GA Processor

dunka, 2024-04-20:
I have the same problem on a atom D2500, probably also grub, I rolled back to 11.5 and things are fine.

aman-sandhu, 2024-04-20:
> @aman-sandhu Can you also check the BIOS version in the System Information (i.e. as on last picture above)? The version should be `00.02.19` (or similar). If not, try updating to the latest BIOS first, as this one seems to work correctly.

I have updated the BIOS on my hp thin client t620 to the version L40 00.02.19. Its all working fine now. Thank you for your help.

feinadam, 2024-04-20:
Same issue with Intel® Desktop Board D2700DC.

AlexanderSmi, 2024-04-20:
Same problem. Intel D2500HN motherboard, SSD - tried different ones. The haos_generic-x86-64-12.2.img image was written by balenaEtcher. BIOS - I also tried all the versions that I could find. BIOS settings - checked all reasonable ones. The result, unfortunately, is negative.

MrBasque, 2024-04-22:
> I have updated the BIOS on my hp thin client t620 to the version L40 00.02.19. Its all working fine now. Thank you for your help.

Thank you @sairon, updating BIOS (L40 0219)  + resetting BIOS settings to default did the trick for my HP 620T

SellSan, 2024-04-22:
> @SellSan Well, the last image shows it is not HP T630 but HP T610, with an entirely different CPU. Quite a difference (EDIT: sorry, I notice you corrected your original post few days later - but I still remembered it was T630). However, it also shows the BIOS is at the latest version available from HP, so it's unlikely an update will fix that. The issue is that even though you replaced the file, another OS update will break it again, unless we find the root cause of the problem.
> 
> @aman-sandhu Can you also check the BIOS version in the System Information (i.e. as on last picture above)? The version should be `00.02.19` (or similar). If not, try updating to the latest BIOS first, as this one seems to work correctly.

Sorry for that I made mistake I have HP T610. I'm totally green and don't know much, but maybe it is something related to the BIOS settings explained in this YT [video](https://youtu.be/zJ64YFDrveU?feature=shared&t=237).

pierrepaap, 2024-04-22:
any news for the Intel D525MW board ?
are we aiming at a fix upstream for those ?

IMDvevey, 2024-04-23:
> We're trying to gather as much information as possible and get a device where the issue can be reproduced, as this one will be harder to debug remotely and simply reverting the upgrade is not viable [...]

We're having the same issue with a similar Atom board with integrated components. The exact board was taken from a Fibaro HomeCenter 2 (It's an Intel DN2800MT Desktop Board).

After updating from 12.1 to 12.2, the HA was no longer accessible and after looking, had the same issue stated above where all four boot options (Slot A, B, Rescue A and B) were not booting (No bootable image) or seeming to try to boot but having a black screen with a fixed typing indicator, not being able to actually type anything.

I'm not really great with Linux so I tried copying the folders within the "Supervisor" folder and restoring them to a fresh install of 12.2 (The fresh install works great), at which point the Home Assistant UI is accessible but unusable (see log file of all components freaking out).
[home-assistant_2024-04-23T12-15-39.892Z.log](https://github.com/home-assistant/operating-system/files/15077104/home-assistant_2024-04-23T12-15-39.892Z.log)

Any help would be appreciated if you guys have any idea.

agners, 2024-04-23:
@IMDvevey you can recover the device by just overwriting the boot loader on the first partition (see https://github.com/home-assistant/operating-system/issues/3305#issuecomment-2055918375).

sairon, 2024-04-23:
The common factor of (almost) all those boards is the Intel NM10 chipset. I managed to get a D525MW boards for testing and found a commit that is breaking it, unfortunately there is no BIOS update that could resolve the issue in the firmware, so until the fix is resolved upstream, the only option is reverting the patch. Thankfully it doesn't seem to be that important for any other functionality, so it might be a way to run GRUB 2.12 on the problematic boards.

Here's a patched version of the `bootx64.efi`, please give it a try by replacing the current one in `/EFI/BOOT/` in the first boot partition:
[grub_2.12-patched.zip](https://github.com/home-assistant/operating-system/files/15078787/grub_2.12-patched.zip)

@SellSan @mwevromans Could you please try if the patched GRUB also loads the system on yours HP T610/T620? Since it's a different chipset, there is possibility it might be a different bug, but let's hope not. Please note that without verifying this, another OS update will break your installation again, so it's worth checking we know what's needed to be fixed. 

---

Btw. when testing the board, I also measured its power consumption which is around 16 Watts when the device is sitting idle :exploding_head: That is an order of magnitude higher than RPi, HA Green, or any modern SBC, that also has ~3-4 times higher computing power. Keep it in mind when calculating the total operating costs of running such old hardware :)

Shad0wguy, 2024-04-23:
@sairon So the bug is in Grub 2.12? Is there a bug report there that we can follow?

As for power, I used this just because I had it sitting around already doing nothing. Definitely will get something more power efficient when it dies.

ahmetem, 2024-04-23:
i dont now my macbook air chipset or bios settings. 

MacBook Air (11-inch, Mid 2011) a1370
64GB flash storage
1.6GHz dual-core Intel Core i5
2GB of 1333MHz DDR3 onboard memory
Advanced Intel HD Graphics 3000.

[55 more replies]

PR #3324 (2024-04-25): Revert GRUB2 patch causing boot failure on some old x86_64 BIOSes
GRUB 2.12 release contains a change of the loader [1] used for loading the kernel on x86_64 platform. This change was identified to cause boot failure on some old Intel Atom boards with the NM10 chipset, and possibly some others. Revert this patch before we get a more proper fix for the issue.

[1] https://git.savannah.gnu.org/cgit/grub.git/commit/?id=cfbfae1aef0694b416aa199291cfef7596cdfc20

Fixes #3305
commit 478538ea (2024-04-25): Revert GRUB2 patch causing boot failure on some old x86_64 BIOSes (#3324)

GRUB 2.12 release contains a change of the loader [1] used for loading
the kernel on x86_64 platform. This change was identified to cause boot
failure on some old Intel Atom boards with the NM10 chipset, and
possibly some others. Revert this patch before we get a more proper fix
for the issue.

[1] https://git.savannah.gnu.org/cgit/grub.git/commit/?id=cfbfae1aef0694b416aa199291cfef7596cdfc20

Fixes #3305
closed 2024-08-06 by sairon