home-assistant/operating-system #3306: v4l2-related tracedump appears in kernel log (dmesg) during boot

jose1711, 2024-04-13:
### Describe the issue you are experiencing

During boot process the system logs the following lines:

```
$ dmesg
..
..
[   67.600202] WARNING: CPU: 0 PID: 3174 at drivers/media/common/videobuf2/videobuf2-core.c:1627 vb2_start_streaming+0xec/0x160 [videobuf2_common]
[   67.600218] Modules linked in: rfcomm nft_chain_nat nf_conntrack_netlink xfrm_user xfrm_algo nft_compat nf_tables nfnetlink algif_hash algif_skcipher af_alg bnep sch_fq_codel vc4 snd_soc_hdmi_codec drm_display_helper btsdio cec spidev ov5647 hci_uart drm_dma_helper btqca drm_kms_helper brcmfmac btrtl snd_soc_core brcmutil btbcm rp1_cfe pisp_be aes_ce_blk btintel aes_ce_cipher rpivid_hevc(C) snd_compress snd_pcm_dmaengine bluetooth snd_pcm cfg80211 ghash_ce snd_timer snd v4l2_mem2mem fb_sys_fops sha2_ce ecdh_generic syscopyarea videobuf2_dma_contig sysfillrect ecc sysimgblt videobuf2_memops libaes videobuf2_v4l2 rfkill videobuf2_common sha256_arm64 sha1_ce i2c_designware_platform gpio_keys spi_bcm2835 raspberrypi_hwmon i2c_designware_core raspberrypi_gpiomem nvmem_rmem pwm_fan rp1_adc uio_pdrv_genirq uio drm drm_panel_orientation_quirks backlight fuse
[   67.600273] CPU: 0 PID: 3174 Comm: ml1:Camera1 Tainted: G         C         6.1.73-haos-raspi #1
[   67.600276] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[   67.600277] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   67.600280] pc : vb2_start_streaming+0xec/0x160 [videobuf2_common]
[   67.600288] lr : vb2_start_streaming+0x74/0x160 [videobuf2_common]
[   67.600296] sp : ffffffc00a343b60
[   67.600297] x29: ffffffc00a343b60 x28: ffffffcfe03b4bb8 x27: 0000000000000000
[   67.600300] x26: 0000000000000000 x25: ffffff8107fa8820 x24: ffffff810d086200
[   67.600303] x23: ffffff815170cc00 x22: 0000000040045612 x21: ffffff8107fa8790
[   67.600306] x20: ffffff8107fa85b8 x19: 00000000ffffffea x18: ffffffc00a343500
[   67.600309] x17: 0000000000000000 x16: ffffffd00cd6c69c x15: ffffffc08a3436b7
[   67.600311] x14: 0000000000000000 x13: 2e64656c62616e65 x12: 20746f6e20736920
[   67.600314] x11: 6b6e696c2065646f x10: 6e203068635f3269 x9 : ffffffd00c1ea1e8
[   67.600317] x8 : 00000000ffffefff x7 : 00000000fffff000 x6 : 0000000000000000
[   67.600320] x5 : ffffff81fef18a10 x4 : ffffff81fef18a10 x3 : 0000000000000000
[   67.600322] x2 : 0000000000000000 x1 : ffffffcfdfe5e000 x0 : 0000000000000004
[   67.600326] Call trace:
[   67.600327]  vb2_start_streaming+0xec/0x160 [videobuf2_common]
[   67.600335]  vb2_core_streamon+0x90/0x198 [videobuf2_common]
[   67.600343]  vb2_ioctl_streamon+0x54/0x98 [videobuf2_v4l2]
[   67.600352]  v4l_streamon+0x28/0x40
[   67.600358]  __video_do_ioctl+0x178/0x3fc
[   67.600362]  video_usercopy+0x21c/0x790
[   67.600365]  video_ioctl2+0x20/0x40
[   67.600368]  v4l2_ioctl+0x48/0x70
[   67.600370]  __arm64_sys_ioctl+0xb0/0x100
[   67.600374]  invoke_syscall+0x50/0x120
[   67.600378]  el0_svc_common.constprop.0+0xd4/0x100
[   67.600381]  do_el0_svc+0x34/0xd0
[   67.600383]  el0_svc+0x2c/0x84
[   67.600387]  el0t_64_sync_handler+0xf4/0x120
[   67.600390]  el0t_64_sync+0x18c/0x190
[   67.600392] ---[ end trace 0000000000000000 ]---
[   67.604510] rp1-cfe 1f00110000.csi: csi2_ch0 node link is not enabled.
[   67.604701] ------------[ cut here ]------------
```
The message repeats several times. I think this may be the reason why HA OS is not detecting a locally connected Pi camera while it works fine in the latest Raspbian on the same board.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Boot the system, check the logs.

### Anything in the Supervisor logs that might be useful for us?

```txt
not relevant
```

### Anything in the Host logs that might be useful for us?

```txt
not relevant
```

### System information

## System Information

version | core-2024.4.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | Europe/Prague
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 13. februára 2025 o 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-17.ui.nabu.casa
certificate_status | ready
instance_id | fd9acab5e43e4efd85ff0db93524adfe
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.04.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 28.7 GB
disk_used | 8.1 GB
healthy | true
supported | true
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (17.2.0), motionEye (0.20.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 0
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 4. apríla 2024 o 15:59
-- | --
current_recorder_run | 13. apríla 2024 o 22:15
estimated_db_size | 127.01 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

sairon, 2024-04-15:
What kernel version have you tested with on Raspberry Pi OS? It's a shot in the dark but you can try the [latest dev build](https://os-artifacts.home-assistant.io/index.html) to see if it's resolved - it is using Linux 6.6.20, which should be the same the current Pi OS is using.

Also, how do you use the camera? I think it won't work out of the box without any changes in `config.txt` and such, so adding more details about the setup would be helpful.

jose1711, 2024-04-15:
Thank you for your response.

> What kernel version

HA OS: `Linux a0d7b954-ssh 6.1.73-haos-raspi #1 SMP PREEMPT Thu Apr 11 10:33:23 UTC 2024 aarch64 Linux`
Raspbian: `Linux raspberrypi 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux`

I'll try to extract and replace required files in `/boot` partition from the dev build and report back if there is any improvement.

> how do you use the camera?

As per https://gist.github.com/enegaard/a57af286205914bd912270c89650fb1b I added the following lines to `config.txt`:
```
[all]
start_x=1
gpu_mem=128
```

In Raspbian `cam -l` returns the following output:
```
[0:00:37.489576290] [1931]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:00:37.501517975] [1935]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 14-02-2024 (14:00:12)
[0:00:37.516460475] [1935]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Available cameras:
1: 'ov5647' (/base/axi/pcie@120000/rp1/i2c@88000/ov5647@36)
```

I cannot reproduce the same in HA OS:
```
$ apk add libcamera-tools; cam -l
..
[0:08:16.035800835] [667]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0
Available cameras:
$
```

jose1711, 2024-04-15:
Replacing the files in the startup partition resulted in a boot loop so I went with a fresh install on another SD card. The tracedump is gone, it even seems that picamera is recognized:
```
$ dmesg
..
[    2.164699] rp1-cfe 1f00110000.csi: Using sensor ov5647 6-0036 for capture
[    2.165212] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-csi2_ch0] node id 0 successfully as /dev/video0
[    2.165708] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-embedded] node id 1 successfully as /dev/video1
[    2.166262] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-csi2_ch2] node id 2 successfully as /dev/video2
[    2.166896] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-csi2_ch3] node id 3 successfully as /dev/video3
[    2.170809] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-fe_image0] node id 4 successfully as /dev/video4
[    2.171422] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-fe_image1] node id 5 successfully as /dev/video5
[    2.172957] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-fe_stats] node id 6 successfully as /dev/video6
```
But unfortunately neither `cam` nor `motioneye` are able to use it. Not sure if this an HA OS limitation or something can be done about the camera support.

closed 2024-07-22 by github-actions[bot]