home-assistant/operating-system #3307: RPI installer using old version

MikeTango, 2024-04-15:
### Describe the issue you are experiencing

Hi,

i updated the RPI image installer on Xubuntu to latest (1.8.5) but it still doesn't let me install 12.2 in the dropdown menu. 

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Download RPI installer
2. Click the things
3. ???
...

### Anything in the Supervisor logs that might be useful for us?

```txt
nope
```

### Anything in the Host logs that might be useful for us?

```txt
nope
```

agners, 2024-04-15:
Nice catch! It seems a change in the version repository broke the image json update (see https://github.com/home-assistant/version/actions/runs/8647310063). I'll have a look.

It does seem with #3308 we have another issue on Raspberry Pi though, so maybe it is better to hold off the update anyways.

MikeTango, 2024-04-15:
\o/ thanks for keeping this ship afloat! Workaround for me was installing the correct image from github via rpi-installer - custom images after unzipping the xz which didn't write the sd card as intended.

closed 2024-05-13 by agners