home-assistant/operating-system #3308: usb_storage.quirks: string doesn't fit in 127 chars

jens-maus, 2024-04-15:
### Describe the issue you are experiencing

With the latest version of HAos on RaspberryPi the `usb_storage.quirks` kernel parameter does not seem to fit into the maximum length (127 bytes) limit anymore.

This can be observed with looking at the bootup messages in `dmesg`:

```
[    0.000000] usb_storage.quirks: string doesn't fit in 127 chars.
[    0.000000] Booting kernel: `174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u,174c:225c:u,7825:a2a4:u,152d:0562:u,125f:a88a:u' too large for parameter `usb-storage.quirks'
```

Thus, the current usb-storage.quirks string is 132 characters long and thus exceeds the limit. Therefore it seems that the whole usb-storage.quirks mechanism is not applied on the RaspberryPi platform versions.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Bootup HAos
2. look at `dmesg` output
3. notice the usb_storage.quirks errors/notices of exceeding 127 characters.

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
See above
```

### System information

n/a

jens-maus, 2024-04-15:
Please note that this issues seems to be discussed also in the raspberrypi fora, here:

https://forums.raspberrypi.com/viewtopic.php?t=338094

There a kernel patch is suggested to increase the 127 character limit, thought.

agners, 2024-04-15:
Ugh! Yeah I guess adding the kernel patch for now makes most sense.

Some PID/VIDs are actually "quirked" in the driver (as in the driver disables UAS). I was actually wondering if these cases are really a USB host interface <> USB device interaction issue, or if they could just be added to those quirks? Maybe the quirks need to be smarter, and quirk based on USB host interface the device is attached to? :thinking:

jens-maus, 2024-04-15:
FYI: See here for the intermediate kernel patch I added to my RaspberryMatic project to fix that issue there:

https://github.com/jens-maus/RaspberryMatic/commit/a5c92c9c051b11a669b346c22444256502679f07

sairon, 2024-04-15:
Thanks for catching this, I opened a PR with the same fix for HAOS. Like I wrote in the PR description, only new installs will be affected (on the other hand, old installs don't get the latest updates of quirks with update), and the future kernel will fix this, so hopefully the impact won't be that big.

PR #3311 (2024-04-15): Increase maximum length of usb_storage.quirks in RPi kernel params
With #3281 we hit the maximum length of the quirks parameter. Since cmdline.txt changes are not applied on OS update, only new installs of 12.2 are affected, effectively disabling all quirks until this patch lands in future OS release.

Fixes #3308
commit 77c28be3 (2024-04-15): Increase maximum length of usb_storage.quirks in RPi kernel params (#3311)

With #3281 we hit the maximum length of the quirks parameter. Since
cmdline.txt changes are not applied on OS update, only new installs of
12.2 are affected, effectively disabling all quirks until this patch
lands in future OS release.

Fixes #3308
closed 2024-04-15 by sairon