home-assistant/operating-system #3309: (OVA) HA OS unable to be updated from 12.1 to 12.2

Onepamopa, 2024-04-15:
### Describe the issue you are experiencing

The update from 12.1 to 12.2 "succeeds", but:
When the VM reboots after the update - it pre-selects "Slot B" which has "unknown filesystem".
I can manually select "Slot A" to boot - it boots 12.1 w/o an update to 12.2.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update from 12.1 to 12.2
2. Wait for it to reboot
3. Open VNC/KVM/etc and observe.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing in any of the HAOS logs indicates an issue.
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing in any of the Host logs indicates an issue.
```

### System information

Proxmox 6.x (Yes, I know it's EOL), LVM storage only (no lvm-thin or disk storage).
I had to convert the qcow2 image into a "raw" LVM disk (which I also resized from 32G to 60G).

qm importdisk VMID THE_DOWNLOADED_DISK.qcow2 Target_LVM_Storage --format raw
qm resize VMID virtio1 +28G (where virtio1 is the imported disk)

After this - the VM booted, HA utilized the whole storage:
df -h |less
[image]
[image]
[image]

And
mount |less
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Restored my backup, all good. All updates work fine, apart from the OS 12.2 one...

Here's the VM config:
```
agent: 1,type=virtio
balloon: 0
bios: ovmf
boot: order=ide2;virtio1;net0
cores: 4
cpu: host,flags=+aes
cpuunits: 150000
efidisk0: NVME_SMS:vm-106-disk-1,size=4M
ide2: none,media=cdrom
machine: q35
memory: 4096
name: HAOS
net0: virtio=16:42:50:80:BF:E5,bridge=vmbr0,firewall=1
numa: 0
onboot: 1
ostype: l26
scsihw: virtio-scsi-single
smbios1: uuid=cf66a7bc-1960-4d89-b3c9-b08c01a9ddf4
sockets: 1
startup: order=0,up=30
usb0: host=3-1,usb3=1
virtio1: NVME_SMS:vm-106-disk-2,iothread=1,size=60G
vmgenid: 4cb1066f-7cdd-4604-8c63-f57c4042eb20
```

agners, 2024-04-15:
From the screenshot it looks like you have two disks attached now (vda and sda). If both are disks with a full HAOS installation, this is very problematic: We use partition and file system labels to identify the right partitions. Now these labels are in the system twice, which means the system might pick one of the partition at random.

Make sure you have only one disk with HAOS installed on attached to your VM.

It seems the main data disk is used from `/dev/vda8`, so I'd try to detach whatever `/dev/sda` is (I guess the SCSI disk).

agners, 2024-04-15:
Also, can you run `blkid` to see all partitions and disks in your system?

Onepamopa, 2024-04-15:
@agners You're spot on!
I completely forgot to detach the old disk, just changed the boot order to boot from the new one.
Its entry is at the very bottom of the VM config and didn't get selected/copied.

Now.. detached the old disk, updated to 12.2 - all good.

closed 2024-04-15 by agners