home-assistant/operating-system #3312: Memory leak (outside home-assistant?)

Werewolfke, 2024-04-15:
### Describe the issue you are experiencing

The vm hassio has 6Gb of ram, and was using almost all of it.
Looking in homeassistant itself, it says it was using only 0.8Gb, rebooting homeassistant didn't help, just a little bit, a small dip, yet it was still above 5Gb of usage

I had to do a full reboot of the virtual machine for it to clear. Uptime of the full vm is around a week, home assistant itself probably has rebooted since then, since it updates automatically and reboots after. 

Maybe something in the host-logs show something? 
(I rebooted around 21h, so weird that nothing shows between 19h20 and then... Or even after the reboot... 
Screenshot shows it gradually climbing... Since the boot of the vm. 

Core
2024.4.3
Supervisor
2024.04.1
Operating System
12.2
Frontend
20240404.2 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Boot hassio
2. Use it for a week
3. Poof
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Not that I see or saw...
```

### Anything in the Host logs that might be useful for us?

```txt
2024-04-15 19:07:28.515 homeassistant kernel: veth6e99efc: entered promiscuous mode
2024-04-15 19:07:28.515 homeassistant kernel: kauditd_printk_skb: 318 callbacks suppressed
2024-04-15 19:07:28.515 homeassistant kernel: audit: type=1700 audit(1713208048.514:122): dev=veth6e99efc prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
2024-04-15 19:07:28.516 homeassistant kernel: audit: type=1300 audit(1713208048.514:122): arch=c000003e syscall=44 success=yes exit=40 a0=c a1=c0013fd590 a2=28 a3=0 items=0 ppid=1 pid=498 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj=unconfined key=(null)
2024-04-15 19:07:28.516 homeassistant kernel: audit: type=1327 audit(1713208048.514:122): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D636F6E7461696E6572643D2F72756E2F636F6E7461696E6572642F636F6E7461696E6572642E736F636B
2024-04-15 19:07:28.516 homeassistant NetworkManager[414]: <info>  [1713208048.5150] manager: (veth8bf85bb): new Veth device (/org/freedesktop/NetworkManager/Devices/11)
2024-04-15 19:07:28.516 homeassistant NetworkManager[414]: <info>  [1713208048.5153] manager: (veth6e99efc): new Veth device (/org/freedesktop/NetworkManager/Devices/12)
2024-04-15 19:07:28.570 homeassistant systemd[1]: Started libcontainer container 90373eb8f151e4002d9c15f877d9ec5a87b0d5415581e2a49f7dc65390fc2747.
2024-04-15 19:07:28.575 homeassistant kernel: audit: type=1334 audit(1713208048.573:123): prog-id=26 op=LOAD
2024-04-15 19:07:28.575 homeassistant kernel: audit: type=1334 audit(1713208048.573:124): prog-id=27 op=LOAD
2024-04-15 19:07:28.575 homeassistant kernel: audit: type=1300 audit(1713208048.573:124): arch=c000003e syscall=321 success=yes exit=16 a0=5 a1=c00016d928 a2=78 a3=0 items=0 ppid=1115 pid=1125 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
2024-04-15 19:07:28.575 homeassistant kernel: audit: type=1327 audit(1713208048.573:124): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F39303337336562386631353165343030326439633135663837
2024-04-15 19:07:28.575 homeassistant kernel: audit: type=1334 audit(1713208048.573:125): prog-id=28 op=LOAD
2024-04-15 19:07:28.575 homeassistant kernel: audit: type=1300 audit(1713208048.573:125): arch=c000003e syscall=321 success=yes exit=18 a0=5 a1=c00016d6c0 a2=78 a3=0 items=0 ppid=1115 pid=1125 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
2024-04-15 19:07:28.575 homeassistant kernel: audit: type=1327 audit(1713208048.573:125): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F39303337336562386631353165343030326439633135663837
2024-04-15 19:07:28.614 homeassistant kernel: eth0: renamed from veth8bf85bb
2024-04-15 19:07:28.620 homeassistant NetworkManager[414]: <info>  [1713208048.6201] device (veth6e99efc): carrier: link connected
2024-04-15 19:07:28.620 homeassistant kernel: hassio: port 3(veth6e99efc) entered blocking state
2024-04-15 19:07:28.620 homeassistant kernel: hassio: port 3(veth6e99efc) entered forwarding state
2024-04-15 19:07:28.724 homeassistant NetworkManager[414]: <info>  [1713208048.7236] manager: (veth051cb41): new Veth device (/org/freedesktop/NetworkManager/Devices/13)
2024-04-15 19:07:28.724 homeassistant NetworkManager[414]: <info>  [1713208048.7240] manager: (veth77c93b3): new Veth device (/org/freedesktop/NetworkManager/Devices/14)
2024-04-15 19:07:28.724 homeassistant kernel: hassio: port 4(veth77c93b3) entered blocking state
[... 144 lines cut ...]

dashboards | 9
-- | --
resources | 15
views | 16
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 5, 2024 at 17:02
-- | --
current_recorder_run | April 15, 2024 at 21:07
estimated_db_size | 371.70 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

[image]
[image]
[image]

Impact123, 2024-04-15:
PVE also includes cache and buffers in this stat. At least on linux VMs. See here: <https://www.google.com/search?q=wrong+memory+usage+site%3Aproxmox.com>
If you want to show PVE statistics please show them in the actual GUI, not a random app.

Werewolfke, 2024-04-16:
The random app is a proxmox monitoring app. I could not get into pve at that time.
I will look into it and see what I can do, but I know I had to increase ram from 4Gb to 6 because it crashed otherwise, so I didn't think it were to be caching.
If it is 100% caching then this issue can be closed.

Impact123, 2024-04-16:
You can check the used/free memory with the `glances` addon or by running `free -h` via SSH. I recommend you visit the discord server for help.